MichaIng/DietPi #1585: DietPi-Software | Replace SubSonic 5 with AirSonic

Fourdee, 2018-03-04:
> https://github.com/Fourdee/DietPi/issues/1582#issuecomment-370235356

Fourdee, 2018-03-04:
Install/run test, no web interface:
```
root@DietPi:~# /usr/bin/java -Xmx700m -Dairsonic.home=/mnt/dietpi_userdata/airsonic -Dserver.context-path=/airsonic -Dserver.port=8080 -jar /mnt/dietpi_userdata/airsonic/airsonic.war
           _                       _
     /\   (_)                     (_)
    /  \   _ _ __  ___  ___  _ __  _  ___
   / /\ \ | | '__|/ __|/ _ \| '_ \| |/ __|
  / ____ \| | |   \__ \ (_) | | | | | (__
 /_/    \_\_|_|   |___/\___/|_| |_|_|\___|

                        10.1.1-RELEASE

2018-03-04 15:41:17.865  INFO --- org.airsonic.player.Application          : Starting Application v10.1.1-RELEASE on DietPi with PID 2486 (/mnt/dietpi_userdata/airsonic/airsonic.war started by root in /root)
2018-03-04 15:41:17.888  INFO --- org.airsonic.player.Application          : The following profiles are active: legacy
2018-03-04 15:41:34.442  INFO --- o.a.p.service.SettingsService            : Java: 1.8.0_131, OS: Linux
2018-03-04 15:42:06.130  INFO --- liquibase                                : Successfully acquired change log lock
2018-03-04 15:42:14.637  INFO --- liquibase                                : Reading from DATABASECHANGELOG
2018-03-04 15:42:15.735  INFO --- liquibase                                : Successfully released change log lock
2018-03-04 15:42:15.759  INFO --- liquibase                                : Successfully acquired change log lock
2018-03-04 15:42:15.811  INFO --- liquibase                                : Reading from DATABASECHANGELOG
2018-03-04 15:42:16.279  INFO --- liquibase                                : Successfully released change log lock
2018-03-04 15:42:18.587  INFO --- o.a.p.service.PodcastService             : Automatic Podcast update scheduled to run every 24 hour(s), starting at Sun Mar 04 15:47:18 GMT 2018
2018-03-04 15:42:19.631  INFO --- o.a.p.s.MediaScannerService              : Automatic media library scanning scheduled to run every 1 day(s), starting at Mon Mar 05 03:00:00 GMT 2018
2018-03-04 15:42:39.888  INFO --- org.airsonic.player.Application          : Started Application in 98.033 seconds (JVM running for 101.093)
```

Fourdee, 2018-03-04:
Online doc : http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=11280#p11280

Fourdee, 2018-03-04:
Completed.

Fourdee, 2018-03-04:
Patch removal of 33

closed 2018-03-04 by Fourdee

MichaIng, 2018-03-04:
👍 Wow that was fast, nice work!!