MichaIng/DietPi #1587: do you always have to run dietpi from and usb..?

janbildsoehansen, 2018-03-04:
## Creating a bug report/issue:
wanna install the x86 on a computer but how do i make it boot from the ssd only with out the usb

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-03-04:
@janbildsoehansen 

> wanna install the x86 on a computer but how do i make it boot from the ssd only with out the usb

#### Method 1:
- Plug the SSD into a USB <> SATA converter. 
- Write the DietPi image to the SSD.
- Plug back into PC and power on.

#### Method 2:
- ```dd```, TBA

janbildsoehansen, 2018-03-04:
Thanks for the quick response :D i will try so :)

best regards Jan

Fourdee, 2018-03-06:
@janbildsoehansen 

I'll mark this as closed, question answered. Please reopen if required.

closed 2018-03-06 by Fourdee