MichaIng/DietPi #1591: ntp behind http/tcp proxy

TheRinger, 2018-03-08:
NTP service is using UDP protocol to sync the time. So HTTP/TCP proxy may not work for it.
An alternative for this is htpdate package available in debian/ubuntu repos, it would allow bootstrapping installs behind http/tcp proxies.

MichaIng, 2018-03-13:
@TheRinger 
Thanks for your request.

Actually we plan to fully switch to systemd-timesyncd, but as far as I could find out, the issue is the same there.

If considering an alternative to sync through proxies, I would take a look at tlsdate: https://github.com/ioerror/tlsdate
Secure TLS connection at least. What I don't like about both projects, is that they are not actively developed for years... But use case is reasonable, although I would expect a local ntp server behind a http proxy always :wink:.

TheRinger, 2018-03-17:
I'll do some reading about systemd-timesyncd. Comcast's public internet gateway (xfinitywifi) in the United States blocks udp so I ended up using htpdate in order to sync.

Fourdee, 2018-07-09:
> Actually we plan to fully switch to systemd-timesyncd

I'll mark as closed.

However, if this is still an issue for any user, please reopen.

closed 2018-07-09 by Fourdee