MichaIng/DietPi #1592: Upgrade from 6.2 to 6.3 resets dietpi user password

ragadeeshu, 2018-03-08:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
6.2 to 6.3
- SBC Device (EG: RPi 3)?
Asus tinkerboard
- Power supply used (EG: 5v 1A RAVpower)?
Rpi official power supply
- SD card used (EG: Sandisk ultra)?
Some fast toshiba thing
- Distro (EG: Jessie) | ```uname -a```?
Linux DietPi 4.4.71+ #1 SMP Wed Nov 15 14:31:48 CST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?
Dunno, don't have the time to check

#### Expected behaviour:

After upgrading to 6.3 all user passwords should be the same as before upgrading
#### Actual behaviour:

The password for the user `dietpi` is reset to `dietpi` after upgrade. This is a huge security concern for devices with an exposed ssh endpoint.

#### Steps to reproduce:

Install dietpi 6.2, change the password for the dietpi user, upgrade to 6.3.

#### Did you submit a ```dietpi-bugreport```?

Nope.

#### Extra details:

MichaIng, 2018-03-08:
@ragadeeshu 
Thanks for your reports. Yes that is true, to implement a new boot service we rerun the core environment script, which also resets dietpi password: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_core_environment#L45

@Fourdee 
Maybe we can check for user existence first before resetting/setting password. Generally I think setting core environment might reset too much things, that user could have adjusted reasonably. Maybe do needed e.g. service patches individually, even if this is higher effort than just running `dietpi-set_core_environment`? Otherwise we can go through and merge uncritical steps into some kind of "patch" option, that excludes critical changes, such as passwords and files that users might want to adjust?

Fourdee, 2018-03-08:
@MichaIng 

> Generally I think setting core environment might reset too much things, that user could have adjusted reasonably. Maybe do needed e.g. service patches individually, even if this is higher effort than just running dietpi-set_core_environment? 

Yep, agree 👍 
Manual patch changes from v6.3 onwards, avoiding the use of ```dietpi-set_core_environment``` during patches.

We'll need to leave the current entries for ```dietpi-set_core_environment``` in the v6.3< patch file, as it will be near impossible to back track all the manual changes required, most likely creating more issues down the line.

> useradd -m -s /bin/bash "$DIETPI_USERNAME" -p "$DIETPI_PASSWORD"

Yep 👍 , we can avoid the password reset in ```dietpi-set_core_environment``` to prevent future occurrences during v6.-1 > v6.3.

@ragadeeshu 

Apologies, password reset for user ```dietpi``` will no longer occur as of v6.3 and onwards.

Fourdee, 2018-03-08:
@MichaIng 
Sent commit: https://github.com/Fourdee/DietPi/commit/87e941005ff62ade3b2b799c1ae9e66bba976d90

Fourdee, 2018-03-09:
Resolved for v6.4.

closed 2018-03-09 by Fourdee