MichaIng/DietPi #1593: Failed update to 6.3 from 6.2

theverymrp, 2018-03-08:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v6.2
- SBC Device     | RPi 2 Model B (armv7l) (index=2)
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:
update completes

#### Actual behaviour:
update fails

#### Steps to reproduce:
run dietpi-update

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
Log file contents:
Get:1 http://download.mono-project.com/repo/debian stretch InRelease [5,863 B]
Ign:1 http://download.mono-project.com/repo/debian stretch InRelease
Hit:2 https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease
Err:2 https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease
  Splitting up /var/lib/apt/lists/www.mirrorservice.org_sites_archive.raspbian.org_raspbian_dists_stretch_InRelease into data and signature failed
Get:3 https://apt.sonarr.tv develop InRelease [13.3 kB]
Ign:3 https://apt.sonarr.tv develop InRelease
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Err:4 https://archive.raspberrypi.org/debian stretch InRelease
  Splitting up /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_InRelease into data and signature failed
Reading package lists...
W: GPG error: http://download.mono-project.com/repo/debian stretch InRelease: Splitting up /var/lib/apt/lists/partial/download.mono-project.com_repo_debian_dists_stretch_InRelease into data and signature failed
W: The repository 'http://download.mono-project.com/repo/debian stretch InRelease' is not signed.
E: Write error - write (28: No space left on device)
E: Write error - write (28: No space left on device)
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease: Splitting up /var/lib/apt/lists/www.mirrorservice.org_sites_archive.raspbian.org_raspbian_dists_stretch_InRelease into data and signature failed
W: GPG error: https://apt.sonarr.tv develop InRelease: Splitting up /var/lib/apt/lists/partial/apt.sonarr.tv_dists_develop_InRelease into data and signature failed
W: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.
E: Write error - write (28: No space left on device)
E: Write error - write (28: No space left on device)
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://archive.raspberrypi.org/debian stretch InRelease: Splitting up /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_InRelease into data and signature failed
E: Write error - write (28: No space left on device)
E: Write error - write (28: No space left on device)
E: The package lists or status file could not be parsed or opened.
```

MichaIng, 2018-03-08:
@theverymrp 
Thanks for your report. I am afraid the fix for your issue is provided with the update, but AFTER the issue already occurs and leads to failure: https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L328

Ref: https://github.com/Fourdee/DietPi/issues/1388

Please do the following before running dietpi-update:
`G_AGI dirmngr`

I am wondering why this suddenly appeared, as before I never faced the need of package `dirmngr` for any repo... Please report back, if it works, which will underline the need of this package nowadays.

theverymrp, 2018-03-08:
thanks @MichaIng. I've done as requested but am now facing a second issue. I can't run dietpi-update/config/launcher anymore, command not found :(

MichaIng, 2018-03-08:
Ah I guess this is due to new updated files are already in place but new necessary services was not yet implemented as part of patch_file process, leading to missing sourceing of dietpi-globals.

Please do:
`source /DietPi/dietpi/func/dietpi-globals`

Fourdee, 2018-03-08:
@theverymrp 

> E: Write error - write (28: No space left on device)

Lets check space, please paste results:
```
df -h
```

> A small hotfix update, what do you think? Sadly would need some manual implementation of G_AGI dirmngr outside of patch_file or ignoring errors during G_AGUP.

@MichaIng 

Yep 👍 I'll setup a system and replicate the issue, just want to runs some checks.

theverymrp, 2018-03-08:
@MichaIng perfect, that worked and update now successfully completed :)

@Fourdee 

Filesystem              Size  Used Avail Use% Mounted on
/dev/root                30G  1.5G   27G   6% /
devtmpfs                484M     0  484M   0% /dev
tmpfs                   489M  8.0K  489M   1% /dev/shm
tmpfs                   489M  6.4M  483M   2% /run
tmpfs                   5.0M     0  5.0M   0% /run/lock
tmpfs                   489M     0  489M   0% /sys/fs/cgroup
tmpfs                    10M  1.5M  8.6M  15% /DietPi
tmpfs                    20M   32K   20M   1% /var/log
tmpfs                   488M  1.5M  487M   1% /tmp
/dev/mmcblk0p1           41M   23M   19M  54% /boot

MichaIng, 2018-03-08:
> E: Write error - write (28: No space left on device)

Ah tats true, just saw the GPG error on apt update and remembered the issues users faced with missing dirmngr.

Fourdee, 2018-03-08:
@theverymrp 

> tmpfs 488M 1.5M 487M 1% /tmp

Thanks. 👍 

@MichaIng 
0 swapfile (tmpfs out of space?), believe this is only on the RPi v6.2 image (bug, swapfile gen set to 0). But we should patch for auto swapfile size and regen during v6.4 just to make sure.

Fourdee, 2018-03-08:
@MichaIng 

Yep, Warnings do not cause ```G_AGUP``` to fail, only the errors, which I believe are related to 0 swapfile and tmpfs out of free space:
```
root@DietPi:~# G_AGUP
[ INFO ] APT update, please wait...
Get:1 http://download.mono-project.com/repo/debian raspbianstretch InRelease [2,224 B]
Hit:2 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch InRelease
Ign:1 http://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Fetched 2,224 B in 1s (1,930 B/s)
Reading package lists...
W: GPG error: http://download.mono-project.com/repo/debian raspbianstretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY A6A19B38D3D831EF
W: The repository 'http://download.mono-project.com/repo/debian raspbianstretch InRelease' is not signed.
[  OK  ] G_AGUP
```

Fourdee, 2018-03-09:
Resolved after v6.4 patch.

closed 2018-03-09 by Fourdee