MichaIng/DietPi #1594: Problem with PREP_SYSTEM_FOR_DIETPI.sh Failure

Elluminatus, 2018-03-08:
#### Required Information:
- DietPi Version | v.
- SBC Device     | NULL (index=22)
- Distro         | jessie (index=3)
- Command        | G_AGDUG
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:
packages shoukd be authenticated

#### Actual behaviour:
packages cannot be authenticated

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
  libkrb5-3 libkrb5support0 libldap-2.4-2 liblocale-gettext-perl liblzma5
  libmount1 libncurses5 libncursesw5 libnewt0.52 libnih-dbus1 libnih1
  libopts25 libp11-kit0 libpam-modules libpam-modules-bin libpam-runtime
  libpam0g libparted2 libpcre3 libpopt0 librtmp1 libsasl2-2
  libsasl2-modules-db libselinux1 libsemanage-common libsemanage1 libsepol1
  libslang2 libsmartcols1 libss2 libssh2-1 libstdc++6 libsystemd0 libtasn1-6
  libtext-charwidth-perl libtext-wrapi18n-perl libtinfo5 libudev1 libusb-0.1-4
  libusb-1.0-0 libustr-1.0-1 libuuid1 locales login lsb-base mawk mount
  multiarch-support nano ncurses-base ncurses-bin net-tools netbase ntfs-3g
  ntp openssl parted passwd perl-base plymouth procps psmisc readline-common
  sed sensible-utils startpar sudo systemd systemd-sysv sysv-rc sysvinit-utils
  tar tzdata udev unzip usbutils util-linux wget whiptail xkb-data xz-utils
  zlib1g
169 upgraded, 37 newly installed, 2 to remove and 0 not upgraded.
Need to get 59.7 MB of archives.
After this operation, 21.2 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  locales libc-l10n libdbus-1-3 libnih-dbus1 libnih1 libc6 libc-bin debconf
  libgpg-error0 libgcrypt20 liblz4-1 libapparmor1 libip4tc0 libpcre3
  libselinux1 libmount1 libseccomp2 libsmartcols1 libtinfo5 libncurses5
  libncursesw5 base-files sensible-utils debianutils bash libfdisk1
  bash-completion util-linux init-system-helpers sysvinit-utils startpar
  insserv sysv-rc initscripts mount kbd coreutils console-setup-linux
  libtext-charwidth-perl perl-base liblocale-gettext-perl e2fslibs e2fsprogs
  klibc-utils initramfs-tools console-setup keyboard-configuration libklibc
  cpio libudev1 libsystemd0 hdparm systemd-sysv udev systemd kmod libkmod2
  gcc-6-base libgcc1 libattr1 acl libacl1 libaudit-common libcap-ng0 libaudit1
  libpam0g libsepol1 libsemanage-common libsemanage1 bzip2 libbz2-1.0
  libustr-1.0-1 libpam-modules-bin libpam-modules passwd adduser libprocps6
  procps initramfs-tools-core linux-base xkb-data lsb-base libcap2
  libdevmapper1.02.1 dmsetup libuuid1 libcryptsetup4 libidn11 liblzma5
  libblkid1 init bsdutils tar libpam-runtime cron dpkg dash diffutils
  findutils install-info grep gzip hostname login ncurses-bin sed libstdc++6
  zlib1g libapt-inst2.0 apt-utils libapt-pkg5.0 apt libdb5.3 gpgv
  libdebconfclient0 base-passwd ncurses-base mawk libgmp10 libnettle6
  libhogweed4 libffi6 libp11-kit0 libtasn1-6 libgnutls30 libntfs-3g871 ntfs-3g
  netbase libbsd0 libedit2 libopts25 libssl1.1 ntp libassuan0 pinentry-curses
  libnpth0 readline-common libreadline7 gnupg-agent libksba8 libsqlite3-0
  gnupg libslang2 libcomerr2 libss2 multiarch-support tzdata libusb-0.1-4
  libelf1 libmnl0 iproute2 iputils-ping libisc-export160 libssl1.0.2
  libdns-export162 isc-dhcp-client isc-dhcp-common libnewt0.52 libpopt0 nano
  libunistring0 libidn2-0 libpsl5 wget whiptail libgnutls-openssl27
  libkeyutils1 libgssapi-krb5-2 libkrb5-3 libk5crypto3 libkrb5support0
  libsasl2-modules-db libsasl2-2 libldap-common libldap-2.4-2 xz-utils
  libnghttp2-14 librtmp1 libssh2-1 libcurl3-gnutls apt-transport-https openssl
  ca-certificates curl libcurl3 dc dosfstools dphys-swapfile fbset fuse
  libfuse2 libcap2-bin libdrm2 parted libparted2 plymouth libpng16-16
  libplymouth4 libtext-wrapi18n-perl libusb-1.0-0 net-tools psmisc sudo unzip
  usbutils fake-hwclock
E: There are problems and -y was used without --force-yes

MichaIng, 2018-03-08:
@Elluminatus 
Thank you for your report.

We define the `--force-yes` (on Stretch+ `--allow-unauthenticated` and others) options but do not apply them within `G_AGDUG` function. Might be due to possible issues of applying all of them, as they are a bid dangerous, or we just forgot it: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-globals#L1438-L1463

Please try the following, from within directory where you downloaded `PREP_SYSTEM_FOR_DIETPI.sh` to:
`sed -i 's/G_AGDUG/apt-get dist-upgrade --force-yes/g' PREP_SYSTEM_FOR_DIETPI.sh`
Then rerun the script.

Note that running the preparation on generic devices is experimental as it is hard to predict exact outcome dependant on initial system. Might I ask which device/os-image you are running the script on?

@Fourdee 
Is the missing `$options` wanted or just forgot to add 😉?
Do you know, what is necessary for authenticating packages? `debian-archive-keyring`, `gnupg` and now `dirmngr` is installed before `G_AGDUG`, maybe the new keys need to be somehow activated/added?
Ref.: https://wiki.debian.org/SecureApt

Fourdee, 2018-03-08:
@MichaIng 

> Is the missing $options wanted or just forgot to add 😉?

Probably forgot 😃. I'll add it back in.

Fourdee, 2018-03-08:
@Elluminatus 

Which device are you running?

Fourdee, 2018-03-08:
@MichaIng 

> Do you know, what is necessary for authenticating packages? debian-archive-keyring, gnupg and now dirmngr is installed before G_AGDUG

Those should cover it, as long as any additional added repos have key installed, where available.

SuBLiNeR, 2018-03-09:
@Fourdee just a lil hint upon version numbering: Just use something like 6.3.1 for low updates and 6.x for higher and better fixes and updates! ;) 
If you'll just use more than 6.x updates, i'll guess we'll see 7.x by June'18 :octopus:

SuBLiNeR, 2018-03-09:
Debian Buster may be a solution..... as myself if i use "Buster" upon Stretch with "Prep", i have no problmes so far..... is it only myself?

MichaIng, 2018-03-09:
@SuBLiNeR @Fourdee 
> Just use something like 6.3.1 for low updates

I thought about the same, doing some subsub versioning for hotfix updates. But as it does not affect function, I would keep this with low priority in mind 😉.

> Debian Buster may be a solution

Distro upgrades always bear the risk of errors, I guess the APT package and system dependencies are just too complex to totally avoid. `--force-yes` (on Jessie) respectively `--allow-unauthenticated` (Stretch and afterwards) will do the job here.

Debian Buster generally works fine, I run a VM here since longer without any issues. Raspbian Buster on the other hand had often package updates with missing dependencies, leading in worst case to the removal of a large bunch of dependant packages, if you force apt (full-)upgrade. Read my status report: https://github.com/Fourdee/DietPi/issues/1286#issuecomment-366984565
Both have in common a still high rate of package updates. Nearly every day I can apt upgrade my Buster systems, even that the amount of installed packages is even lower than on regular DietPi and therefore **way** lower than on regular Debian/Raspbian. If you wait for 2 weeks, than the high amount of updates is already risky on it's own (same as distro upgrades), thus I would so for definitely not recommend Buster for end users.

Elluminatus, 2018-03-17:
Hi I am using a banana pi m3 with a raspbian jessie using kernel BPI-M3 kernel 3.4
Hope this might help?

Elluminatus, 2018-03-17:
Okay i tried it again. Now it seems like the whole process is working. But when i try to reboot after the system told me to do the system is not starting the heartbeat led is blinking three to four time and the banana pi logo shows up, but then nothing else happens...

k-plan, 2018-03-18:
@Elluminatus 

> the system restarts and I can see 4 failures in red passing by, but to fast to read.

Without a the output of serial port while boot up (or not) you will never will get closer to the issue.

http://linux-sunxi.org/Sinovoip_Banana_Pi_M3#Adding_a_serial_port

Elluminatus, 2018-03-19:
Hi thak you for the hint. I found a usb to uart adapter and monitored the output: See here:

`HELLO! BOOT0 is starting!
boot0 version : 4.0.0
rtc[1] value = 0x00010000
rtc[2] value = 0x00020000
rtc[3] value = 0x00030000
DRAM DRIVE INFO: V1.0
DRAM Type =7 (2:DDR2,3:DDR3,6:LPDDR2,7:LPDDR3)
DRAM normal_mode value: 00000001
DRAM CLK =672 MHZ
PMU: AXP81X
ddr voltage = 1200 mv
using ic A83T
using axp AXP813
DRAM zq value: 00003bfb
dram size =2048
card boot number = 0
card no is 0
sdcard 0 line count 0
[mmc]: mmc driver ver 2014-8-11 15:06:39
[mmc]: ***Try SD card 0***
[mmc]: SD/MMC Card: 4bit, capacity: 15193MB
[mmc]: vendor: Man 00035344 Snr ab691800
[mmc]: product: SB16G
[mmc]: revision: 8.0
[mmc]: ***SD/MMC 0 init OK!!!***
sdcard 0 init ok
ERROR! NOT find the head of uboot.
The size of uboot is 000d8000.
sum=ce3ec68c
src_sum=ce3ec68c
Succeed in loading uboot from sdmmc flash.
Ready to disable icache.
Jump to secend Boot.
[      0.314]

U-Boot 2011.09-rc1-00000-gfe0f876 (May 03 2016 - 13:13:31) Allwinner Technology

[      0.323]version: 2.1.0
normal mode
[      0.403]pmbus:   normal or secure os
ready
[      0.407]PMU: AXP81X
[      0.410]PMU: AXP81X found
bat_vol=4504, ratio=100
[      0.415]PMU: dcdc2 900
[      0.418]PMU: pll1 1008 Mhz
AXI0=336 Mhz,PLL_PERIPH =1200 Mhz AHB1=200 Mhz, APB1=100 Mhz
set power on vol to default
dcdc1_vol = 3300, onoff=1
dcdc2_vol = 900, onoff=1
dcdc3_vol = 900, onoff=1
dcdc4_vol = 900, onoff=1
dcdc6_vol = 900, onoff=1
aldo1_vol = 1800, onoff=0
aldo2_vol = 1800, onoff=1
aldo3_vol = 3000, onoff=1
eldo2_vol = 1800, onoff=0
dldo1_vol = 3300, onoff=0
dldo2_vol = 3300, onoff=0
dldo3_vol = 2500, onoff=1
dldo4_vol = 2800, onoff=0
fldo2_vol = 900, onoff=1
gpio0_vol = 3300, onoff=1
gpio1_vol = 3300, onoff=1
find power_sply to end
supply=dcdc1
axp=axp81x, supply=dcdc1, vol=3300
reg addr=0x1c20b04, value=0xd, pmu_vol=3300
[... 989 lines cut ...]
[    6.892701] [<c00103c4>] (__irq_svc+0x44/0x78) from [<c001158c>] (default_idle+0x38/0x40)
[    6.892701] [<c001158c>] (default_idle+0x38/0x40) from [<c00119bc>] (cpu_idle+0xac/0x108)
[    6.892701] [<c00119bc>] (cpu_idle+0xac/0x108) from [<c075d4e8>] (secondary_start_kernel+0x10c/0x130)
[    6.892701] [<c075d4e8>] (secondary_start_kernel+0x10c/0x130) from [<4075cad4>] (0x4075cad4)
[    6.892705] CPU5: stopping
[    6.892705] [<c0017b8c>] (unwind_backtrace+0x0/0x128) from [<c07604e0>] (dump_stack+0x20/0x24)
[    6.892705] [<c07604e0>] (dump_stack+0x20/0x24) from [<c0016410>] (handle_IPI+0x17c/0x308)
[    6.892705] [<c0016410>] (handle_IPI+0x17c/0x308) from [<c000a3d4>] (gic_handle_irq+0xa4/0xb4)
[    6.892705] [<c000a3d4>] (gic_handle_irq+0xa4/0xb4) from [<c00103c4>] (__irq_svc+0x44/0x78)
[    6.892705] Exception stack(0xe6927f68 to 0xe6927fb0)
[    6.892705] 7f60:                   00000007 c0b5d19c 00000000 00000000 e6926000 c0c93b48
[    6.892705] 7f80: c076e7a4 c0b61810 4000406a 410fc075 00000000 e6927fbc e6927fb0 e6927fb0
[    6.892705] 7fa0: c0011588 c001158c 60000113 ffffffff
[    6.892705] [<c00103c4>] (__irq_svc+0x44/0x78) from [<c001158c>] (default_idle+0x38/0x40)
[    6.892705] [<c001158c>] (default_idle+0x38/0x40) from [<c00119bc>] (cpu_idle+0xac/0x108)
[    6.892705] [<c00119bc>] (cpu_idle+0xac/0x108) from [<c075d4e8>] (secondary_start_kernel+0x10c/0x130)
[    6.892705] [<c075d4e8>] (secondary_start_kernel+0x10c/0x130) from [<4075cad4>] (0x4075cad4)
[    8.912543] sun8i hotplug: try to kill cpu:0 failed!
[    9.921477] sun8i hotplug: try to kill cpu:1 failed!
[   10.930345] sun8i hotplug: try to kill cpu:2 failed!
[   11.939129] sun8i hotplug: try to kill cpu:3 failed!
[   12.947850] sun8i hotplug: try to kill cpu:4 failed!
[   13.956495] sun8i hotplug: try to kill cpu:5 failed!
[   14.965066] sun8i hotplug: try to kill cpu:7 failed!
`

Fourdee, 2018-03-19:
@Elluminatus 
Thanks 👍 

Notes:
```
[ 6.769861] No filesystem could mount root, tried: ext3 ext2 ext4 squashfs vfat msdos iso9660 ntfs fuseblk udf
[ 6.783765] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)
```

After running PREP, and before rebooting, please try disabling FS expansion:
```
		rm /etc/systemd/system/dietpi-fs_expand.service
		rm /var/lib/dietpi/fs_partition_resize.sh
		rm /etc/systemd/system/dietpi-fs_partition_resize.service
		systemctl daemon-reload
```

Also, if we can get the following before the reboot, may help:
```
blkid
df -h
cat /etc/fstab
```

Elluminatus, 2018-03-19:
blkid:
root@bpi-iot-ros-ai:/home/pi# blkid
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="BPI-BOOT" UUID="CE95-5411" TYPE="vfat" PARTUUID="000bc5c9-01"
/dev/mmcblk0p2: LABEL="BPI-ROOT" UUID="4e362d90-5b5d-4e56-bdc8-493003f13ef7" TYPE="ext4" PARTUUID="000bc5c9-02"
/dev/sda1: LABEL="Volume" UUID="98E440A0E440830C" TYPE="ntfs" PARTUUID="b5b1c71d-01"
/dev/mmcblk1p1: SEC_TYPE="msdos" LABEL="BPI-BOOT" UUID="CE95-5411" TYPE="vfat" PARTUUID="70b18045-01"
/dev/mmcblk0: PTUUID="000bc5c9" PTTYPE="dos"
/dev/mmcblk1: PTUUID="70b18045" PTTYPE="dos"
/dev/mmcblk1p2: PARTUUID="70b18045-02"

df -h
root@bpi-iot-ros-ai:/home/pi# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       4.5G  640M  3.7G  15% /
devtmpfs        750M     0  750M   0% /dev
tmpfs          1006M     0 1006M   0% /dev/shm
tmpfs          1006M   18M  989M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs          1006M     0 1006M   0% /sys/fs/cgroup
/dev/mmcblk0p1  256M  190M   67M  74% /boot
tmpfs           202M     0  202M   0% /run/user/1000
/dev/mmcblk1p1  256M     0  256M   0% /media/pi/BPI-BOOT
tmpfs          1005M     0 1005M   0% /tmp
tmpfs            20M     0   20M   0% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
/dev/sda1       299G   96M  298G   1% /mnt/98E440A0E440830C

cat /etc/fstab
root@bpi-iot-ros-ai:/home/pi# cat /etc/fstab
#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#TMPFS / MISC ------------------------------------------------------
proc            /proc           proc    defaults                               0 0
tmpfs /tmp tmpfs defaults,size=1005M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs                   /var/log                tmpfs   defaults,size=20m,noatime,nodev,nosuid,mode=1777        0 0
tmpfs                   /DietPi                 tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777        0 0

#Internal Drives---------------------------------------------------
UUID=CE95-5411    /boot    auto    defaults,noatime    0 0
UUID=4e362d90-5b5d-4e56-bdc8-493003f13ef7    /    auto    defaults,noatime    0 0
UUID=98E440A0E440830C    /mnt/98E440A0E440830C    auto    defaults,noatime,x-systemd.automount,nofail    0 0

MichaIng, 2018-03-22:
@Elluminatus 
The error messages you faced (besides DietPi-RAMlog) appear due to failed DietPi-RAMdisk.
Could you show output of
- `systemctl status dietpi-ramdisk`
- `journalctl -u dietpi-ramdisk`
- /var/tmp/dietpi/logs/dietpi-ramdisk.log
- /var/tmp/dietpi/logs/dietpi-ramlog.log

And could you try to start RAMdisk manully, maybe this gives more output: `/boot/dietpi/dietpi-ramdisk 0`

Elluminatus, 2018-03-23:
**systemctl status dietpi-ramdisk**

root@bpi-iot-ros-ai:/home/pi# systemctl status dietpi-ramdisk
● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor p
reset: enabled)
   Active: inactive (dead) since Fri 2018-03-23 16:55:04 GMT; 8min ago

Mar 23 16:53:11 bpi-iot-ros-ai systemd[1]: Starting DietPi-RAMdisk...
Mar 23 16:53:12 bpi-iot-ros-ai systemd[1]: Started DietPi-RAMdisk.
Mar 23 16:53:12 bpi-iot-ros-ai systemd[1]: dietpi-ramdisk.service: Faile
d to set invocation ID on control group /system.slice/dietpi-ramdisk.service, ig
noring: Operation not supported
Mar 23 16:55:03 bpi-iot-ros-ai systemd[1]: Stopping DietPi-RAMdisk...
Mar 23 16:55:04 bpi-iot-ros-ai systemd[1]: Stopped DietPi-RAMdisk.

**journalctl -u dietpi-ramdisk**

root@bpi-iot-ros-ai:/home/pi# journalctl -u dietpi-ramdisk
-- Logs begin at Wed 2016-07-13 04:34:29 BST, end at Fri 2018-03-23 17:03:20 GMT
. --
Mar 23 16:53:11 bpi-iot-ros-ai systemd[1]: dietpi-ramdisk.service: Faile
d to set invocation ID on control group /system.slice/dietpi-ramdisk.service, ig
noring: Operation not supported
Mar 23 16:53:11 bpi-iot-ros-ai systemd[1]: Starting DietPi-RAMdisk...
Mar 23 16:53:12 bpi-iot-ros-ai systemd[1]: Started DietPi-RAMdisk.
Mar 23 16:53:12 bpi-iot-ros-ai systemd[1]: dietpi-ramdisk.service: Faile
d to set invocation ID on control group /system.slice/dietpi-ramdisk.service, ig
noring: Operation not supported
Mar 23 16:55:03 bpi-iot-ros-ai systemd[1]: Stopping DietPi-RAMdisk...
Mar 23 16:55:04 bpi-iot-ros-ai systemd[1]: Stopped DietPi-RAMdisk.

**/var/tmp/dietpi/logs/dietpi-ramdisk.log**

Sat 24 Mar 00:53:11 CST 2018 | DietPi-Ramdisk: Starting
Sat 24 Mar 00:53:11 CST 2018 | DietPi-Ramdisk: Completed
Fri 23 Mar 16:55:03 GMT 2018 | DietPi-Ramdisk: Stopping
Fri 23 Mar 16:55:03 GMT 2018 | DietPi-Ramdisk: Completed

**/var/tmp/dietpi/logs/dietpi-ramlog.log**

DietPi-Ramlog: Starting
cp: cannot stat '/var/lib/dietpi/dietpi-ramlog/storage/*': No such file or dire$
DietPi-Ramlog: Completed
DietPi-Ramlog: Stopping
DietPi-Ramlog: Completed

**/boot/dietpi/dietpi-ramdisk 0**

root@bpi-iot-ros-ai:/home/pi# /boot/dietpi/dietpi-ramdisk 0
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
Fri 23 Mar 17:14:17 GMT 2018 | DietPi-Ramdisk: Starting
Fri 23 Mar 17:14:17 GMT 2018 | DietPi-Ramdisk: Completed

Ready for the next input/step :)

Greetings Elluminatus

Fourdee, 2018-03-23:
> Failed to set invocation ID on control group /system.slice/dietpi-ramdisk.service, ig
noring: Operation not supported

https://github.com/systemd/systemd/issues/5236#issuecomment-277779394?

Elluminatus, 2018-03-23:
Seemann like dietpi 6.4. is runing on a banana pi m3 great work!

Fourdee, 2018-03-28:
I'll mark this as closed, as the original issue has been resolved (https://github.com/Fourdee/DietPi/issues/1594#issue-303552262).

If (https://github.com/Fourdee/DietPi/issues/1594#issuecomment-375979423) is still occurring, please reopen.

closed 2018-03-28 by Fourdee