MichaIng/DietPi #1595: Remove login?

MrGlasspoole, 2018-03-08:
Is it possible to remove the login and DietPi output?
[image]

I already have:
`quiet splash plymouth.ignore-serial-consoles logo.nologo consoleblank=0 loglevel=1`
in the cmdline.txt.

It would be nice to only see it if a keyboard is attached.

MichaIng, 2018-03-09:
@MrGlasspoole 
<s>The DietPi login screen is handled within `~/.bashrc` => `/DietPi/dietpi/login`. Besides the visual output it does some other useful/necessary/wanted steps, thus it should not be removed!
But you could disable the visual output to screen by editing .bashrc:
`sed -i 's|/dietpi/login|/dietpi/login &> /dev/null|' /root/.bashrc`</s>

Whoopsie, it's about the postboot screen:
After v6.3 (wait for v6.4 which will be hotfix release today) you could do:
```
sed -i 's/StandardOutput=tty/StandardOutput=syslog/' /etc/systemd/system/dietpi-postboot.service
systemctl daemon-reload
```
Afterwards output should be redirected to syslog on next boot.

I don't know how to reliably detect plugged keyboard, would need some testing with e.g. `lsusb`: https://serverfault.com/a/126317
But I think it is unlikely that we will add this to DietPi as default. Would need additional checks, systemd unit adjustment daemon-reload in earlier boot stage, leading to longer boot for everyone, just to have default Debian 2 liners output in cases, where I do not see the big benefit 😉.

Fourdee, 2018-03-09:
I believe @MichaIng answer should offer a solution.

I'll mark this as closed, due to the following:
- DietPi boot messages are part of the DietPi experience.
- We do not officially endorse/support removing the DietPi boot messages.

closed 2018-03-09 by Fourdee