MichaIng/DietPi #1598: 3.5 inch LCD white screen

ccc0der, 2018-03-08:
Hello.
I'm trying to set up the [3.5 inch LCD with touch screen](https://ru.aliexpress.com/item/2016-Original-3-5-LCD-TFT-Touch-Screen-Display-with-Acrylic-Case-Compatible-for-Raspberry-Pi/32701628698.html?spm=a2g0s.9042311.0.0.qbVYhv). It was working in both Raspbian and Ubuntu Mate.
But after installing drivers (I tried 3 versions: 170703, 160715 and 151102) I always get the same white screen and nothing happens. I run LCD35-show.
Please, help

Fourdee, 2018-03-09:
@ccc0der 

> But after installing drivers (I tried 3 versions: 170703, 160715 and 151102) 

Hi,

How are you installing the drivers? We'll need to see what commands/scripts are run to assist.

ccc0der, 2018-03-09:
@Fourdee 
I used all drivers from [here](http://www.raspberrypiwiki.com/index.php/How_to_use_LCD_driver).
I started from [170703](http://www.raspberrypiwiki.com/index.php/File:LCD-show-170703.tar.gz). I unzipped it in the home folder, and then I run LCD35-show. After installing it automatically reboots, and the 3.5 screen is still not working.
I repeated all actions with [151102-V2](http://www.raspberrypiwiki.com/index.php/File:LCD-show-151102-V2.zip) and [160715](http://www.raspberrypiwiki.com/index.php/File:LCD-show-160715.zip), but nothing changed

Fourdee, 2018-03-10:
**LCD35-show:**
_I've seen some "bad" bash scripts in my time, but this is pure dedication lol._
- ```/boot/config.txt``` is completely overwritten, this may cause issues with expected scrapes from DietPi if the file lacks default RPi items.

Ok, I believe the solution here is to copy /boot/config.txt to DietPi-RAMdisk after running ```./LCD35-show``` eg:
- Install driver ```./LCD35-show```
- ```cp /boot/config.txt /DietPi/config.txt```
- reboot system

---

```
i=1
for lines in `cat ./boot/mark`
do
  case ${i} in
  1) var1=${lines};;
  esac
  i=`expr ${i} + 1`
done
#echo "$var1"

if test "$var1" = "0";then
#sudo mkdir /etc/X11/xorg.conf.d
sudo apt-get install xserver-xorg-input-evdev
sudo cp -rf /usr/share/X11/xorg.conf.d/10-evdev.conf /usr/share/X11/xorg.conf.d/45-evdev.conf
echo "1" > ./boot/mark
fi

if test "$1" = "0" -o "$#" = "0";then
sudo cp -rf ./etc/X11/xorg.conf.d/99-calibration.conf-35  /usr/share/X11/xorg.conf.d/99-calibration.conf
sudo cp ./boot/config-35.txt /boot/config.txt
echo "LCD configrue 0"
elif test "$1" = "90";then
sudo cp -rf ./etc/X11/xorg.conf.d/99-calibration.conf-35-90  /usr/share/X11/xorg.conf.d/99-calibration.conf
sudo cp ./boot/config-35.txt-90 /boot/config.txt
echo "LCD configrue 90"
elif test "$1" = "180";then
sudo cp -rf ./etc/X11/xorg.conf.d/99-calibration.conf-35-180  /usr/share/X11/xorg.conf.d/99-calibration.conf
sudo cp ./boot/config-35.txt-180 /boot/config.txt
echo "LCD configrue 180"
elif test "$1" = "270"; then
sudo cp -rf ./etc/X11/xorg.conf.d/99-calibration.conf-35-270  /usr/share/X11/xorg.conf.d/99-calibration.conf
sudo cp ./boot/config-35.txt-270 /boot/config.txt
echo "LCD configrue 270"
fi

sudo cp ./waveshare35a-overlay.dtb /boot/overlays/waveshare35a.dtbo
sudo cp ./waveshare35a-overlay.dtb /boot/overlays/
sudo cp -rf ./usr/share/X11/xorg.conf.d/99-fbturbo.conf  /usr/share/X11/xorg.conf.d/
if [ -b /dev/mmcblk0p7 ]; then
sudo cp ./cmdline.txt-noobs /boot/cmdline.txt
else
sudo cp ./cmdline.txt /boot/
fi

sudo cp ./inittab /etc/

if test "$#" = "0" -o "$1" = "0" -o "$1" = "90"  -o "$1" = "180" -o "$1" = "270"; then
sudo reboot
echo "reboot now"
else
echo "Invalid parameter,Usage:LCD35-show [0] [90] [180] [270]"
fi
```

ccc0der, 2018-03-10:
@Fourdee 
I tried copying after installing both 151102-V2 and 170703 virsions, but it still not working

closed 2018-04-13 by Fourdee

SteveJohnsson, 2019-02-24:
Hi guys!

I'm not able to get my Chinese TFT 3,5" LCD to work on DietPi. It works fine with Raspbian and this driver (https://github.com/goodtft/LCD-show). 

I did everything described in this issue, but the screen is still white.