MichaIng/DietPi #1599: Wifi Hotspot / Tor Hotspot

symbios24, 2018-03-09:
Hello i installed both Wifi Hotspot / Tor Hotspot in latest dietpi for RaspberryPi3 if i want o use only the WiFI Hotspot without the Tor how can i do it without uninstalling the Tor?

Thanks

Fourdee, 2018-03-09:
@symbios24 

Its a little complicated, due to the iptables that are configured for Tor, during installation.

Currently, to switch, you need to remove Tor, which also removes WiFi hotspot as it needs to flush everything. Then reinstall WiFi Hotspot.

SuBLiNeR, 2018-03-09:
@symbios24 as @Fourdee answered, it depends upon the iptaples config. 

for TOR:
     sudo iptables -t nat -A PREROUTING -i wlan0 -p tcp --dport 22 -j REDIRECT --to-ports 22
     sudo iptables -t nat -A PREROUTING -i wlan0 -p udp --dport 53 -j REDIRECT --to-ports 53
     sudo iptables -t nat -A PREROUTING -i wlan0 -p tcp --syn -j REDIRECT --to-ports 9040

for only Hotspot:
    sudo sysctl -w net.ipv4.ip_forward=1 && sudo sed -i "s/#net.ipv4.ip_forward=1/net.ipv4.ip_forward=1/g" /etc/sysctl.conf && sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE && sudo iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE && sudo bash -c "iptables-save > /etc/iptables.rules"

that's like i was formely answering on to your question.... ;)

symbios24, 2018-03-10:
Hi, thanks for the response it will be nice to add an option inside the dietpi-config to switch between normal hotspot and tor

symbios24, 2018-03-11:
Thank you but with my myopia will take me half a hour to write this every time, if you can add this option in the config sometime in the feature will be nice thanks

garywill, 2018-11-04:
This script is used for creating wifi hotspot: [garywill/linux-router](https://github.com/garywill/linux-router)
It supports normal hotspot and transparent proxy.
```
# lnxrouter --ap wlan0 SSID --password PASSWD
```
will create normal hotspot. 
To switch to tor hotspot, `control-c` to stop it, then
```
# lnxrouter --ap wlan0 SSID --password PASSWD --tp 9049 --dns-proxy 9053
```
That script handles ip, hostapd, dhcp&dns (dnsmasq). So hotspot is completely handled by it.
It should work on all Linux. Hope this helps.

MichaIng, 2018-11-04:
@garywill 
Thanks for sharing your great work, I was thinking if we could implement this into DietPi, replacing our own Hotspot and Tor installs. We could build a wrapper GUI for it into DietPi-Config and use it during install as well. Better we help develop your great code then double effort on our own.
- Dependencies match, `dnsmasq` is there additionally, but it totally makes sense to have the Hotspot serving DNS as well.
- As far as I could see, it does not break other DietPi feature, nor does DietPi break it's features, when our related snippets are replaced accordingly.
- Licences match (?), however we would not do any modifications, instead contributing to your project, in case. Source code link in our README.md and your NOTICE prompt on install is self-evident.
- Would you agree?

But one question: Why does the script need to stay running? See no reason why exiting (after everything else finished) would break anything? hostapd, dnsmasq and dhcp server should keep everything running based on the config that the script did?

garywill, 2018-11-04:
@MichaIng Thank you. 
I forked that script and modified for my own use, the upstream project is oblique's create_ap.
License of create_ap is BSD 2-Clause, I forked it and license my script LGPL, both are FOSS. It's surely OK to link to/use/fork my script, as long as keeping necessary NOTICE.

I keep the script running, so when the script receive SIGINT, SIGUSR1 etc. signals, it restore iptables , terminates `dnsmasq` and other stuff. (I usually run it on PC in terminal emulater and use `control-c` to quickly clear all changes).
There is `--daemon ` option to put it to background (script still running). 
Use `--stop <id>` to stop it. (This is buggy now, may not work)