MichaIng/DietPi #1600: DietPi-Software | New version of UrBackup server available (2.2.7)

LexiconCode, 2018-03-09:
## Creating a bug report/issue:
There is a new version of UrBackup server available (2.2.7). Download it [here](http://www.urbackup.org/download.html).

#### Required Information:
- DietPi Version 63
- SBC Device (EG: C2)
- Distro (Jessie) 

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

LexiconCode, 2018-03-09:
How would you recommend upgrading UrBackup?

Fourdee, 2018-03-09:
@LexiconCode 

Thanks for the heads up 👍 

Updates are handled by us, via patched re-installs.

I'll update the install versions and run a patch for v6.4

Fourdee, 2018-03-09:
@LexiconCode 

Completed, 2.2.8 will be installed during patch: https://github.com/Fourdee/DietPi/commit/0fd3bf315f2f0c538027a39b22c19d7d95d33159

v6.4 is planned for release today.

closed 2018-03-09 by Fourdee