MichaIng/DietPi #1602: DietPi-Set_dphys-swapfile | Failure with option 1

MichaIng, 2018-03-09:
#### Required Information:
- DietPi Version | v6.3
- SBC Device     | RPi 2 Model B (armv7l) (index=2)
- Distro         | buster (index=5)
- Command        | dphys-swapfile setup
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-dphys-swapfile

#### Expected behaviour:
Swapfile should be renewed

#### Actual behaviour:
Program fails on `dphys-swapfile setup`, as swapfile got not successfully removed before.

#### Steps to reproduce:
`/DietPi/dietpi/func/dietpi-set_dphys-swapfile 1`

#### Additional logs:
```
 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Generating new swapfile

[ INFO ] DietPi-dphys-swapfile | Size     = 1072 MB
[ INFO ] DietPi-dphys-swapfile | Location = /mnt/sda/dietpi_userdata/.swapfile
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/sda/dietpi_userdata/.swapfile)

[FAILED] DietPi-dphys-swapfile | dphys-swapfile setup
[ INFO ] DietPi-dphys-swapfile | exit_code = 1
[ INFO ] DietPi-dphys-swapfile | VERSION:v6.4 | HW_MODEL:2 | HW_ARCH:2 | DISTRO:5
[ INFO ] DietPi-dphys-swapfile | Log file contents:
want /mnt/sda/dietpi_userdata/.swapfile=1072MByte, checking existing
...
2018-03-09 21:03:16 root@micha:~# rm $(cat /etc/dphys-swapfile | grep -m1 '^CONF_SWAPFILE=' | sed 's/.*=//')
2018-03-09 21:03:45 root@micha:~# l /mnt/sda/dietpi_userdata/.swapfile
ls: cannot access '/mnt/sda/dietpi_userdata/.swapfile': No such file or directory
2018-03-09 21:03:50 root@micha:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 1
[  OK  ] Root access verified.

 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Generating new swapfile

[ INFO ] DietPi-dphys-swapfile | Size     = 1072 MB
[ INFO ] DietPi-dphys-swapfile | Location = /mnt/sda/dietpi_userdata/.swapfile
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/sda/dietpi_userdata/.swapfile)

[  OK  ] DietPi-dphys-swapfile | dphys-swapfile setup
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapon
[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 1024M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=1024M tmpfs /tmp
```
- As can be seen removing the swapfile manually works great, afterwards command runs nice. I even did some `bash -x` run and found the exact same rm command.
- Adding some G_RUN_CMD rm ... leads to the following:
```
 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Generating new swapfile

[ INFO ] DietPi-dphys-swapfile | Size     = 1072 MB
[ INFO ] DietPi-dphys-swapfile | Location = /mnt/sda/dietpi_userdata/.swapfile
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/sda/dietpi_userdata/.swapfile)

[FAILED] DietPi-dphys-swapfile | rm /mnt/sda/dietpi_userdata/.swapfile
[ INFO ] DietPi-dphys-swapfile | exit_code = 1
[ INFO ] DietPi-dphys-swapfile | VERSION:v6.4 | HW_MODEL:2 | HW_ARCH:2 | DISTRO:5
[ INFO ] DietPi-dphys-swapfile | Log file contents:
rm: cannot remove '/mnt/sda/dietpi_userdata/.swapfile': Operation not permitted
```
- Don't get it, as command of course runs with root permissions.
....
Found the issue:
- `dphys-swapfile swapoff` on my RPi2 does not really disable swapfile. `htop` still shows it in use.
- `swapoff -a` on the other hand works, as well as `dphys-swapfile uninstall`. To stay with `dphys-swapfile` commands (which seem just wrappers for the default swapon/off from `mount` package I think `dphys-swapfile uninstall` needs to be added before `dphys-swapfile setup` / before swapfile removal / as swap file removal.

MichaIng, 2018-03-09:
On VM the command does not fail, but the file is not removed as well:
```
 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Generating new swapfile

[ INFO ] DietPi-dphys-swapfile | Size     = 1059 MB
[ INFO ] DietPi-dphys-swapfile | Location = /mnt/dietpi_userdata/.swapfile
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/dietpi_userdata/.swapfile)

[  OK  ] DietPi-dphys-swapfile | dphys-swapfile setup
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapon
[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 1024M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=1024M tmpfs /tmp
root@VM-Stretch:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 1
[  OK  ] Root access verified.

 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Generating new swapfile

[ INFO ] DietPi-dphys-swapfile | Size     = 1059 MB
[ INFO ] DietPi-dphys-swapfile | Location = /mnt/dietpi_userdata/.swapfile
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/dietpi_userdata/.swapfile)

[  OK  ] DietPi-dphys-swapfile | dphys-swapfile setup
[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapon
[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 1024M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=1024M tmpfs /tmp
root@VM-Stretch:~# l /mnt/dietpi_userdata/.swapfile
-rw------- 1 root root 1110441984 Mar  9 23:15 /mnt/dietpi_userdata/.swapfile

 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Disable swapfile

[  OK  ] DietPi-dphys-swapfile | dphys-swapfile swapoff
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/mnt/dietpi_userdata/.swapfile)

[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 494M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=494M tmpfs /tmp
root@VM-Stretch:~# l /mnt/dietpi_userdata/
total 1084908
drwxrwxr-x 2 dietpi   dietpi         4096 Mar  1 22:49 downloads
-rw-r--r-- 1 root     root         476627 Mar  4 19:01 mariadb-database-backup.sql
drwxrwxr-x 2 dietpi   dietpi         4096 Mar  1 22:49 Music
drwxrwx--- 6 www-data www-data       4096 Feb 26 22:42 nextcloud_data
drwxrwxr-x 2 dietpi   dietpi         4096 Mar  1 22:49 Pictures
-rw------- 1 root     root     1110441984 Mar  9 23:15 .swapfile
drwxrwxr-x 2 dietpi   dietpi         4096 Mar  1 22:49 Video
root@VM-Stretch:~# l /mnt/dietpi_userdata/.swapfile
-rw------- 1 root root 1110441984 Mar  9 23:15 /mnt/dietpi_userdata/.swapfile
```

Maybe just on RPi, maybe just or in combination with swapfile on external drive, the command fails. But on every device removal of swapfile does not work. Somehow it does work if done manually. But within script `dphys-swapfile uninstall` seems to be necessary.

Fourdee, 2018-03-10:
@MichaIng 

Confirmed, good spot 👍 

Notes (i'll edit/add more during testing):
```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 20 /mnt/dietpi_userdata/.swapfile

Log file contents:
swapon: /mnt/63bb57bf-839e-4aaa-8f63-bacfa3349c78/dietpi_userdata/.swapfile:
swapon failed: Device or resource busy
```

raibtoffoletto, 2018-03-11:
Hey guys,
I was attempting to install DietPi and also got the same error.
I'm not sure if it is because I'm attempting to install it on a 2gb SD
(my 16gb died yeterday), I can try later in a bigger one to check if its
not just lackage of disk space since my disk is like this:

Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        1817488 1801088         0 100% /

==LOG==

#### Required Information:
- DietPi Version | v6.3
- SBC Device     | RPi Zero W (armv6l) (index=1)
- Distro         | stretch (index=4)
- Command        | dphys-swapfile setup
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-dphys-swapfile

#### Additional logs:
```
Log file contents:
want /var/swap=1567MByte, generating swapfile ...
```

Fourdee, 2018-03-11:
@raibtoffoletto 

> I'm not sure if it is because I'm attempting to install it on a 2gb SD

Hi,

DietPi requires a 4GB minimum SD card, to ensure a stable experience and sufficient swapfile based on total memory: http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

Previously, it was possible to set swapfile size to 0 in ```dietpi.txt```, however, due to previous issues with swapfile generation on some images, we had to forcefully recreate them for all devices in v6.4 patch, overwriting any previous setting.

Fourdee, 2018-03-11:
@raibtoffoletto 

Try running the following command to revert swapfile to 0MB:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 0
```

raibtoffoletto, 2018-03-11:
@Fourdee 
Thanks a lot! I'm so sorry, I didn't see the 4gb requirement on the instructions.
(I was looking at an out-of-date guide somewhere and I didn't check it).

Anyway, I managed by copying var/ to an external usb and edit it to mount via /etc/fstab before first boot.
The install went without problems.

The problem was that with the swap error, the installation breaks and I don't have access anymore do 
DietPi scripts... couldn't figure it out why. Anyway, I was attempting something out of the min specs....

Fourdee, 2018-03-12:
@MichaIng 

Resolved:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_dphys-swapfile#L102-L124
- Removed dphys-swapfile
- Do it ourselves in script above
- ```faallocate``` for speedy swapfile generation. Downside is possible fragmentation, however, shouldnt affect solid state drives.

Tested:
- 🈯️ 1 /var/swap > 1 /mnt/dietpi_userdata/.swap
- 🈯️ 1 /mnt/dietpi_userdata/.swap > 1 /var/swap
- 🈯️ 0 FP works fine

MichaIng, 2018-03-12:
@Fourdee 
Very good, I thought already the same, after some research, that the main benefit of `dphys-swapfile` seems to be the automatic swap size determination, which we anyway choose ourself. Thus should be possible to skip this package and use native swap file methods of `mount`.

This eliminates other ideas I had to install/uninstall `dphys-swapfile` on setting/removing swapfile, e.g. for VMs, where we want to avoid them anyway.

Fourdee, 2018-03-13:
@MichaIng 

> and add swap location to dietpi.txt instead? I would also like to see swap location within dietpi-config,

Yep to both 👍 

I also missed a few things in dietpi-software. I'll make the changes and send a commit.

Fourdee, 2018-03-14:
Completed.

closed 2018-03-14 by Fourdee