MichaIng/DietPi #1603: Kodi on Rock64

TheExpertNoob, 2018-03-10:
#### Required Information:
- DietPi Version? **v6.4**
  but reads:
  6
  4
- SBC Device? **Rock64 (4Gb)**
- Power supply used? **5v 3A**
- SD card used? **N/A eMMC (16Gb)**
- Distro? **Linux DietPi 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux**

#### Additional Information (if applicable):
- Software title? **Kodi**
- Can this issue be replicated on a fresh installation of DietPi?
  Yes, due to lack of GPU support

#### Expected behaviour:
Rock64 device should boot into KODI GUI/Desktop

#### Actual behaviour:
ERROR: Unable to create GUI. Exiting
*** Error in `/usr/lib/aarch64-linux-gnu/kodi/kodi.bin': double free or corruption (out): 0x000000556a92bf00 ***
Aborted

#### Steps to reproduce:
Install KODI
run KODI

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
I assume this is due to lack of GPU support for the Rock64 and not an aarch64 issue? How soon will GPU support be available from Pine64?(not a question you could answer but might have insight on) How does one build a custom kernel for diet pi and **TEST** it instead of implementing it and the need to reflash?

TheExpertNoob, 2018-03-10:
haha, should have read the CHANGELOG

Fourdee, 2018-03-10:
@TheExpertNoob 

Hi,

Thanks for the report.

GPU features are disabled for the Rock64 (and Kodi), we do not support it currently.

I'll edit ```dietpi-software``` to disable Kodi for this device.

TheExpertNoob, 2018-03-10:
>  Its already disabled, did you install Kodi via apt?
yes I did.

After the fact I read over
https://github.com/Fourdee/DietPi/issues/380
since its dated, I may look over the repositories mentioned and see if anything has been updated, and maybe help out in getting GPU working if modules build properly.

TheExpertNoob, 2018-03-20:
Placing link here for future referencing and experimentation.
https://github.com/ayufan-rock64/linux-build/blob/master/recipes/video-playback.md

closed 2018-07-09 by Fourdee