MichaIng/DietPi #1604: Desktop not working Pine64

Drew80, 2018-03-10:
## Creating a bug report/issue:
Desktop Environment not loading on Pine 64 A+

#### Required Information:
- **DietPi Version |** v6.4
- **SBC Device |** Pine64 A+ 1GB (Kickstarter edition)
- **Power supply used |** 5v 2.5A bought from Pine.org
- **SD card used |** Adata 16gb class 10 
- **Distro |** Linux DietPi 3.10.104-2-pine64-longsleep # 113 SMP PREEMPT Thu Dec 15 21:46:07 CET 2016 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title - LXDE or MATE desktop installation
- Can this issue be replicated on a fresh installation of DietPi - Yes I have tried multiple times.  I have successfully installed other DE using images from the Pine.org website however I find support isn't great over there.  Also I have had success with DietPi in the past in and is my preferred OS of choice whenever possible

#### Expected behaviour:
LXDE or MATE should install and load on the 'start' command from the terminal

#### Actual behaviour:
Nothing.  Desktop starts to load then I get an error

original error......
EE - fbturbo: module ABI major version (18) doesn't match the server's version (23)
EE - Server terminated with error (1). Closing log file.log" for additional information
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

*** noticed this and preformed "sudo apt-get update" now errors reads as follows

new error......
EE - Server terminated with error (1). Closing log file.log" for additional information
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

#### Steps to reproduce:
Fresh install of DietPi
Choose LXDE or MATE as the desktop environment

#### Did you submit a ```dietpi-bugreport```?
Reference Code 459f33f4-ac12-4910-a589-eb589a0c1889-0

#### Extra details:
From what I have been reading and testing the desktop should be able to run and run without troubles.  I have had nothing but troubles getting it going.

Multiple fresh installs tried.

Installed software modules
5 - ALSA: linux sound system
6 - Xserver: linux display system --> this is what most of the errors are pointing to
24 - MATE: desktop environment (Also tried with LXDE, same issue.)
103 - DietPi-Ramlog: minimal, optimized logging
104 - Dropbear: ssh server
152 - Avahi-Daemon: hostname broadcast (I prefer to login with HOSTNAME.local on my network)

Note: The error I received with LXDE was similar but I do not think it was the exact same. Willing to test both or either to help further DietPi OS

Thanks

- Drew

Fourdee, 2018-03-10:
@Drew80 

Thanks for the report 👍 , looks like fbturbo driver being outdated against Xorg. We'll take a look.

Fourdee, 2018-03-10:
@Drew80 

Binaries updated for Stretch, please run the following command to reinstall Xserver + fbturbo drivers:
```
dietpi-software reinstall 6
reboot

#startx
```

Please let us know if this resolves.

Drew80, 2018-03-10:
@Fourdee 

YES! That did it.  I don’t have time this morning to test it out with LXDE, but I will have time to double check it this weekend on a fresh install.

Thank you so much Finally can move forward with my magic mirror project on the Pine64 

Which would be nice as an install option / feature request. 

I found this on the MagicMirror.builders website  So I’m going to be giving it a try

https://github.com/stacywebb/magicmirror_arm64

Thanks for the fast response time too. Another reason the DietPi community rocks!

Drew80, 2018-03-10:
@Fourdee 
I did a fresh install on the same brand of SD card (ADATA 16GB) and installed LXDE expecting the same issue however it installed fine this time.  I was unable to reproduce the error I was getting with LXDE the first time but I can confirm that both LXDE & MATE have been installed and working on the Pine64 A+ 1G board with DietPi v6.4  Hopefully the bug will get it's self worked out but at least we have a working solution that did not take too long to correct.

```
dietpi-software reinstall 6
reboot

#startx
```

Thanks for the help

- Drew

Fourdee, 2018-03-11:
@Drew80 

> however it installed fine this time. I was unable to reproduce the error

Hi Drew,

Yep, this is because we updated the binaries on dietpi.com which get downloaded during the install of v6.4.

Thanks for testing this and confirming resolved 👍 . I'll mark this as completed.

closed 2018-03-11 by Fourdee

darren-ang, 2020-06-13:
**DietPi Version |** v6.30.0
**SBC Device |** Pine64PINE A64 2GB rev 2
This seems like old issue but I encounter same same issue(but diff version ) today

Error as below
[    62.884] (II) Module fbturbo: vendor="X.Org Foundation"
[    62.884]    compiled for 1.19.2, module version = 0.5.1
[    62.884]    Module class: X.Org Video Driver
[    62.884]    ABI class: X.Org Video Driver, version 23.0
[    62.884] (EE) fbturbo: module ABI major version (23) doesn't match the server's version (24)
[    62.884] (EE) Failed to load module "fbturbo" (module requirement mismatch, 0)
[    62.884] (EE) No drivers available.
[    62.884] (EE)
Fatal server error:
[    62.884] (EE) no screens found(EE)
[    62.884] (EE)

Joulinar, 2020-06-13:
Which user you are going to use to start the desktop?

darren-ang, 2020-06-13:
thanks for reply

I use root, 

This issue happens on LXDE

XFCE4 works without issue

Joulinar, 2020-06-13:
looks like incorrect driver is loaded. 

`[ 62.884] (EE) fbturbo: module ABI major version (23) doesn't match the server's version (24)`

@MichaIng 
 can you have a look pls?

MichaIng, 2020-06-13:
Very strange that it is limited to one desktop. Indeed we need to binaries for Buster, I guess.

@darren-ang 
Just to be sure, you use a Debian Buster image, right? `echo $G_DISTRO_NAME`

darren-ang, 2020-06-13:
@MichaIng 

`root@DietPi:~# echo $G_DISTRO_NAME`
`buster`

MichaIng, 2020-06-14:
Okay I read a bit into the topic in general and compiled a new `libump` and `xf86-video-fbturbo` for Debian Buster. Please try the following:
```
rm /usr/lib/libUMP* # Remove the files we moved manually there
cd /tmp
wget https://dietpi.com/downloads/binaries/buster/libump_aarch64.deb
dpkg -i libump_aarch64.deb
rm libump_aarch64.deb
wget https://dietpi.com/downloads/binaries/buster/xf86-video-fbturbo_aarch64.deb
dpkg -i xf86-video-fbturbo_aarch64.deb
rm xf86-video-fbturbo_aarch64.deb
ldconfig
startx
```

darren-ang, 2020-06-14:
Hi Michalng,

Thanks , your solution works!!

Output of the execution as below
root@DietPi:~# rm /usr/lib/libUMP*
root@DietPi:~# cd /tmp
root@DietPi:/tmp# wget
https://dietpi.com/downloads/binaries/buster/libump_aarch64.deb
--2020-06-14 11:47:47--
https://dietpi.com/downloads/binaries/buster/libump_aarch64.deb
Resolving dietpi.com (dietpi.com)... 172.67.173.4, 104.27.179.199,
104.27.178.199, ...
Connecting to dietpi.com (dietpi.com)|172.67.173.4|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9612 (9.4K) [application/x-debian-package]
Saving to: ‘libump_aarch64.deb’

libump_aarch64.deb  100%[===================>]   9.39K  --.-KB/s    in
0.001s

2020-06-14 11:47:48 (8.19 MB/s) - ‘libump_aarch64.deb’ saved [9612/9612]

root@DietPi:/tmp# dpkg -i libump_aarch64.deb
(Reading database ... 59686 files and directories currently installed.)
Preparing to unpack libump_aarch64.deb ...
Unpacking libump (3.0-dietpi1) over (3.0-1) ...
dpkg: warning: unable to delete old directory '/usr/local/lib': Directory
not empty
dpkg: warning: unable to delete old directory '/usr/local/include':
Directory not empty
dpkg: warning: unable to delete old directory '/usr/local': Directory not
empty
Setting up libump (3.0-dietpi1) ...
root@DietPi:/tmp# rm libump_aarch64.deb
root@DietPi:/tmp# wget
https://dietpi.com/downloads/binaries/buster/xf86-video-fbturbo_aarch64.deb
--2020-06-14 11:48:25--
https://dietpi.com/downloads/binaries/buster/xf86-video-fbturbo_aarch64.deb
Resolving dietpi.com (dietpi.com)... 104.27.178.199, 172.67.173.4,
104.27.179.199, ...
Connecting to dietpi.com (dietpi.com)|104.27.178.199|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27800 (27K) [application/x-debian-package]
Saving to: ‘xf86-video-fbturbo_aarch64.deb’

xf86-video-fbturbo_ 100%[===================>]  27.15K  --.-KB/s    in
0.004s

2020-06-14 11:48:26 (5.96 MB/s) - ‘xf86-video-fbturbo_aarch64.deb’ saved
[27800/27800]

root@DietPi:/tmp# dpkg -i xf86-video-fbturbo_aarch64.deb
dpkg: warning: downgrading xf86-video-fbturbo from 1.0-1 to 0.5.1-dietpi1
(Reading database ... 59675 files and directories currently installed.)
Preparing to unpack xf86-video-fbturbo_aarch64.deb ...
Unpacking xf86-video-fbturbo (0.5.1-dietpi1) over (1.0-1) ...
Setting up xf86-video-fbturbo (0.5.1-dietpi1) ...
root@DietPi:/tmp# rm xf86-video-fbturbo_aarch64.deb
root@DietPi:/tmp# ldconfig
root@DietPi:/tmp# startx

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.0-8-arm64 aarch64 Debian
Current Operating System: Linux DietPi 5.4.43-sunxi64 #20.05.2 SMP Tue Jun
2 17:20:17 CEST 2020 aarch64
Kernel command line: root=UUID=05f23878-fded-416f-821e-44cb78f788a8
rootwait rootfstype=ext4  consoleblank=0 loglevel=7 ubootpart=898c1213-01
usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=memory
swapaccount=1
Build Date: 05 March 2019  08:11:12PM
xorg-server 2:1.20.4-1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
Before reporting problems, check http://wiki.x.org
to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
(++) from command line, (!!) notice, (II) informational,
(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sun Jun 14 11:49:36 2020
(==) Using config file: "/etc/X11/xorg.conf"
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module mali not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module mali_drm not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module sunxi_cedar_mod not found in directory
/lib/modules/5.4.43-sunxi64
SunxiMaliDRI2_Init: drmOpen failed!

MichaIng, 2020-06-14:
Great to hear. I'll update `dietpi-software` accordingly.
```
modprobe: FATAL: Module g2d_23 not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module mali not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module mali_drm not found in directory
/lib/modules/5.4.43-sunxi64
modprobe: FATAL: Module sunxi_cedar_mod not found in directory
/lib/modules/5.4.43-sunxi64
SunxiMaliDRI2_Init: drmOpen failed!
```
Expected kernel modules were not found. The failing part of the code is here: https://github.com/ssvb/xf86-video-fbturbo/blob/f9a6ed78419f0b98cf2c3ce3cdd4c97fe9a46195/src/sunxi_mali_ump_dri2.c#L1086-L1113

Not sure whether 2D acceleration is at least active partly 🤔.

I checked the kernel config and it seems that the mali hardware info is added via device tree, so I guess is expected to be not available as module.

Found the same set or errors here: https://github.com/ssvb/xf86-video-fbturbo/issues/10#issuecomment-24293992
Statement on the errors:
> The driver works on Rockchip devices (tested on RK3188). Naturally the advantage is only what regards to usage of ARM NEON instructions, since the Allwinner 2D accelerator is not present on RK CPUs.

I guess it is only available for A10, A13 and A20 but not for A64.