MichaIng/DietPi #1605: Default Nextcloud install doesn't use configured userdata

JonasDralle, 2018-03-10:
#### Required Information:
- DietPi Version | **6.2**
- SBC Device | **Raspberry Pi 3b**
- Power supply used | **direct to wall socket**
- SD card used | **SanDisk 16 GB**
- Distro | **Linux cloud 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux**

#### Additional Information (if applicable):
- **NextCloud 13** on **Lighthttpd**
- Can this issue be replicated on a fresh installation of DietPi?  
**Yes and I will add a step by step guide subsequently**

#### Expected behaviour:
When external hard drive is configured as userdata location the nextcloud data should be stored there.

[image]

#### Actual behaviour:
The folder ```/mnt/***/dietpi_userdata/nextcloud_data``` exists but looks like this:
[image]

user "jonas" is missing. Instead data is stored in ```/var/www/nextcloud/data```
[image]

which in turn results maxing out the SD card and thus internal server errors on write:
[image]

#### Steps to reproduce:
[ i will add this later, sorry for the inconvenience ]

#### Did you submit a ```dietpi-bugreport```?
fc8c991a-a683-4a7d-9a0a-a6057b31fd02-0

#### Extra details:

JonasDralle, 2018-03-10:
⚠️ When I first installed NextCloud on DietPi 6.2 I had trouble connecting to the MariaDB. [This YouTube video shows the same misconfig at about 1:50](https://www.youtube.com/watch?v=O8h_Dlf2MOA).
I just tried with 6.4 again and NextCloud was set up correctly (including admin account with password "dietpi".

### Cannot reproduce error!

~

What I did:

(1) Fresh SD Card flash with following image: **DietPi_v6.2_RPi-ARMv6-Stretch.img**
SHA1: DAD2740F0F8081439394C6F78153A73EBFA9B16F
[image]

(2) Config Diet Pi
(a) External Drive formatted <sub>[img](https://i.imgur.com/uXYKegB.png)</sub>
(b) External Drive is for user data <sub>[img](https://i.imgur.com/mHpkUad.png)</sub>
(c) install NextCloud with Lighthttpd
[image]

(3) Going to NextCloud and uploading a file to check file upload location (```http://dietpi/nextcloud/```)

[image]

The file is saved in the correct location and thus invalidating my ticket

[image]

Thanks for keeping DietPi awesome! With DietPi I was able to setup a fully functioning NextCloud Server in less than a hour.

closed 2018-03-10 by JonasDralle

MichaIng, 2018-03-11:
@JonasDralle 
Thanks for your report and positive feedback after all 👍.

The issue you faced first occurs if something went wrong with the Nextcloud command line installation: https://docs.nextcloud.com/server/13/admin_manual/configuration_server/occ_command.html#command-line-installation-label
This includes database and database user creation and setting up data folder. But if somehow on MariaDB installation/setup/startup something went wrong, then `occ` command fails and you see the web ui installation page from the video with default username `admin` and default data location `/var/www/nextcloud/data`. So then a careful look into MariaDB status/installation/logs could help to find the issues root.