MichaIng/DietPi #1607: External drive containing userdata isn't automounted

phdelodder, 2018-03-10:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | **6.4**
- SBC Device | **Rock64**
- Power supply used | **Original Wall Socket**
- SD card used | **SanDisk 16GB**
- Distro (EG: Jessie) | **Linux copperhead 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux**

#### Additional Information (if applicable):
- Software title? **dietpi-drive_manager**
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

#### Expected behaviour:
The dietpi_userdata should automaticaly be mapped

#### Actual behaviour:
the external drive which contains the dietpi_userdata isn't mounted resulting docker containers not starting. 

#### Steps to reproduce:
Move the userdata using the dietpi-drive_manager and reboot. 

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
Dmesg extract:
`Failed to insert module 'autofs4': No such file or directory`
`Starting of mnt-342eea89\x2ddf7f\x2d4af7\x2d813f\x2de97ccb446ada.automount not supported.`

Fstab line:

`UUID=342eea89-df7f-4af7-813f-e97ccb446ada    /mnt/342eea89-df7f-4af7-813f-e97ccb446ada    auto    defaults,noatime,x-systemd.automount,nofail    0 0`

phdelodder, 2018-03-10:
Rebooting a few times solved it ...

Fourdee, 2018-03-10:
Due to ```CONFIG_AUTOFS4_FS=n``` in kernel config, with the pre-image/kernel we use: https://github.com/ayufan-rock64/linux-build/releases/tag/0.5.15

Which is out of our control unfortunately.

### Workaround is to remove the automount option from fstab:
@phdelodder 

```
sed -i 's/,x-systemd.automount//g' /etc/fstab
```

Fourdee, 2018-03-10:
I'll see if we can add a workaround for Rock64 in source.

Fourdee, 2018-03-10:
#### Resolved for v6.5:
- DietPi-Drive_Manager | Resolved an issue where x-systemd.automount would fail if autofs4 was disabled in kernel/modules (eg: Rock64). x-systemd.automount is now disabled for systems which fail autofs4 detection: https://github.com/Fourdee/DietPi/issues/1607
- Fix applied during patch.

I'll mark this as closed.

@phdelodder 
Please let us know if problems persist after running the workaround:
https://github.com/Fourdee/DietPi/issues/1607#issuecomment-372030565

closed 2018-03-10 by Fourdee

phdelodder, 2018-03-10:
@Fourdee the workaround fixed the issue.