MichaIng/DietPi #1608: When installing a single webclient offer the option to server content on root directory

JonasDralle, 2018-03-10:
When installing a `$webthing` dietpi serves this service via `/$webthing` (for example `/wordpress` when installing Wordpress), leaving `/` as the default webserver post-install page.
That's all fine when installing multiple optimized software packages but odd when installing only a single one.

I'd like to have an option **"serve `$webthing` at root"** in the dietpi setup.
Bitnami has a `bnconfig --appurl /` for this <sub>[bitnami doc](https://docs.bitnami.com/installer/components/bnconfig/)</sub>.

MichaIng, 2018-03-10:
@JonasDralle 
Thank you for your feature request. I already thought the same when another user wanted to have Nextcloud available on web root. This totally makes sense if you do not plan to install other web services besides. Currently if user configures its webserver manually for this, it will break further web service installations by dietpi-software.

It might be thinkable to implement a variable web root within DietPi instead of static /var/www. Then add option to change it via dietpi.txt or better automatically to chosen web service. But for practical reasons I would leave all installations of web services as they are. Just save a flag about changed web root and inform users on another installation attempt that web root needs to be reverted with offer to do that automatically.

MichaIng, 2018-04-01:
Note:
- Easiest would be perhaps to use mod_alias
  - Lighttpd:
```
server.modules += ( "mod_alias" )
alias.url = ( "/" => "/var/www/nextcloud/" )
```
  - Apache:
`Alias / "/var/www/nextcloud"`
  - Nginx: http://nginx.org/en/docs/http/ngx_http_core_module.html#alias
```
location / {
    alias /var/www/nextcloud/;
}
```

But it needs adjustments of webserver location config and in case web service config as well, e.g. ownCloud/Nextcloud `config.php`. Not too fast to implement...

FredericGuilbault, 2019-06-30:
On big servers, each website have it's own user and webroot usualy stored in the `/home/[USER]/public_html`

`/var/www` for apache and `/var/www/html` for nginx are default linux locations. Then yes, if you have more then one web service, plan to have more then one or even don't know if there will be an other one in the future. And plan to run it under the same user you should def have seperated folder under `/var/www` for it. 
To have nice URLs  AFAIK  `mod_rewrite` is the way to go.

MichaIng, 2019-06-30:
@FredericGuilbault 
Even on Apache the default is `/var/www/html`. Regardless we change all to `/var/www` to be consistent and I don't see much reason for the html subdir as webroot.

All webserver related software have reasonable/meaningful subdirs, so no rewrite required then. However the question is which method is the easiest to change or redirect or rewrite or proxy the webroot to some websites sub directory, so that when installing another website, this can be easily reverted.

A simple redirect would be IMO easiest solution. But the URL then will change from plain root to related website subdir. But this would not require any other change.

Any solution that would preserve the URL so that the subdir is not contained, requires configuration changes for many web applications, as the URI is often used to verify/authenticate. E.g. ownCloud and Nextcloud have the `trusted_domains` array that needs to contain every base URI that you want to access with. Also they have the `overwrite.cli.url` which needs change then, as if `http://localhost/nextcloud` is rewritten to `http://localhost/nextcloud/nextcloud`, this would break things. Or we need to actively need to exclude these cases, which is actually not too hard. However many other such things for different web applications need to be considered. So this would make things much more complicated.