MichaIng/DietPi #1613: Squeezebox not running!

hifitime, 2018-03-13:
dietpi 6.4
Squeezebox  not running
==================================

root@DietPi:~# squeezeboxserver --d
# **/usr/bin/perl: symbol lookup error: /usr/share/squeezeboxserver/CPAN/auto/EV/EV.so: undefined symbol: Perl_xs_apiversion_bootcheck**
=======================================

root@DietPi:~# service logitechmediaserver start
Failed to start logitechmediaserver.service: Unit logitechmediaserver.service not found.

The errors that are always displayed are“/usr/bin/perl:symbol lookup error:/usr/share/squeezeboxserver/cpan/auto/ev/ev.so:undefined symbol:perl_ Xs_apiversion_bootcheck”

======================================
root@DietPi:~# perl -v

This is perl 5, version 24, subversion 1 (v5.24.1) built for aarch64-linux-gnu-thread-multi
(with 75 registered patches, see perl -V for more detail)

Copyright 1987-2017, Larry Wall

Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.

Fourdee, 2018-03-13:
@hifitime 

Hi,

Thanks for the report.

Which device are you running this on?

Fourdee, 2018-03-13:
notes:
```
#C2
dietpi-software install 35
2018-03-13 19:18:57 (3.46 MB/s) - ‘/mnt/dietpi_userdata/Music/Absolute-Radio-00s.pls’ saved [127/127]

[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Software | DietPi-Software installation completed.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[  OK  ] DietPi-Services | start : SqueezeBox
[ SUB2 ] DietPi-Process_tool > Apply
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20568) : Nice      0
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20568) : Affinity  0-3
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20568) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20586) : Nice      0
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20586) : Affinity  0-3
[FAILED] DietPi-Process_tool | Squeezebox Server (LMS) (20586) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process Tool | An issue has occured
```

hifitime, 2018-03-14:
hi  Fourdee

tks for you reply

My device is not in the existing support list

I compiled it through prep_system_for_dietpi.sh.

Marvell ARMADA 3700LP (88F3720) ARM Cortex A53

hifitime, 2018-03-14:
I tried to remove squeezebox through the control panel and then through sudo dpkg-i logitechmediaserver_7.9.1~1520781386_arm.deb Then, it works fine.

hifitime, 2018-03-15:
I tried to remove squeezebox through the control panel and then through
dpkg -i logitechmediaserver_7.9.1~1521043527_arm.deb

Then, it works fine.

hifitime, 2018-03-15:
May be a problem with the Logitechmediaserver_7.9.1~15xxxxx_all.deb installation package on the dietpi.com server?

hifitime, 2018-03-15:
maby  the problem is  http://dietpi.com/downloads/binaries/all/DietPi-LMS7.9-CPAN_arm64.zip

Fourdee, 2018-03-15:
@hifitime 

Confirmed issue is limited to ARMv8 under Debian Stretch:
```
Mar 15 16:11:44 DietPi squeezeboxserver[22584]: /usr/bin/perl: symbol lookup error: /usr/share/squeezeboxserver/CPAN/auto/EV/EV.so: undefined symbol: Perl_xs_apiversion_bootcheck
```

Leave it with us, we'll try to resolve.

Fourdee, 2018-03-15:
@hifitime 

Will be fixed for v6.5.

Please try the following to resolve now (copy and paste all):
```					INSTALL_URL_ADDRESS='http://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.1_CPAN-armv8-stretch.zip'
G_CHECK_URL "$INSTALL_URL_ADDRESS"
wget "$INSTALL_URL_ADDRESS" -O package.zip
unzip -o package.zip -d /usr/share/squeezeboxserver
rm package.zip
dietpi-services restart
```

Fourdee, 2018-03-15:
Resolved for v6.5:
- 🈯️ C2 + Jessie 
- 🈯️ C2 + Stretch

closed 2018-03-15 by Fourdee