MichaIng/DietPi #1614: MPD: Alsa device does not appear as DLNA endpoint player in the network

Knitas, 2018-03-13:
'Dietpí' or Alsa device does not appear as DLNA endpoint player in, for example, Bubble-up.  

- DietPi Version 6.2 -6.4
- RPi 3 **Model B** '(conscious choice because of its discrete regulators) 

Did several attempts wit fresh install, re-install. No error message.

#### Expected behaviour:
Alsa device name (mpd.conf) should appear in the DLNA players list.

Fourdee, 2018-03-13:
@Knitas 

Hi,

Many thanks for the report 👍 

Confirmed, currently our compiled MPD binary has UPnP disabled:
```
--disable-upnp
```

I'll need to re-compile binaries with UPnP enabled.

Fourdee, 2018-03-25:
```
root@DietPi:~# mpd -V
Music Player Daemon 0.20.18

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright 2008-2017 Max Kellermann <max.kellermann@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local curl

Neighbor plugins:
 upnp
```

- ARMv6
- 🈯️ ARMv7
- 🈯️ ARMv8
- 🈯️ x86_64

Fourdee, 2018-03-25:
Completed.

Patch reinstall for v6.5.

closed 2018-03-25 by Fourdee