MichaIng/DietPi #1617: Weird characters when typing special characters

grenzor, 2018-03-13:
When I try to type some special character it never types it but instead types some other random character. For example when `/` key is typed it types `-` and for other special character keys it types other stuff as shown in screenshot below. Makes it almost impossible to use terminal commands. Letters and numbers type out just fine.

Screenshot: https://i.imgur.com/Zl5uJGE.png

Using the latest DietPi Virtualbox image on Windows host.

MichaIng, 2018-03-13:
@grenzor 
Thanks for your report.

Jep, that was a mistake by me, leaving german keyboard layout which does not get overwritten by dietpi.txt default "gb" (UK layout) during 1st run setup. To fix you should be able to just switch keyboard layout from within `dietpi-config`.

This issue is fixed with the newest VirtualBox image, uploaded yesterday: http://dietpi.com/downloads/images/

grenzor, 2018-03-13:
Thanks, changed keyboard layout and all is fine now.

closed 2018-03-13 by grenzor