MichaIng/DietPi #1620: Shairport-sync not working on fresh install

ezekini, 2018-03-14:
#### Required Information:
- DietPi Version: 6.4
- SBC Device: ODROID C2
- Power supply used: Generic
- SD card used: eMMC by Hardkernel
- Distro: Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Expected behaviour:
process should start normally. 

#### Actual behaviour:
The process doesn't start apparently due to some problem with libraries libssl.

root@DietPi:~# systemctl status shairport-sync -l
● shairport-sync.service - Shairport Sync - AirPlay Audio Receiver
   Loaded: loaded (/lib/systemd/system/shairport-sync.service; disabled)
   Active: failed (Result: exit-code) since mar 2018-03-13 23:17:16 -03; 10h ago
  Process: 1684 ExecStart=/usr/local/bin/shairport-sync (code=exited, status=127)
 Main PID: 1684 (code=exited, status=127)

mar 13 23:17:16 DietPi systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
mar 13 23:17:16 DietPi shairport-sync[1684]: /usr/local/bin/shairport-sync: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
mar 13 23:17:16 DietPi systemd[1]: shairport-sync.service: main process exited, code=exited, status=127/n/a
mar 13 23:17:16 DietPi systemd[1]: Unit shairport-sync.service entered failed state.

#### Steps to reproduce:
Make fresh install of Dietpi on C2. Try to install shairport-sync

Fourdee, 2018-03-14:
@ezekini 

Thanks for the report 👍 

We'll try to replicate.

Fourdee, 2018-03-14:
Notes:
```libssl.so.1.1``` due to stretch compiles, Odroid C2 running Jessie.

Fourdee, 2018-03-14:
@ezekini 

Please run the following commands (copy paste all) to fix this for C2 now:
```
wget http://ftp.us.debian.org/debian/pool/main/o/openssl/libssl1.1_1.1.0f-3+deb9u1_arm64.deb -O package.deb; dpkg -i package.deb; rm package.deb
```

We'll also get this resolved for v6.5 update.

Fourdee, 2018-03-14:
Resolved for v6.5. 
Workaround for v6.4 + C2 now = https://github.com/Fourdee/DietPi/issues/1620#issuecomment-373088691

closed 2018-03-14 by Fourdee

ezekini, 2018-03-14:
You are fast! Thank you for the workaround. Long live diet-pi and happy pi day

amk221, 2018-07-26:
I am running a fresh install of 6.11, and then ran `dietpi-update` to 6.12
and getting:

```
root@KitchenStreamer:~# systemctl status shairport-sync -l
● shairport-sync.service - Shairport Sync - AirPlay Audio Receiver
   Loaded: loaded (/lib/systemd/system/shairport-sync.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-07-26 12:40:44 BST; 5min ago
  Process: 1254 ExecStart=/usr/local/bin/shairport-sync (code=exited, status=1/FAILURE)
 Main PID: 1254 (code=exited, status=1/FAILURE)

Jul 26 12:32:31 KitchenStreamer systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
Jul 26 12:32:31 KitchenStreamer shairport-sync[1254]: Successful Startup
Jul 26 12:40:44 KitchenStreamer shairport-sync[1254]: fatal error: audio_alsa: Sample format 7 not available for device "defa
ult": Invalid argument
Jul 26 12:40:44 KitchenStreamer systemd[1]: shairport-sync.service: Main process exited, code=exited, status=1/FAILUR
E
Jul 26 12:40:44 KitchenStreamer systemd[1]: shairport-sync.service: Unit entered failed state.
Jul 26 12:40:44 KitchenStreamer systemd[1]: shairport-sync.service: Failed with result 'exit-code'.
```

Fourdee, 2018-07-27:
@amk221 

Hi,

Thanks for the report 👍 

> fatal error: audio_alsa: Sample format 7 not available for device "defa
ult": Invalid argument

Please change the bitrate to 32bit:
- ```nano /usr/local/etc/shairport-sync.conf```
- Replace this line ```//  output_format = "S16";```
- With this ```output_format = "S32";```
- Save changes CTRL+X, Y, ENTER
- Restart services ```dietpi-services restart```

The reason for this, when we reinstall/update applications the default applied is 16bit for all systems.

amk221, 2018-08-07:
Actually, it already was S32. I've just tried a fresh installed of Allo GUI @ 6.11 (because that is all that is available currently).  And as soon as I chose to Airplay form iTunes, the service gets switched off.

Fourdee, 2018-08-08:
@amk221 

Strange, can we see the full contents of this file please?
```
cat /usr/local/etc/shairport-sync.conf
```

amk221, 2018-08-08:
```bash
general =
{
  name = "%H";
  interpolation = "soxr";
};

metadata =
{
	enabled = "yes";
	include_cover_art = "no";
	pipe_name = "/tmp/shairport-sync-metadata";
	pipe_timeout = 5000;
	socket_address = "226.0.0.1";
	socket_port = 5555;
	socket_msglength = 65000;
};

alsa =
{
//  mixer_control_name = "PCM";
output_rate = 44100;
output_format = "S16";
};
```
^ (Note, I changed it to S16, and it seemed to work)

```bash
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndallodigione [snd_allo_digione], device 0: Allo DigiOne HiFi wm8804-spdif-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```bash
aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=sndallodigione
    snd_allo_digione,
    Default Audio Device
dmix:CARD=sndallodigione,DEV=0
    snd_allo_digione,
    Direct sample mixing device
dsnoop:CARD=sndallodigione,DEV=0
    snd_allo_digione,
    Direct sample snooping device
hw:CARD=sndallodigione,DEV=0
    snd_allo_digione,
    Direct hardware device without any conversions
plughw:CARD=sndallodigione,DEV=0
    snd_allo_digione,
    Hardware device with all software conversions
```

Fourdee, 2018-08-08:
@amk221 

> output_format = "S16";

You should be able to get 24bit

Try setting/testing this to ```S24``` or ```S24_3LE```, restart services after reach try:
```
dietpi-services restart
```