MichaIng/DietPi #1621: Credits | Rework to separate DietPi code, base image and final image creator

MichaIng, 2018-03-14:
According to: http://dietpi.com/phpbb/viewtopic.php?f=9&t=2960&p=11407#p11407

@k-plan mentioned an important point about current credits appearance:
- http://dietpi.com/phpbb/viewtopic.php?f=9&t=2960#p11396
- http://dietpi.com/phpbb/viewtopic.php?f=9&t=2960#p11402

We should rework credits to reflect:
- Who is responsible for DietPi frontend (coding)
- Who is responsible for base image.
- Who created the final image by using base image above and DietPi-Prep.

@Fourdee 
What do you think?

Fourdee, 2018-03-14:
@MichaIng @k-plan 

> @k-plan mentioned an important point about current credits appearance:

Yep all good points. 👍 

> Who is responsible for DietPi frontend (coding)

Open to suggestions, @MichaIng how you like your name mentioned? 😃 

> Who is responsible for base image.
Who created the final image by using base image above and DietPi-Prep.

We could do a inputbox prompt during PREP. Requesting "Please enter your name", this could then be displayed in the DietPi banner (as opposed to current)?

k-plan, 2018-03-14:
@MichaIng 

Erst einmal Danke für die Beachtung und Verständnis des Themas.

Eigentlich wollte diese Diskussion von Anfang an aus dem Weg gehen.

Daher auch die damalige Ansage, `PREP_SYSTEM_FOR_DIETPI.sh` gehört nicht in End-User Hände.
Aber es wurde anders entschieden. Gut, damit könnte ich leben. Die Mehrzahl wird es eh' nicht hinbekommen. Sie sind einfach zu bequem oder wurde von uns zu sehr verwöhnt. Können nur konsumieren und Forderungen stellen. Bei so einer Grundhaltung bekomme ich schon Bluthochdruck.
Wer es hinbekommt, kann gerne im stillen Kämmerchen für sich selbst und den privaten Gebrauch damit ein Image basteln. Auf welcher Basis auch immer. Das entzieht sich unserm Einfluss und wäre somit auch völlig irrelevant. 

Was aber nach meinem Dafürhalten gar nicht geht, ist dass das Ergebnis dann nach Belieben im Internet verteilt werden darf und unter dem Name `DietPi`, `Created by : Daniel Knight` und ` Web: http://DietPi.com` erscheint.
Wo bitte landet dann der Support und wer leistet den bitte? Wir kommen jetzt schon auf keinen grünen Zweig.

War übrigens auch das gleiche und einzige Argument von Igor, welches ich gelten lassen würde in der gesamten unsäglichen Diskussion.

Also würde ich für mich persönlich das gleiche Argument in Anspruch nehmen wollen.

Vor allem wenn ich auf mehrmalige konkrete Nachfrage lese:

> It operates and boots fine...but offering support...I wouldn't know how to troubleshoot

Sorry, ... ohne Worte!

Die Anzahl der `single-posting-user`  ist nun seit geraumer Zeit schon exorbitant angewachsen.
Die Qualitätsniveau der Fragen ist dabei unterirdisch. Klar,  posten ist deutlich einfacher als sich selbst um eine Antwortfindung zu bemühen.
Fast alle Antworten sind durch Lesen im Forum oder suchen im Internet beantwortbar.
Mehr kann und tue ich auch nicht. Somit sollte fast jeder Forenteilnehmer dies auch können, der länger als drei Wochen dabei ist und DietPi nutzt.

Was zeigt mir das alles bitte?
Die "Community" funktioniert einfach nicht. Und das nicht erst seit gestern. 
Fast alle nur Konsumenten mit hoher Erwartungshaltung und Ansprüchen.
Kaum einer will was tun, Aufgaben übernehmen und anpacken.
Und wenn doch, dann aber nicht auf Dauer.
Glaube nach 2 1/2 darf ich das behaupten.

k-plan, 2018-03-14:
> Open to suggestions, @MichaIng how you like your name mentioned? smiley

Yep! :100:  :+1: 

> Requesting "Please enter your name", this could then be displayed in the DietPi banner (as opposed to current)?

Not could ... it is must! Without no PREP !

```
───────────────────────────────────────
UserPi | 22:04 | Tue 13/03/18
───────────────────────────────────────
v6.4 | OrangePi Zero (armv7l)
───────────────────────────────────────
eth0 | 192.168.0.98
───────────────────────────────────────

Created by : John Doe 
Web : http://john.doe.net

Device image possible thanks to: - DietPi

Unsupported user build image version,
no longer supported by DietPi, so you are on your own.

```

But this only produce unnecessary workload.
Disallow spreading the world with unsupportable versions.

> Being open source, you can use any pre-image you like.

Correct. For personal use and for your self, it's no problem.
https://github.com/Fourdee/DietPi#dietpi-gplv2-license

But:
> You should have received a copy of the GNU General Public License along with this program. 

Same for the image base and you have to mention, how is the owner/maker/builder.

> You can also host the images (unofficially) you create anywhere you like.

What is the benefit of allow to do this?
If we force this, hassle is inevitable.

Fourdee, 2018-03-14:
@k-plan 

> The number single-posting-user has been growing for some time already exorbitant. 
The quality level of the questions is underground. Of course, posting is much easier than trying to find an answer. 
Almost all answers can be answered by reading in the forum or searching the internet. 
The "community" just does not work. And not just since yesterday. 

Yep agree. 
I'am tempted to disable the forum registrations and revert back to Github for support. The forum is simply a spamfest place for "Can I do this?" questions without the user being bothered to even try it.
There is about 0.1%? of posts being general bug reports or issues, but aside from that, its a shambles and a waste of our time in most regards.

@MichaIng @k-plan 
Thoughts on disable forums?

At least with GitTickets, we have a good template that should force the user to provide the details (and in some cases, question their original intentions)

> You can also host the images (unofficially) you create anywhere you like.
What is the benefit of this? 
If we force this, hassle is inevitable. See ARM ... before.

Generally we cannot control what users post/host outside of dietpi.com. So yes, as per your idea, we need to ensure in PREP that any image we have not made, is labeled as "unofficial/not supported" in some method.

SuBLiNeR, 2018-03-14:
Hi there @all 
first thing what should be done at least is to set dietpi.com as an https-site with working SSL and maybe md5's or GPG's provided downloads so ppl will more head over to #DietPi as an alternative to Raspbian Lite images.
In my opinion the DietPi is the most affordable image for the ppl who are familiar with the CLI and not addicted to GUIs.
The ppl who are asking different questions on the Forum or even on this git, are nearly almost to lazy to search the web or find solutions on their own.
The problem is stated within the behave at least, that the ppl only want to have some final solutions to their problems given and not do some research by themselves to get it solved. Main concern upon this is, if they found a solution, they'll post it or they'll post it with wrong solutions. It may depend at least on personal interests or not, but most of the times, the people will behave bad on you if not clearly stated what they're upon off.

So, even if this is somehow not logical or whatever i've written down, this is what i think! Mine intention is, that your work should be more being "grateful" (?)! 

DietPi is one of my personal favorites over the rest of supported images... even if it has no SSL or MD5s etc.... 

Keep up the good work n don't let you get down!

k-plan, 2018-03-14:
@Fourdee 

> Thoughts on disable forums?

puuh ...  I really don't know. 

> and revert back to Github for support. 

Hmm ... github is not a forum or user plattform. I dislike this idea. 
It's more difficult to handle. For user and for us. 
Most are not possible to fill out template or copy your great new error message system.
Sorry.

> Generally we cannot control what users post/host outside of dietpi.com.

That's correct! :+1: 

But you have said:  `You can also host the images (unofficially) you create anywhere you like`

This will sound for most every one like a challenge to do this with your agreement.

So I can understand others will not approve this. Sorry.

MichaIng, 2018-03-15:
@Fourdee @k-plan @SuBLiNeR 
Oh wow, my intention was not to start such a deep discussion, thought it a bid easier, but okay.

1. **Resharing/GPL**
As far as I know/understand (fast check wikipedia) and also as I like it, the GPL licence clearly allows free usage and resharing of the product, as long as the user/resharer applies the same GPL licence again. Resharing always bears the risk, that end users address issues wrong, but this is nothing new: e.g. SBC producers offer crappy images and end users sending but reports to Debian bug tracker or something like that 😆. But I love this licence concept which guarantees that good code is never doomed to die and can be freely used/enhanced/forked/reshared to make people lives better. The only thing we can and should do to counteract wrong addressed issues is clarifying who is responsible for which part of the final product.

2. `It operates and boots fine...but offering support...I wouldn't know how to troubleshoot`
Hmm, maybe I see this from a totally different angle: I appreciate that someone not only wants to create an image for an now officially unsupported device for himself, but offers to share this to other users who suffer the same missing official image 👍. Of course the support he can offer is limited to what it took to create the image and high likely can not include DietPi code related issues. These on the other hand is what we WANT to be reported here to be able to fix for all and make DietPi better. The only thing that indeed needs to be clear there, is the above: DietPi code is still from us, but image not created/tested by us. Of course it is guaranteed that issues can still be addressed wrong, but this is an absolutely usual problem in foss world, e.g. kernel or repo related issues mentioned here, web server related issues addressed to Nextcloud, Windows related problems addressed to software producers or the other way round, ah even on non-foss 😆... I guess using 3rd parties work as basement still saves more time than inventing the whole thing new from ground, redirecting issues/requests is a damn low price for this, that sometimes others need to pay because of us, sometimes us because of others 🤔.

3. **Forum**
Hah, same discussion was on Nextcloud forum. Not about closing the forum, damn no, but about how to handle issue sorting and in case redirection to and from github better, as both are used overlapping. But especially from end user perspective it think it is beneficial to have a forum outside of github. Github at least looks like the coders world, where new issues usually contain a large template to fill and give the impression that some kind of background knowledge is necessary, no matter if it is used/moderated strictly like that or not. A "simple" forum on the other hand usually looks more inviting to new users to freely ask questions, discuss etc. Or course most users will not stay for long after their question was answered, I mean how much forums I would need to check regularly if doing this for every piece of software I once used and had questions/issues with. I stay active where I am long time user and **like** and feel able to contribute. That some users need to be taken by their hand, where others just need a pointer into the right direction is something that can be seen on every forum and is mostly handled by itself. Also on DietPi forum we home quite active users regularly helping and discussing others questions/issues. No idea about usual daily active forum user numbers compared to product user numbers, blaming anyone, if it doesn't feel too self-sustaining will definitely not make something better 😛.

4. **Waste of time?**
Compared maybe to github, I see a forums task actually in being a time waste, from the pure productive point of view. As said, users might (and should?) feel free to wish everybody a nice weekend or ask questions that may sound dump from others perspective, who then on the other hand should be able to answer them without our attention, even if it's sometimes a rough goo*** link 😉. How often I see this on so many forums 😆.
I think if our time is rare, we just need to feel free to not answer every simple topic on the forum. But sometimes I enjoy just reading a bid and dealing with non-coding issues. Everything that might need our attention should be reposted here on github of course, which works already quite well?
________________________
Back to initial topic, I tought about something like this:
```
DietPi code created by : Daniel Knight
Image created by       : <custom entry, e.g. MichaIng + email/link if I want to share>
Image based on         : <custom entry, e.g. https://downloads.raspberrypi.org/raspbian_lite/ or just Raspbian Lite>
```
Maybe we can make DietPi-Prep then add `(official image)` in the back of the custom name, if `Daniel Knight`/`Fourdee`, `k-plan` or `MichaIng` was found within the string, using an "officials" array or something 😄.

Fourdee, 2018-03-15:
@MichaIng 

> Maybe we make DietPi-Prep then add (official image)in the back of the custom name, if Daniel Knight/ Fourdee, k-planor MichaIngwhat found within the string, using "officials" array or something

Yep, had the same thoughts, looks good 👍 

Run the whiptail prompts in PREP, then create a file in ```/DietPi/dietpi/.prep_info``` which we can scrape from ramdisk and banner.

> Image based on Raspbian Lite

Yep

> https-site with working SSL

@SuBLiNeR 
Not only do we need SSL, but also a webserver upgrade. We need to migrate from Wheezy to Stretch, without downtime. So i'll discuss with the hosting provider and see if they can offer us another server for the switch.

k-plan, 2018-03-15:
@Fourdee

> > Thoughts on disable forums?
> 
> puuh ... I really don't know.

Had to sleep over this.
Vote for leave it online. 
Why? All your documentation and discretion for DietPi is there.
Will be a lot of work to get and host it somewhere else.
And dislike, to get totally depend from a third party platform.
If github will change their terms of use  or want to get paid for their service  ....

@MichaIng 

> 1. Resharing/GPL
Sehe, du hast dich belesen und auch Gedanken gemacht. :+1: 
Mehr als das die meisten "image-sharer" tun werden.
Davon ab, so weit ich das verstehe, muss auch je nach Lizenzmodel z.B. eine Kopie der GPL mit dem Produkt überreicht werden und man muss zu mindestens auf den zu Grunde liegenden Quellcode verweisen oder zur Verfügung stellen.
[DietPi macht das](https://github.com/Fourdee/DietPi#dietpi-gplv2-license) ,  ist sogar bei vielen Geräten so z.B. Internet Router. Übrigens kann ich bei ARM... keine solche Hinweise entdecken.  Ist aber nicht unser Thema hier.

> 2. Hmm, maybe I see this from a totally different angle:   I appreciate that  ....
Richtig, anderer Blickwinkel und der sei dir auch ungenommen.
Ich versuche aus "Fehlern" der Vergangenheit etwas zu lernen.
So wie ich das verstehe, ist Igor erst so angepisst, auf Grund der Nutzer Aussagen Dietpi wäre das bessere ARM... und wenn es kein DietPi für das Gerät gibt, dann wäre ARM... so ganz okay.
Das das so Aussagen einen tief treffen können und man überreagiert, kann ich durchaus nachvollziehen und habe auch Verständnis dafür. Die Verärgerung dann aber an der Person Fourdee auszulassen, war voll daneben. Genau wie einige andere Aussagen.
Und solche Vorfälle würde ich gerne zukünftig von vorne herein ausschließen und vermeiden.
Nur wenn man die Leistung und Mühen anderer respektiert, kann man selber Respekt erwarten.
Und nein, eine `.img` und ein fertiges Script, beides von Dritten erstellt, aus dem Internet herunter zu laden, durchlaufen lassen und zu verteilen, ist nach meinem Dafürhalten keine persönliche Leistung.
So gar nicht, wenn man sich nicht im Voraus mit den möglichen (persönlichen) Folgen für andere auseinander gesetzt hat (Support, Vorwürfe gegen die Person, Reputations- u. Vertrauensverlust).

Daher würde ich begrüßen, Aussagen wie  `You can also host the images (unofficially) you create anywhere you like` zu revidieren. Erzeugt bei kritischen Betrachtern nur Unmut und schafft unnötige Angriffsfläche. Sorry.

> 3. Forum
In den meisten Punkten teile ich deine Einschätzungen.
> I stay active where I am long time user and like and feel able to contribute. 
> Also on DietPi forum we home quite active users regularly helping and discussing others questions/issues.
Nur diese zwei druurfte ich in den letzen Jahren anders erleben oder empfinden.
Da regelt sich gar nix selber, außer stetig wachsenden Ansprüchen. Auch der Umgangston ist deutlich rauer.

> 4. Waste of time?
> I think if our time is rare, we just need to feel free to not answer every simple topic on the forum. 
Leider deutet der Betrieb eines Forums neben der Last auch eine Verpflichtung.
Daher kann ich Fourdee Gedankengang zur Abschaltung schon nachvollziehen.
Wie bei allem, verteilt ich die Last auf zu wenige Schultern.

Fourdee, 2018-04-11:
[image]

Fourdee, 2018-04-12:
[image]

Fourdee, 2018-04-12:
Completed.

Fourdee, 2018-04-12:
@MichaIng @k-plan 

When using PREP, enter following image creator name (number):

- Micha = 1
- K-Plan = 2

closed 2018-04-12 by Fourdee

MichaIng, 2018-04-12:
@Fourdee 
Looks great, also nice solution to prevent the usage of our names 👍.
2 Lines more, but still fully fits onto VirtualBox default resolution terminal 😄:
[image]

k-plan, 2018-04-12:
@Fourdee 

> When using PREP, enter following image creator name (number):

>  - Micha = 1
> - K-Plan = 2

... and what is with:
- Fourdee = 3

:question:  :fearful: 

But joking aside.

1.) Hmmm ... I personally would prefer:

-  DietPi Project Team = 0

and finish. 
No need to edit, if anything will change in future.
And, because Fourdee have to upload the image to dietpi.com, he always know, who as build it from DietPi Project Team.

2.) No need to give `additional credits` to k-plan (me), because  I can and have nothing to do with image build process. Images have exist and will exist without me.
But will not exist without @Fourdee  and @MichaIng .

Only hunting some bugs and give some advice, nothing more. So no need to see my name in this line.

MichaIng, 2018-04-12:
@k-plan 
I guess Daniel Knight / Fourdee = 0 😉.

Agree with "DietPi Project Team = 0" or "DietPi Project Official" or something. In the end issues with images from us three should be addressed the same way (here or forum) anyway.

Additional credits:
Hehe, DietPi would run with Fourdee alone as well. But as long as we have official contributors, helping the one or the other way, it's fine to give credits to all of them, otherwise to non. All fine, I mean you can review contributors here on github and see admins and moderators on the forum as well. I guess it's quite rare to give credits to those that prominent within the software, but quite nice for me on non-profit FOSS projects.

Fourdee, 2018-04-12:
@k-plan 

> No need to give additional credits to k-plan (me), because I can and have nothing to do with image build process. Images have exist and will exist without me.

True, but you have/still provide valuable input and insight over the years, without which, DietPi would not be as good as it is today. So your name is staying in the credits 😃 

@k-plan @MichaIng 

> Agree with "DietPi Project Team = 0"

Yep, good clean idea 👍 , i'll make the change and send commit.