MichaIng/DietPi #1623: Can not update to version 6.4

nakulbende, 2018-03-15:
## Issue:

I can not update to 6.4. I have run all the apt commands to get any dependencies, made sure  I have internet and I can ping dietpi.com. Here is the output: 

```
DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     4  100     4    0     0     12      0 --:--:-- --:--:-- --:--:--    12
[  OK  ] Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6

[  OK  ] Update available
[ INFO ] Current Version : 6.1
[ INFO ] Server Version  : 6.4
[ INFO ] Detecting drives, please wait...
[ INFO ] Detected PARTUUID mount: /dev/mmcblk0p1 > /boot
[ INFO ] Detected PARTUUID mount: /dev/mmcblk0p2 > /
[ INFO ] Processing drive information, please wait...
[ INFO ] Checking available free space on RootFS, please wait...
[  OK  ] 5327 MB available, 500 MB required

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] stop : cron
[  OK  ] stop : proftpd
[  OK  ] stop : apache2
[  OK  ] stop : lighttpd
[  OK  ] stop : php7.0-fpm
[  OK  ] stop : noip2
[  OK  ] stop : dnsmasq
[  OK  ] stop : pihole-FTL
[  OK  ] stop : openvpn
[  OK  ] stop : avahi-daemon
[  OK  ] stop : netdata
[  OK  ] stop : plexmediaserver
[  OK  ] stop : qbittorrent
[  OK  ] stop : blynkserver
[  OK  ] stop : motioneye
[  OK  ] stop : cups
[  OK  ] stop : plexpy
[  OK  ] stop : RaspimJpeg

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:19 --:--:--     0curl: (6) Could not resolve host: dietpi.com
[FAILED] Download failed, unable to run update. Please try running dietpi-update again.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] Current Version : 6.1
[ INFO ] Server Version  : 6.4
[  OK  ] Update completed

Please reboot your system now, using the command reboot
```
#### System Information:
- DietPi Version | 6.1
- RPi 3
- 5v 2.5A
- Sandisk ultra

k-plan, 2018-03-15:
@nakulbende 

> If you notice, the update script can connect to dietpi.com initially, but not the second time around. Any thoughts?

Yes.

> - DietPi Version | 6.1 
>
> ```
> DietPi-Services
> ─────────────────────────────────────────────────────
>  Mode: stop
> ...
> [  OK  ] stop : dnsmasq
> 
> ...
> 
> curl: (6) Could not resolve host: dietpi.com
> [FAILED] Download failed, unable to run update. Please try running dietpi-update again.
> 
> ```

Bug is fixed with DietPi v6.2

https://github.com/Fourdee/DietPi/commit/640ff4f03300772be2f694ef371e75ba5070b0f9

To get `dietpi-update` running now, please edit as `root`:

`root@Odroid-C1:~# nano /DietPi/dietpi/dietpi-services`

and rem out (#) the line 53:

```              .....
		'ympd'
 		'minidlna'
 		'noip2'
  	        #'dnsmasq' #: https://github.com/Fourdee/DietPi/issues/1501
 		'pihole-FTL'
 		'subsonic'
 		'webiopi'
                 ...
```

Save file, and start `dietpi-update`

k-plan, 2018-03-16:
@nakulbende 

> ... - still got the same error.

1.) -  Can you please share the output again?

-----

```
DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

....
[  OK  ] stop : openvpn
...
```
2.) - Is WAN (internet) connection of your RPi via a VPN Tunnel Provider?

----------

3.) - Can you please show:

- ping ip address of your default gateway (router): `root@RPi:~# ping "ip-address-of-your-router"`

- look at you RPi ip address: `root@RPi:~# ip address`

- look at your RPi Kernel IP routing table: `root@RPi:~# route`

- look at your RPi DNS server: `root@RPi:~# cat /etc/resolv.conf`

nakulbende, 2018-03-16:
My Ethernet connection is directly from the router - the openvpn client actually uses RPi as a host for me to VPN into my network from outside. 

Here is the ping output:
```
--- 192.168.0.1 ping statistics ---
32 packets transmitted, 32 received, 0% packet loss, time 32159ms
rtt min/avg/max/mdev = 0.363/0.446/0.529/0.041 ms
```
IP address output (my DCHP gives static address to alll): 
```
root@DietPi:~# ip address
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:ed:fc:2e brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.201/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:feed:fc2e/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:b8:a9:7b brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.200/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:feb8:a97b/64 scope link
       valid_lft forever preferred_lft forever
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 100
    link/none
    inet 10.8.0.1/24 brd 10.8.0.255 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::d245:6841:f527:2d67/64 scope link flags 800
       valid_lft forever preferred_lft forever
```
Routing table:
```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         Archer50        0.0.0.0         UG    0      0        0 eth0
10.8.0.0        0.0.0.0         255.255.255.0   U     0      0        0 tun0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 wlan0
```

Namesevers: (First one is raspPi itself, as I use pi-hole. But the second one is quad-8)
```
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
nameserver 192.168.0.201
nameserver 8.8.8.8
```

Here is the ``dietpi-update`` output:
```
[  OK  ] Root access verified.
[ INFO ] Checking RootFS R/W access. Please wait...
[  OK  ] RootFS R/W access.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     4  100     4    0     0     10      0 --:--:-- --:--:-- --:--:--    10
[  OK  ] Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6

[  OK  ] Update available
[ INFO ] Current Version : 6.1
[ INFO ] Server Version  : 6.4
[ INFO ] Detecting drives, please wait...
[ INFO ] Detected PARTUUID mount: /dev/mmcblk0p1 > /boot
[ INFO ] Detected PARTUUID mount: /dev/mmcblk0p2 > /
[ INFO ] Processing drive information, please wait...
[ INFO ] Checking available free space on RootFS, please wait...
[  OK  ] 5356 MB available, 500 MB required

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] stop : cron
[  OK  ] stop : proftpd
[  OK  ] stop : apache2
[  OK  ] stop : lighttpd
[  OK  ] stop : php7.0-fpm
[  OK  ] stop : noip2
[  OK  ] stop : pihole-FTL
[  OK  ] stop : openvpn
[  OK  ] stop : avahi-daemon
[  OK  ] stop : netdata
[  OK  ] stop : plexmediaserver
[  OK  ] stop : qbittorrent
[  OK  ] stop : blynkserver
[  OK  ] stop : motioneye
[  OK  ] stop : cups
[  OK  ] stop : plexpy
[  OK  ] stop : RaspimJpeg

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:19 --:--:--     0curl: (6) Could not resolve host: dietpi.com
[FAILED] Download failed, unable to run update. Please try running dietpi-update again.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] Current Version : 6.1
[ INFO ] Server Version  : 6.4
[  OK  ] Update completed

Please reboot your system now, using the command reboot
```

nakulbende, 2018-03-16:
Oh wait, I just removed DNS servers from my router settings, and it worked! Looked like it was forced to take the fist DNS name server, and did not go to the backup one.

nakulbende, 2018-03-16:
Actually, should not have spoken so soon. I got an error, and when it rebooted, it is on v6.2. Once I get home, I will get the error dump. 

I guess as far as this issue goes, we can close it (?) I will open a new one for my update woes.

Fourdee, 2018-03-16:
@nakulbende 

> Actually, should not have spoken so soon. I got an error, and when it rebooted, it is on v6.2. Once I get home, I will get the error dump.
I guess as far as this issue goes, we can close it (?) I will open a new one for my update woes.

Continue with this ticket, we can edit the title if needed 😃

k-plan, 2018-03-16:
@nakulbende 

You have several problems. But point by point.

----------
> Here is the `dietpi-update` output:

```
DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
....
```
Looks good for now.  
`dnsmasq` will not be stopped any more, while `dietpi-update` is running.
You have edit `/DietPi/dietpi/dietpi-services` correct and it is working. :+1: 

-------------------

>My Ethernet connection is directly from the router - the openvpn client actually uses RPi as a host for me to VPN into my network from outside.

Okay. OpenVPN tunnelling is not you problem at the moment. :+1: 

---------------

> `--- 192.168.0.1 ping statistics ---`
> `0% packet loss`

Okay, you default gateway is reachable :+1:  and if I see it correct, it's a` TP Link Archer 50`

-------------

> ```
> 2: eth0: ... state UP ...
>     inet 192.168.0.201/24 brd 192.168.0.255 scope global eth0
>
> 3: wlan0: ... state UP ...
>     inet 192.168.0.200/24 brd 192.168.0.255 scope global wlan0
> ```

1.) - you have to disable WiFi  via `dietpi-config`  so `wlan0` don't get a IP address any more.

2.) - you have to set Wired Connection (`eth0`) via `dietpi-config` to `STATIC` eg.  192.168.0.201

```
Change Mode    STATIC                                 
Copy           Copy Current address to Static         
Static Ip      192.168.0.201                         
Static Mask    255.255.255.0                          
Static Gateway 192.168.0.1                            
Static DNS     8.8.8.8                          
Disable        Disable Ethernet adapter              
Apply          Save Changes (Restart Networking) 
```

> IP address output (my DCHP gives static address to alll):

Please disable this for 192.168.0.201 

----------------

> Routing table:
>
> ```
> Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
> default         Archer50        0.0.0.0         UG    0      0        0 eth0
> 10.8.0.0 [...]

It should now look like this:

```
root@RPi-Zero:~# route -n

Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1        0.0.0.0         UG    0      0        0 eth0
10.8.0.0        0.0.0.0         255.255.255.0   U     0      0        0 tun0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
```

-------------------

> Namesevers: (First one is raspPi itself, as I use pi-hole.

Your` /etc/resolv.conf` with running `pihole` should look like this:

```
root@RPi-Zero:~# cat /etc/resolv.conf

# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
nameserver 127.0.0.1

```
But to simple edit the file will not do the trick.

- Check if pihole is up and running correct:

```
root@RPi-Zero:~# pihole status

  [✓] DNS service is running
  [✓] Pi-hole blocking is Enabled

```

Then as `root`:

```
## Backup lighttpd.conf
mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak

## Reconfigure or Repair Pi-hole subsystems
pihole -r

## Restore lighttpd.conf
cp /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

## Restart all running services
dietpi-services restart

## Display the running status of Pi-hole subsystems
pihole status

## check resolv.conf
cat /etc/resolv.conf

## finished

```

Now DNS, routing and `dietpi-update` should work.

nakulbende, 2018-03-16:
Hey @k-plan , thanks a lot. I will go home and redo this (for some reason, my VPN client is not working, so I can't VPN in and do changes from work. I remember restarting the pi after update to 6.2, so we will see). 

Some small questions:

> 2.) - you have to set Wired Connection (eth0) via dietpi-config to STATIC eg. 192.168.0.201

So only here? `Change Mode    STATIC`                           

>Copy           Copy Current address to Static         
>Static Ip      192.168.0.201                         
>Static Mask    255.255.255.0                          

And Manually set this? `Static Gateway 192.168.0.1 `                           

>Static DNS     8.8.8.8                          
>Disable        Disable Ethernet adapter              
>Apply          Save Changes (Restart Networking) 
>IP address output (my DCHP gives static address to alll):

>Please disable this for 192.168.0.201

Will do. Out of curiosity, DHCP is giving the pi a static address, so why does it matter that we need to remove this, and have pi setup as static. 

As far as the pihole goes, Do i still use sudo nano to open this file, change nameserver to localhost and save it, follow your steps to make it permanent?

>root@RPi-Zero:~# cat /etc/resolv.conf
> Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
> DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
>nameserver 127.0.0.1

Would I have to change it back? or should the nameserver for pi be always localhost?

k-plan, 2018-03-16:
@nakulbende 

> So only here?

It is whiptail menu. Move with courser keys and press enter to edit.

After this and hit `Save Changes (Restart Networking)`

your settings should look like this and `#Please use DietPi-Config to modify network settings.`:

`root@RPi-Zero:~# cat /etc/network/interfaces`

```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

...
# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.201
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8

# Wifi
#allow-hotplug wlan0
...

```

> so why does it matter that we need to remove this, and have pi setup as static.

Pi-Hole installation told you to set a static ip to get work properly.

This is a little bit outdated, but overall you can see some nice screen shots:
https://jacobsalmela.com/2015/08/26/a-new-easy-installation-method-for-the-ad-blocking-pi-hole/ 
It shows, what `dietpi-software`  shows you before it will start the pihole installer

> As far as the pihole goes, Do i still use sudo nano to open this file, change nameserver to ...

Sorry I read in the file :

```
Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
```
`dietpi-config`, `dietpi-software`, `pihole installer` ... use whiptail menus to set and edit config files.

So please use this menu and do not edit config files they need or set by hand.
Yes you can do this, but only if you know what you are doing.

> or should the nameserver for pi be always localhost?

pihole did this since some past version. It's needed for pihole to work correct. But pihole installer did it for you  while install pihole, pihole-FTL, dnsmasq, and many other packets.

nakulbende, 2018-03-16:
So, not sure what just happened. I tried to login with SSH, and this is what I got. 

```
login as: root
root@192.168.0.201's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
grep: /DietPi/dietpi.txt: No such file or directory
-bash: gettext.sh: No such file or directory
-bash: /DietPi/dietpi/func/dietpi-globals: No such file or directory
-bash: /DietPi/dietpi/login: No such file or directory
```

None of the dietpi-xxx commands work. What is wrong with my dietpi? I did take a backup using dietpi-backup, if needed.

Fourdee, 2018-03-18:
@nakulbende 

Please paste results:
```
systemctl status dietpi-ramdisk -l
```

nakulbende, 2018-03-19:
Hey all, 

So - I basically lost access to SSH altogether after i posted last time. When I connected it to the a screen, I saw a Ext4-fs error (something like can not find itable). I powered off my Pi, and tried to open the SD card on my computer, but it was corrupted. I can see some files, but even config.txt and contents of dietpi folder were missing! Seems like my SD card was corrupted while I had to pull the plug on my Pi, due to unresponsiveness which happened a couple of times during update.  It seems like the issue I had was much alike this [one](http://dietpi.com/phpbb/viewtopic.php?f=11&t=1135). 

I decided that I gotta just start with a fresh installation - buti did run into some problems (seems like the default webserver changed in 6.4 - failed installation of webserver prevented a successful pi-hole admin webadmin, so i uninstalled lighttpd and installed apache. Seems to work fine for now).  I have a screnshot on my home desktop for the error - will post here if you guys are interested. 

In short - the new installation was not as error free as my other dietpi installations, but I think it has been working. I am hoping that it was a read/ write intteruption that caused my SD card to go. Will update!

Thanks a lot for both of your help!

PS: for my education: what does this do?

```
systemctl status dietpi-ramdisk -l
```

nakulbende, 2018-03-19:
Sorry - this was scrolling really fast, so I could only capture this much. 

[image]

k-plan, 2018-03-20:
@nakulbende 

> PS: for my education: what does this do?

- you can check status of the essential service, which must run in the background, to get all dietpi scripts to work: 

```
root@RPi-Zero-W:~# systemctl status dietpi-ramdisk -l

● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2018-03-18 23:04:24 CET; 1 day 9h ago
   CGroup: /system.slice/dietpi-ramdisk.service

Mar 18 23:04:22 RPi-Zero-W systemd[1]: Starting DietPi-RAMdisk...
Mar 18 23:04:24 RPi-Zero-W systemd[1]: Started DietPi-RAMdisk.
```

or you can proof it with:

```
root@RPi-Zero-W:~# service dietpi-ramdisk status

● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2018-03-18 23:04:24 CET; 1 day 9h ago
   CGroup: /system.slice/dietpi-ramdisk.service

Mar 18 23:04:22 RPi-Zero-W systemd[1]: Starting DietPi-RAMdisk...
Mar 18 23:04:24 RPi-Zero-W systemd[1]: Started DietPi-RAMdisk.
```

Can't explain it better when in the header comments of the script.:

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-ramdisk#L3-L21

```
	#////////////////////////////////////
	# DietPi RAMDISK Script
	#
	#////////////////////////////////////
	# Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
	#
	#////////////////////////////////////
	#
	# Info:
	# - Copies dietpi files from /boot to /DietPi (ram)
	# - /DietPi is mounted as tmpfs (ram)
	# - This allows us to vastly reduce filesystem IO across DietPi.
	# - Creates a file when active (/DietPi/.ramdisk)
	#
	# Usage:
	# - Called from /DietPi/dietpi/conf/dietpi-service
	# - /DietPi/dietpi/dietpi-ramdisk 0				= Copy from /boot (disk) to /DietPi (ram)
	# - /DietPi/dietpi/dietpi-ramdisk 1				= Copy from /DietPi (ram) to /boot (disk)
	#////////////////////////////////////
```

and the result:

```
root@RPi-Zero-W:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.3G  969M  6.1G  14% /
devtmpfs        237M     0  237M   0% /dev
tmpfs           241M     0  241M   0% /dev/shm
tmpfs           241M  3.3M  238M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           241M     0  241M   0% /sys/fs/cgroup
tmpfs            10M  1.5M  8.6M  15% /DietPi
tmpfs            20M   24K   20M   1% /var/log
tmpfs           490M     0  490M   0% /tmp
/dev/mmcblk0p1   42M   23M   19M  54% /boot
```

> I decided that I gotta just start with a fresh installation - ....

A good decision. 
After your previous post `... No such file or directory` we normally say to start with a fresh image on [a full formated sd card](https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267), because your file system is corrupt.
But most user don't want do hear this and get instantly acidify.
So I have decided to leave well alone, to get around this excruciating discussion.
It's the downside of sd card flash technology, not a issue of any distro or DietPi itself.
Use the best sd card you can get and never pull power down while running.

DietPi try to get this more robust and reliable and this is, what `dietpi-ramdisk` does.
But miracle are usually impossible.

k-plan, 2018-03-20:
@Fourdee 

> buti did run into some problems (seems like the default webserver changed in 6.4 - failed installation of webserver

We have to take a look on it.

```
#       Webserver Preference Selection:
#       NB: This will get ignored, if you have manually selected any WEBSERVER_Stacks.
#               0=Apache2
#               -1=Nginx
#               -2=Lighttpd
AUTO_SETUP_WEB_SERVER_INDEX=-1
```

[image]

Uppps ...

nakulbende, 2018-03-20:
Hey all, 

I soon went with ngix, I wish I chose that in the config file as you said - because now I keep getting this error when i install some apps (I was installing iptraf and some other network monitoring things when this happened). It seems like it does not remember that I uninstalled Lighttpd and installed ngix.

[image]

Any way to choose default webservers? Is it form the whiptail menu you showed in last post? If so, I will check it as soon as I get back - pretty sure it was set to ngix.

nakulbende, 2018-03-20:
>We have to take a look on it.

@k-plan  and @Fourdee : If it helps, here is a partial log for my installation, during which I believe webserver was creating issues. I might be wrong. 

```
[  OK  ] DietPi-Services | dietpi_controlled : qbittorrent
[  OK  ] DietPi-Services | dietpi_controlled : blynkserver
[  OK  ] DietPi-Services | dietpi_controlled : motioneye
[  OK  ] DietPi-Services | dietpi_controlled : cups
[  OK  ] DietPi-Services | dietpi_controlled : plexpy
[  OK  ] DietPi-Services | dietpi_controlled : raspimjpeg
[ SUB2 ] DietPi-Set_Hardware > gpumemsplit (128)
[  OK  ] gpumemsplit 128 | Completed
[  OK  ] DietPi-Software | DietPi-Software installation completed.
[ SUB2 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[  OK  ] DietPi-Services | start : proftpd
[  OK  ] DietPi-Services | start : apache2
[  OK  ] DietPi-Services | start : lighttpd
[  OK  ] DietPi-Services | start : php7.0-fpm
[  OK  ] DietPi-Services | start : noip2
[  OK  ] DietPi-Services | start : pihole-FTL
[  OK  ] DietPi-Services | start : avahi-daemon
[  OK  ] DietPi-Services | start : netdata
[  OK  ] DietPi-Services | start : plexmediaserver
[  OK  ] DietPi-Services | start : qbittorrent
[  OK  ] DietPi-Services | start : blynkserver
[  OK  ] DietPi-Services | start : motioneye
[  OK  ] DietPi-Services | start : cups
[  OK  ] DietPi-Services | start : plexpy
[  OK  ] DietPi-Services | start : raspimjpeg
[ SUB3 ] DietPi-Process_tool > Apply
[FAILED] DietPi-Process_tool | NetData (23028) : Nice      19
[FAILED] DietPi-Process_tool | NetData (23028) : Affinity  0-3
[FAILED] DietPi-Process_tool | NetData (23028) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | NetData (23029) : Nice      19
[FAILED] DietPi-Process_tool | NetData (23029) : Affinity  0-3
[FAILED] DietPi-Process_tool | NetData (23029) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | NetData (23030) : Nice      19
[FAILED] DietPi-Process_tool | NetData (23030) : Affinity  0-3
[FAILED] DietPi-Process_tool | NetData (23030) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Pi-hole FTL (22873) : Nice      0
[FAILED] DietPi-Process_tool | Pi-hole FTL (22873) : Affinity  0-3
[FAILED] DietPi-Process_tool | Pi-hole FTL (22873) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Webserver Apache2 (23362) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Apache2 (23362) : Affinity  0-3
[FAILED] DietPi-Process_tool | Webserver Apache2 (23362) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22803) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22803) : Affinity  0-3
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22803) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22804) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22804) : Affinity  0-3
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22804) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22805) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22805) : Affinity  0-3
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22805) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22806) : Nice      0
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22806) : Affinity  0-3
[FAILED] DietPi-Process_tool | Webserver Php-fpm (22806) : Scheduler SCHED_OTHER 0
[FAILED] DietPi-Process Tool | An issue has occured
[ INFO ] DietPi-Patchfile | APT installation for: dirmngr, please wait...
```

MichaIng, 2018-04-03:
@nakulbende 
Hui, sorry for late reply from our side. Did you managed it in between?

From both of your logs I see at least 2 webserver installed at a time. `dietpi-software` should warn to just select one webserver stack, recommending to just select the webserver preference and the frontend software you need, letting DietPi automatically choose the right webserver stack. But it does not block parallel webserver installations, which then break each other.

If I got it right, you want to use nginx? No database with it?

Try the following:
1. Uninstall all non-nginx webservers + stacks: `dietpi-software uninstall 75 76 81 82 83 84`
2. Reinstall nginx and PHP: `dietpi-software reinstall 84 89`

k-plan, 2018-04-07:
@MichaIng 

> ... not block parallel webserver installations, which then break each other.

Hmm... habe da einmal ein wenig herum gespielt.
Der zweite `webserver` `lighttpd` kommt von `pihole` Installer. 
Spielt auch keine Rolle, ob man via `dietpi-software` installiert oder nicht.

Test:

- Frisches Images ohne jegliche Zusatzsoftware
- in `dietpi-software` =>>  nun `Webserver Preference : Nginx` ausgewählt
[image]

- via `dietpi-software` =>> `pihole` installieren

1. `dietpi-software` installiert `Nginx`. 
2. `dietpi-software` lädt und startet den `pihole installer` und dieser installiert dann `lighttpd`

=>> wir haben nun auf dem System 2x `webserver` laufen, von denen einer chrashed.

... und anders herum:

- Frisches Images ohne jegliche Zusatzsoftware
- in `dietpi-software` =>>  als Standard Vorgabe ist `Webserver Preference : Lighttpd` ausgewählt

- via `dietpi-software` =>> `pihole` installieren

1. `dietpi-software` installiert `lighttpd`. 
2. `dietpi-software` lädt und startet den `pihole installer` und alles ist schick. (bis auf die Update Problematik)

Aber wenn man nun `Webserver Preference` nachträglich noch einmal ändern möchte, funktioniert das nicht (dauerhaft).

Die  2x `webserver` "Problematik" taucht auch auf, wenn man Software zur Installation auswählt, die einen andern Webserver als `lighttpd` wie `nginx` `tomcat` oder `apache` benötigt.

----------------------

`pihole` mit `Nginx` läuft zwar, aber man muss dann auch bei jedem `pihole` Update wieder per Hand dran herum machen. Gerade noch einmal geprüft.

```
───────────────────────────────────────
 DietPi     | 13:00 | Sat 07/04/18
 ───────────────────────────────────────
 v6.6       | RPi Zero W (armv6l)
 ───────────────────────────────────────

root@RPi-Zero-W:~# pihole -v
  Pi-hole version is v3.3 (Latest: v3.3.1)
  AdminLTE version is v3.3 (Latest: v3.3)
  FTL version is v3.0 (Latest: v3.0)
```

```
root@RPi-Zero-W:~# pihole -up
  [i] Checking for updates...
  [i] Pi-hole Core:     update available
  [i] FTL:              up to date
  [i] Web Interface:    up to date

  [i] Pi-hole core files out of date
  [✓] Check for existing repository in /etc/.pihole
  [✓] Update repo in /etc/.pihole

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
[... 196 lines cut ...]
Uppps ... dann sollte man das fixen mit:

`root@RPi-Zero-W:~# apt purge lighttpd* && apt autoremove -y`

Danach Gegenprobe:

```
root@RPi-Zero-W:~# dietpi-services status
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | php7.0-fpm   active (running) since Mon 2018-04-02 23:45:39 CEST; 4 days ago
[  OK  ] DietPi-Services | nginx        active (running) since Mon 2018-04-02 23:45:50 CEST; 4 days ago
[  OK  ] DietPi-Services | pihole-FTL   active (running) since Sat 2018-04-07 13:08:06 CEST; 12min ago
[  OK  ] DietPi-Services | cron active (running) since Mon 2018-04-02 23:45:51 CEST; 4 days ago
[  OK  ] DietPi-Services | dropbear     active (running) since Mon 2018-04-02 23:45:16 CEST; 4 days ago
[  OK  ] DietPi-Services | dnsmasq      active (running) since Sat 2018-04-07 13:07:46 CEST; 12min ago
[  OK  ] DietPi-Services | dietpi-boot  active (exited) since Mon 2018-04-02 23:45:34 CEST; 4 days ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Mon 2018-04-02 23:45:34 CEST; 4 days ago
root@RPi-Zero-W:~#
```
Läuft ... alles schick! 😃

MichaIng, 2018-04-10:
Close in favour of https://github.com/Fourdee/DietPi/issues/1696

closed 2018-04-10 by MichaIng