MichaIng/DietPi #1626: Please add Apple Pi DAC support

orchardaudio, 2018-03-16:
## Creating a bug report/issue:

Hello,

Please add ApplePi DAC overlay to your system.

Thanks;
Leo

#### Required Information:
- DietPi Version: All Versions 
- SBC Device: All Raspberry Pi Devices
- Power supply used: 5V 3A or 5.25V 3A
- SD card used: Any SD Card
- Distro (EG: Jessie) | Jessie

#### Expected behaviour:
Add overlay so that ApplePi DAC driver can be loaded with config.txt file similar to official Raspian release.

Please see here for additional info:
https://github.com/raspberrypi/linux/issues/2302

Fourdee, 2018-03-16:
@orchardaudio 

Hi,

Thanks for the request. Overlay appears to be in current APT kernel:
```
root@DietPi:~# ls -lha /boot/overlays/ | grep apple
-rwxr-xr-x 1 root root 1.5K Mar 14 16:33 applepi-dac.dtbo
root@DietPi:~# uname -a
Linux DietPi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```

I'll add this to ```dietpi-config```.

Fourdee, 2018-03-16:
Completed, available as option in ```dietpi-config``` with DietPi v6.5. ETA 1-2 weeks release.

closed 2018-03-16 by Fourdee

orchardaudio, 2018-07-24:
Thanks for taking care of it.