MichaIng/DietPi #1627: Image update | Support RPi 3 B+

Fourdee, 2018-03-16:


Fourdee, 2018-03-16:
Completed:
http://dietpi.com/download

Fourdee, 2018-03-16:
Kernel dump/hang when enabling WiFi modules in ```dietpi-config```

Fourdee, 2018-03-18:
Other notes:
```
root@DietPi:~#  vcgencmd get_config int
arm_freq=1400
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=400
desired_osc_freq=0x33e140
desired_osc_freq_boost=0x3c45b0
disable_commandline_tags=2
disable_l2cache=1
disable_splash=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
force_eeprom_read=1
force_pwm_open=1
framebuffer_depth=16
framebuffer_height=720
framebuffer_ignore_alpha=1
framebuffer_swap=1
framebuffer_width=1280
gpu_freq=300
hdmi_force_cec_address=65535
init_uart_clock=0x2dc6c00
initial_turbo=20
lcd_framerate=60
over_voltage_avs=0x10c8e
over_voltage_avs_boost=0x2932e
over_voltage_sdram_p=2
pause_burst_frames=1
program_serial_random=1
sdram_freq=500
sdram_schmoo=0x2000020
temp_limit=75
```

xmijo, 2018-03-19:
Getting quite a few syntax errors and 'command not found' in dietpi/boot and dietpi/dietpi-cpu_set on boot. Also, "gettext.sh: No such file or directory" on login.

Using the latest 6.4 image on the new Pi 3 B+.

Fourdee, 2018-03-19:
@xmijo 

Thanks for the report 👍  
I can confirm we are able to replicate the same issues.

We are unsure the cause at the moment, however, learning towards possible kernel teething issues.

> Quite a few syntax errors and 'command not found' in the boot scripts (/DietPi/dietpi/boot and /DietPi/dietpi/dietpi-cpu_set). Prints on every boot. Also, "gettext.sh: No such file or directory" on login.

Linked to: https://github.com/Fourdee/DietPi/issues/1631#issuecomment-373965406

xmijo, 2018-03-19:
I just edited my post, I see now that you guys are already aware of the wifi issues and the boot error messages seem to have disappeared after a re-image. Not sure what caused it in the first place but seems to work fine now. But if you were able to replicate it was probably not an isolated incident..

MichaIng, 2018-03-19:
I also faced the missing gettext.sh message. Can investigate this evening.

k-plan, 2018-03-19:
@MichaIng 

> Can investigate this evening.

https://github.com/Fourdee/DietPi/commit/5fda19cdf8879e04eab72fabe0324f0144c3dc8d

Hope this will not open a new case full of worms. :smile: 

But: `Why try wifi to start and Failed to try-restart ntp.service: Unit ntp.service not found.` is still open, if you have spare time: https://github.com/Fourdee/DietPi/issues/1631#issuecomment-374240509

Fourdee, 2018-03-19:
Reference: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=207997#p1288529

xmijo, 2018-03-20:
Has anyone else experienced kernel freezing (occasionally, not every time) when issuing the reboot command? Or did I also get a faulty unit

Fourdee, 2018-03-20:
> Has anyone else experienced kernel freezing (occasionally, not every time) when issuing the reboot command? Or did I also get a faulty unit

Unsure, however, i'll re-create our DietPi image using a RPi 3 (non-plus) just to be sure.

Fourdee, 2018-03-20:
Image updated, same random instability issues for me.
1/5 reboots results in instant kernel dump. There is no consistency to it.

k-plan, 2018-03-20:
@Fourdee 

> ... i'll re-create our DietPi image using a RPi 3 (non-plus) just to be sure.

by this, can you please check wifi connection on RPi 3 (non-plus)?
Lack of hardware at the moment.

[Can't get wifi working on RPI3](http://dietpi.com/phpbb/viewtopic.php?f=11&t=2970)

[WiFi Problems (RPi 3)](http://dietpi.com/phpbb/viewtopic.php?f=11&t=2976)

Fourdee, 2018-03-20:
@k-plan 

> by this, can you please check wifi connection on RPi 3 (non-plus)?

Yep, just tested fine with v6.4 image on RPi 3 (non-plus).

Zvezke, 2018-03-22:
From reading the posts above, I understand that there are various issues - but I also understand that the latest version of DietPi should be bootable on 3B+.
I've downloaded (DietPi_v6.0_RPi-ARMv6-Stretch.img), flashed and attempted to boot, but no.
Am I missing something?

xmijo, 2018-03-23:
@Zvezke The latest version of DietPi is 6.4, not 6.0. Get the latest version from [dietpi.com/download](url)

Zvezke, 2018-03-23:
@xmijo Thanks for replying, Xmijo. I only see one option to download - choosing 'Download image' - which automatically starts downloading 'DietPi_RPi-ARMv6-Stretch.7z'. Is there a way to choose v6.4?
[image]

Fourdee, 2018-03-23:
@Zvezke 

The version number is listed inside the .7z file. Download the file and you'll see its v6.4 :)

Zvezke, 2018-03-23:
Oh, indeed it is. Sorry for the inconvenience.

Fourdee, 2018-03-24:
WiFi solution is:
```
rfkill unblock wifi
```

Once run, WiFi + scan is now functional. Seems a permanent setting on subsequent reboots, once applied.

Fourdee, 2018-03-24:
WiFi fixed: https://github.com/Fourdee/DietPi/commit/02e1f6bebb09a93c89e896fb7b606f25ee8952db

My device works fine with LAN disabled and WiFi use only. But, i will be sending it back due to fault: 
https://github.com/Fourdee/DietPi/issues/1627#issuecomment-374669173

Completed.

closed 2018-03-24 by Fourdee

xmijo, 2018-03-24:
Great news that wifi is fixed

Is the kernel hang on reboot the reason for returning your device @Fourdee ? Do the issues persist on regular Rasbian, too? (Haven't got access to my device atm so cannot test it myself)