MichaIng/DietPi #1630: DietPi-Image | DietPi-Automation < v6.2

k-plan, 2018-03-17:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 

DietPi-image < v6.2

#### Additional Information (if applicable):
- Software title?

DietPi-Automation

- Can this issue be replicated on a fresh installation of DietPi?

only on fresh image installation

#### Expected behaviour:

- Automated setup a fresh installation during first boot up

#### Actual behaviour:

- Need user interaction because of whiptail message boxes

#### Steps to reproduce:

`dietpi.txt`

```
AUTO_SETUP_AUTOMATED=1
## eg.
AUTO_SETUP_LOGGING_INDEX=-2
AUTO_SETUP_TIMEZONE=Europe/Berlin

```

#### Extra details:

- take a Dietpi image < v6.2
- write it to sd card
- edit `dietpi.txt`
```
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_LOGGING_INDEX=-2
AUTO_SETUP_TIMEZONE=Europe/Berlin
```

- power up the device with fresh written image on sd card

- ... and see what will happen (if you can eg. SBC has a HDMI output)

- it will stuck on `dietpi-update` during first boot up  running `patch_file` : 
https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L132
and
https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L232

Fourdee, 2018-03-18:
@k-plan

👍  

Confirmed, auto login results in ```G_USER_INPUTS=1```

Fourdee, 2018-03-18:
Existing code should prevent this, missing from v6.2<?
login
```
		#Start DietPi-Software if required
		G_DIETPI_INSTALL_STAGE=$(cat /DietPi/dietpi/.install_stage)
		if (( $G_DIETPI_INSTALL_STAGE == 0 )); then

			#Set non-interactive shell, if automated installation (as .bashrc run via STDIN check is interactive)
			if (( $(grep -ci -m1 '^AUTO_SETUP_AUTOMATED=1' /DietPi/dietpi.txt) )); then

				export G_USER_INPUTS=0

			fi

			/DietPi/dietpi/dietpi-software | tee "$FP_DIETPI_FIRSTRUNSETUP_LOG"

			#	reset to auto detection
			unset G_USER_INPUTS

		fi
```

Fourdee, 2018-03-18:
Test:
- 🈯️ RPi 3 > PREP > v6.0 > ```AUTO_SETUP_AUTOMATED=1``` > ```reboot```

@k-plan 
Which image/device used?

k-plan, 2018-03-18:
@Fourdee 

> Which image/device used?

```
Filename          : DietPi_v6.0_RPi-ARMv6-Stretch.img
MD5               : d269e041c5a55251573a4f5de2f0aec2
SHA1              : c55936d5cee04cd945a84aa79da3b29fecec64ae
CRC32             : 
SHA-256           : f118dffdd0dda446bc21dacf55a0a55b16c8e478bd84882a768fa49c6081c9ea
SHA-512           : 
SHA-384           : 
Full Path         : \\Fuzon\wd_1tb\#Backups\_Daniel\Projects\DietPi\DietPi_v6.0_RPi-ARMv6-Stretch.img
Modified Time     : 10/01/2018 13:14:40
```

- RPi Zero W

- `/boot/dietpi/login`

```
	#Start DietPi-Software if required
		G_DIETPI_INSTALL_STAGE=$(cat /DietPi/dietpi/.install_stage)
		if (( $G_DIETPI_INSTALL_STAGE == 0 )); then

			/DietPi/dietpi/dietpi-software | tee "$FP_DIETPI_FIRSTRUNSETUP_LOG"

		fi

	#----------------------------------------------------------------
	#DietPi finishing up bootloader.
	elif (( $G_DIETPI_INSTALL_STAGE == -1 )); then

		/DietPi/dietpi/dietpi-banner 0
		echo -e " >> DietPi System prep is nearly completed: \n Please run /DietPi/dietpi/login after a few seconds"

	#----------------------------------------------------------------

```

k-plan, 2018-03-18:
@Fourdee 

before you ask:

I want to check what is the different/problem with `DietPi_v6.4_RPi-ARMv6-Stretch.img`

(see: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2970)

Have downloaded and install `DietPi_v6.2_RPi-ARMv6-Stretch.img` at Thursday. It install and works fine for me without any problems, except of some annoying error messages. But this image file is on my computer at work ...

So can only check out at the moment image files I have here lieing around`DietPi_v6.0_RPi-ARMv6-Stretch.img`
Tried out same automated installation and noticed it will hang at some point.
But Wifi is working on RPi Zero W without problems, except of some annoying error messages: 

[DietPi-config | enable Wifi on RPi Zero W](https://github.com/Fourdee/DietPi/issues/1631)

k-plan, 2018-03-19:
@Fourdee 

- found another `DietPi_v6.0_RPi-ARMv6-Stretch.img` (I now have three different v6.0 img for RPi!)

```
==================================================
Filename          : DietPi_v6.0_RPi-ARMv6-Stretch.img
MD5               : 0ad92e00df8db0340dea59fd42a8a1c3
SHA1              : ac661ff30a48043f79cf63b14396d81a115589e9
CRC32             : 
SHA-256           : a4069a9625a7d6df99180c9520fd94dff462aefc6149c519f8abb4b1626b4c81
SHA-512           : 
SHA-384           : 
Full Path         : Z:\#Backups\_Daniel\Projects\DietPi\DietPi_v6.0_RPi-ARMv6-Stretch.img
Modified Time     : 20/01/2018 14:22:59
Created Time      : 07/01/2018 14:15:31
Entry Modified Time: 20/01/2018 14:22:59
```

If you are sure, where is no DietPi v6.0 for any device to download, < `20/01/2018`, we can close this ticket. Can't download, unpack and check all available `DietPi_v6.0_xxx-ARMxxx-xxxx.7z` for all different devices at the moment. 
Thanks.

Fourdee, 2018-03-20:
@k-plan 

> DietPi_v6.0

👍 

I've checked the current 6.0 images (Asus TB/C1/C2/Sparky), all have the updated code (https://github.com/Fourdee/DietPi/issues/1630#issuecomment-374010963).

DietPi_v6.0_RPi-ARMv6-Stretch.img must of been a very early pre-v6.0 image before we added the automation code needed. As all images the current images have been checked, i'll mark this as closed.

closed 2018-03-20 by Fourdee