MichaIng/DietPi #1631: DietPi-config | enable Wifi on RPi Zero W

k-plan, 2018-03-17:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?

```
 ───────────────────────────────────────
 DietPi     | 20:23 | Sat 17/03/18
 ───────────────────────────────────────
 v6.4       | RPi Zero W (armv6l)
 ───────────────────────────────────────
```

- Power supply used (EG: 5v 1A RAVpower)?

official RPi PSU 5.1 V / 2.5 A

- SD card used (EG: Sandisk ultra)?

SanDisk Mobile Ultra 4GB

- Distro (EG: Jessie) | ```uname -a```?
```

root@RPi-Zero-W:~# cat /proc/version
Linux version 4.9.80+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1098 Fri Mar 9 18:51:28 GMT 2018
```

Base: 
`DietPi_v6.0_RPi-ARMv6-Stretch.img ` => updated to v6.4

- Networking `eth0` via USB adapter:
```
root@RPi-Zero-W:~# lsusb
Bus 001 Device 003: ID 0bda:8153 Realtek Semiconductor Cor
```

```
root@RPi-Zero-W:~# dmesg
....
[    1.753652] usb 1-1.3: new high-speed USB device number 3 using dwc_otg
[    1.807979] systemd[1]: System time before build time, advancing clock.
[    1.884844] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=8153
[    1.887547] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    1.889970] usb 1-1.3: Product: USB 10/100/1000 LAN
[    1.892307] usb 1-1.3: Manufacturer: Realtek
[    1.894676] usb 1-1.3: SerialNumber: 000001000000
....
```

#### Extra details:

Some fluster messages appears:

--------------

- Login screen:

```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

-bash: gettext.sh: No such file or directory

[  OK  ] Root access verified.
 ───────────────────────────────────────
 DietPi     | 20:23 | Sat 17/03/18
 ───────────────────────────────────────
 v6.4       | RPi Zero W (armv6l)
 ───────────────────────────────────────
```

Why `-bash: gettext.sh: No such file or directory` ?

----------------

-  enable Wifi via `dietpi-config` : 

```
### enable WiFi (first time)

 , please wait
[ SUB1 ] DietPi-Set_Hardware > wificreds (set)
[  OK  ] wificreds set | Completed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Set_Hardware > wifimodules (enable)
[ INFO ] DietPi-Set_Hardware | Checking for pre-req APT packages: crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant

[ INFO ] DietPi-Set_Hardware | Flagged for installation: firmware-ralink
[ INFO ] DietPi-Set_Hardware | APT installation for: firmware-ralink, please wait...
[  OK  ] DietPi-Set_Hardware | G_AGI: firmware-ralink

[ INFO ] DietPi-Set_Hardware | Please wait, enabling WiFi Modules...
[  OK  ] wifimodules enable | Completed
[ INFO ] DietPi-Config | Dropping connections, please wait...
[ INFO ] DietPi-Config | Restarting networking services, please wait...
```

```
### connect with corect wificreds 

Connecting to Wifi
 , please wait
[ SUB1 ] DietPi-Set_Hardware > wificreds (set)
[  OK  ] wificreds set | Completed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Set_Hardware > wifimodules (enable)
[ INFO ] DietPi-Set_Hardware | Checking for pre-req APT packages: crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant

[ INFO ] DietPi-Set_Hardware | Flagged for installation: firmware-ralink
[ INFO ] DietPi-Set_Hardware | APT installation for: firmware-ralink, please wait...
[  OK  ] DietPi-Set_Hardware | G_AGI: firmware-ralink

[ INFO ] DietPi-Set_Hardware | Please wait, enabling WiFi Modules...
[  OK  ] wifimodules enable | Completed
[ INFO ] DietPi-Config | Dropping connections, please wait...
[ INFO ] DietPi-Config | Restarting networking services, please wait...
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb....
Sending on   LPF/wlan0/b8:27:eb....
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPREQUEST of 192.168.0.100 on wlan0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.0.100 from 192.168.0.1
DHCPACK of 192.168.0.100 from 192.168.0.1

Failed to try-restart ntp.service: Unit ntp.service not found.
invoke-rc.d: initscript ntp, action "try-restart" failed.
Unit ntp.service could not be found.

bound to 192.168.0.100 -- renewal in 3295 seconds.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process Tool | Completed
```

Why  `APT installation for: firmware-ralink, please wait..` every time?

Testing:

```
root@RPi-Zero-W:~# apt-get install firmware-ralink
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'firmware-misc-nonfree' instead of 'firmware-ralink'
firmware-misc-nonfree is already the newest version (1:20161130-3+rpt3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```

Why try wifi to start and  `Failed to try-restart ntp.service: Unit ntp.service not found.` ?

CyberWeapon, 2018-03-18:
"Why -bash: gettext.sh: No such file or directory?"
The error is because the gettext package is not installed. I had the same issue and 'apt install gettext' solved it. I started with a fresh install and didn't see the error. It showed up after I added software (Node-Red, I2C, RPI.gpio and WiringPi). The message is coming from the /etc/profile.d/wifi-country.sh file which is part of the raspberrypi-sys-mods package.

k-plan, 2018-03-18:
@CyberWeapon 

Thanks for your answer and the hint to `/etc/profile.d/wifi-country.sh` 👍 

> I started with a fresh install and didn't see the error. It showed up after I added software ...

Here it's started with a fresh install and try to configure wifi first time.
It is no additional software installed.

So let us hope @Fourdee and @MichaIng  can fix it. 😉 

As well with `firmware-ralink` and `ntp.service` error message ...

Fourdee, 2018-03-19:
1. 🈯️ resolved
2. 🈯️ resolved
3. ```Why try wifi to start and Failed to try-restart ntp.service: Unit ntp.service not found.```
@k-plan 
Cant seem to replicate this or find any reference to ntp service in our code. standard timesync mode used in dietpi-config?

```
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
DHCPREQUEST of 192.168.0.104 on eth0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.0.104 from 192.168.0.1
DHCPACK of 192.168.0.104 from 192.168.0.1
bound to 192.168.0.104 -- renewal in 626013797 seconds.
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb:db:06:0e
Sending on   LPF/wlan0/b8:27:eb:db:06:0e
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process Tool | Completed
```

k-plan, 2018-03-19:
@Fourdee 

> Cant seem to replicate this or find any reference to ntp service in our code. standard timesync mode used in dietpi-config?

:cry: 

```
root@RPi-Zero-W:~# cat /var/lib/dietpi/.dietpi_image_version
6
4
```
:wink: 

[image]

```
Connecting to WiFi
 , please wait
[ SUB1 ] DietPi-Set_Hardware > wificreds (set)
[  OK  ] wificreds set | Completed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Set_Hardware > wifimodules (enable)
[ INFO ] DietPi-Set_Hardware | Checking for pre-req APT packages: crda firmware-atheros firmware-brcm80211 iw rfkill wireless-tools wpasupplicant
[ INFO ] DietPi-Set_Hardware | Pre-req APT packages are installed
[ INFO ] DietPi-Set_Hardware | Please wait, enabling WiFi Modules...
[  OK  ] wifimodules enable | Completed
[ INFO ] DietPi-Config | Dropping connections, please wait...
[ INFO ] DietPi-Config | Restarting networking, please wait...
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb:00:00:10
Sending on   LPF/wlan0/b8:27:eb:00:00:10
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPREQUEST of 192.168.0.100 on wlan0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.0.100 from 192.168.0.1
DHCPREQUEST of 192.168.0.100 on wlan0 to 255.255.255.255 port 67
DHCPACK of 192.168.0.100 from 192.168.0.1
Failed to try-restart ntp.service: Unit ntp.service not found.
invoke-rc.d: initscript ntp, action "try-restart" failed.
Unit ntp.service could not be found.
bound to 192.168.0.100 -- renewal in 2811 seconds.
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process Tool | Completed
```

> Failed to try-restart ntp.service: Unit ntp.service not found.
> invoke-rc.d: initscript ntp, action "try-restart" failed.
> Unit ntp.service could not be found.

```
root@RPi-Zero-W:~# find / | grep invoke-rc.d
/usr/sbin/invoke-rc.d
/usr/share/bash-completion/completions/invoke-rc.d
```

SuBLiNeR, 2018-03-20:
It seems like the user is not using the internal WiFi, but is using an USB-Key maye. If not using provided Wifi by the Zero W (WH) itself, there is no problem at all with DietPi. It may be a damaged board or whatever finally.....

Fourdee, 2018-03-25:
@k-plan 

I'll mark this as closed, as unable to replicate https://github.com/Fourdee/DietPi/issues/1631#issuecomment-374284550 in v6.5 testing branch.

If problems persist with v6.5 after release, please reopen with detailed steps so we can try and replicate.

closed 2018-03-25 by Fourdee