MichaIng/DietPi #1632: no open-ssh on "DietPi_OdroidC2-ARMv8-Stretch.7z"

pgferr, 2018-03-18:
in optimezed software i found only dropbear and i have problem connecting by winscp with sftp protocol. If i use "DietPi_OdroidC2-ARMv8-Jessie.7z" is available also open-ssh. Is there any issue?

#### Required Information:
- DietPi Version 6.4
- SBC Device odroid c2
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used samsun 64
- Distro stretch

#### Additional Information (if applicable):
open ssh
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

k-plan, 2018-03-18:
@pgferr 

> If i use "DietPi_OdroidC2-ARMv8-Jessie.7z" is available also open-ssh. Is there any issue?

[image]

[image]

You have to change SSH Server choice here.
After hit < `OK` > you have to choose `Install` hit < `OK` >

[image]

pgferr, 2018-03-18:
i need php7 for moodle 3.4 and php7 is avilable only for stretch:
also if i use apache 2 i cannot see it.

open ssh is absolutely unavailable

k-plan, 2018-03-18:
@pgferr 

1. - http://dietpi.com/#download => is only `DietPi_v6.0`_**OdroidC2**`-ARMv8-`**Jessie**`.img` at the moment. So this is supported.

2. - `DietPi_v6.2_OdroidC2-ARMv8-Stretch.img` is unready beta/testing image not fully supported.

> open ssh is absolutely unavailable

open-ssh is absolutety working on Odroid C2 DietPi v6.4- Stretch.
Can see any problems:

```
 ───────────────────────────────────────
 DietPi     | 15:51 | Sun 18/03/18
 ───────────────────────────────────────
 v6.4       | Odroid C2 (aarch64)
 ───────────────────────────────────────

root@Odroid-C2-Stretch:~# uname -a
Linux Odroid-C2-Stretch 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

root@Odroid-C2-Stretch:~# cat /proc/version
Linux version 3.14.79+ (root@odroid-jessie64) (gcc version 4.9.2 (Debian/Linaro 4.9.2-10) ) #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017

root@Odroid-C2-Stretch:~# cat /etc/debian_version
9.4

root@Odroid-C2-Stretch:~# cat /DietPi/dietpi/.version
6
4

root@Odroid-C2-Stretch:~# service ssh status
● ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-03-18 15:50:48 GMT; 2min 18s ago
  Process: 498 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
 Main PID: 537 (sshd)
   CGroup: /system.slice/ssh.service
           ├─ 537 /usr/sbin/sshd -D
           ├─1839 sshd: root@pts/0
           ├─1844 -bash
           ├─2017 systemctl status ssh.service
           └─2022 pager

Mar 18 15:50:47 Odroid-C2-Stretch systemd[1]: Starting OpenBSD Secure Shell server...
Mar 18 15:50:48 Odroid-C2-Stretch sshd[537]: Server listening on 0.0.0.0 port 22.
Mar 18 15:50:48 Odroid-C2-Stretch sshd[537]: Server listening on :: port 22.
Mar 18 15:50:48 Odroid-C2-Stretch systemd[1]: Started OpenBSD Secure Shell server.
Mar 18 15:51:25 Odroid-C2-Stretch sshd[1839]: Accepted password for root from 172.16.10.102 port 50746 ssh2
Mar 18 15:51:25 Odroid-C2-Stretch sshd[1839]: pam_unix(sshd:session): session opened for user root by (uid=0)
```

[image]

Fourdee, 2018-03-23:
Marking as closed, please reopen if required.

closed 2018-03-23 by Fourdee