MichaIng/DietPi #1633: apache installation non recognized from "optimized-software"

pgferr, 2018-03-18:
i am using "DietPi_OdroidC2-ARMv8-Stretch.7z"  and i have installed lamp (maria db) server environment. But in management script there are two prpblems: apache installation is not recognized from "optimized-software" 
I can see only lighttpd and i cannot install phpmyadmin?
tks

#### Required Information:
- DietPi Version | 6.4
- Distro stretch
#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

k-plan, 2018-03-18:
@pgferr 

can you please show us your first screen of `dietpi-software` ?

example:

[image]

Fourdee, 2018-03-30:
No response, marking as closed. Please reopen if required.

closed 2018-03-30 by Fourdee