MichaIng/DietPi #1634: wordpress permalink not work

msongz, 2018-03-18:
lighthttpd

the link like : pi2/wordpress/post-name will not works
but : pi2/wordpress/index.php/post-name will work

something about mod_rewrite configuration
any help ?

MichaIng, 2018-03-20:
@msongz 

Nextcloud and ownCloud nginx configs contain a line to enable front controller which enables pretty URLs without /index.php/: https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/nginx.sites-dietpi.nextcloud.config#L52

Should be possible for Lighttpd as well, we just need to verify the right syntax. Then we can also add this e.g. to new Nextcloud/ownCloud installations.

msongz, 2018-03-20:
@MichaIng 
```
server.modules += ("mod_rewrite")
url.rewrite-once = (
"^/wordpress/(wp-.+)$" => "$0",
"^/wordpress/(.*)$" => "/wordpress/index.php/$1"
)
```
finally fix that !

MichaIng, 2018-03-20:
Thanks, I can adapt this for ownCloud + Nextcloud as well :+1:.

I will check if enabling front controller works as well, as then we could do this for several web services at once.

MichaIng, 2018-12-01:
Required for Pydio client sync: https://github.com/Fourdee/DietPi/issues/2308

MichaIng, 2018-12-02:
Same solution for Pydio works for Nextcloud as well, to reach the pages via URL without /index.php/. But what is not working is that the links inside Nextcloud WebUI open pages with pretty URLs as well.  Not sure if this is part of the front controller directive. Will delay this to v6.20. Only Pydio will be done in v6.19.

danrneal, 2020-12-31:
At the risk of necro-posting on an old issue, is it yet possible to have pretty urls with nextcloud and lighttpd?

MichaIng, 2020-12-31:
The rewrites work fine, but the Nextcloud internal URL generator still generates "ugly" URLs, i.e. all links on the website still contain the PHP scripts in path. For a different reason I had a look into the code section a few days ago, so probably I'll manage to understand what is required. But it's no high priority from my end.

danrneal, 2021-01-01:
Interesting that that nextcloud internal url genenrator would only affect lighttpd and not nginx. :shrug: It's not that big of a deal since nginx works out of the box.

MichaIng, 2021-01-01:
Yes it must be some variable missing or being set differently in Lighttpd compared to Apache2 and Nginx. However, aside of having one additional path element in URLs, it is all functional.