MichaIng/DietPi #1636: DietPi-Config | Testing - v6.5 Layout

k-plan, 2018-03-18:
Layout `dietpi-config` whiptail menu:

Dietpi v6.4:

[image]

Dietpi v6.5:

[image]

Fourdee, 2018-03-19:
@k-plan 

Due to roll out of G_WHIP in DietPi code. This automatically scales window based on screen dimensions against text lines and optional content: https://github.com/Fourdee/DietPi/issues/1546#issuecomment-370173876

#### Completed:
- DietPi-Software
- DietPi-Cron

#### Started
- DietPi-Config

k-plan, 2018-03-19:
@Fourdee 

> Allows us to remove the hard defined whiptail scaling.
> - DietPi-Config
>
> I'am rolling this out gradually in the code, when possible.

Ahh ... okay. Make sense. :+1: 

A little side note:
 - it will looks not so puzzling, if you will centered the first line ( `Hardware : RPi Zero W` ) as well, so it will more look like the old whiptail dialogue before.

Can we close this ticket?

Fourdee, 2018-03-19:
@k-plan 

> it will looks not so puzzling, if you will centred the first line ( Hardware : RPi Zero W ) as well, so it will look more like the old whiptail dialogue before.

Can't centre the text in whiptail, we'd need to resize X based on max string length per line.

Would be nice to have implemented, leave ticket open for now, i'll see if I can get round to it.

MichaIng, 2018-03-20:
If dialog size and text length is known, we could calculate how much space we need to add to beginning of header string to make it look aligned to centre.

Or set max dialog length based on header/text length.

Fourdee, 2018-03-20:
@MichaIng 

> If dialog size and text length is known, we could calculate how much space we need to add to beginning of header string to make it look aligned to centre.

👍 

My only concern is the processing required to check each line of text for its length, and, all the menu/checklist entries (eg: dietpi-software 100+ ). I feel the tradeoff may overall slow the whiptail loading times.

MichaIng, 2018-03-26:
@Fourdee
Yeah didn't see that the "header" in G_WHIP_MENU is message body, which can be of several lines of course. We would need to handle \n the as well... Might be better than to solve it by G_WHIP_SIZE_X_MAX in urgent cases then.

Fourdee, 2018-04-01:
> Might be better than to solve it by G_WHIP_SIZE_X_MAX in urgent cases then.

Agree. I've set a limit for all exit prompts.
[image]

Fourdee, 2018-04-01:
I'll leave the remaining whiptail menus as automatic X max size for now. Yes, we could hard limit them all, however, should we send a commit that exceeds limits, adds additional checks that defeats the purpose of G_WHIP (to make whiptail + dev simple and effective)

Automatic X scaling based on input text contents is also a no go, due to the additional processing required to check each line of text.

I'll mark this as closed.

closed 2018-04-01 by Fourdee