MichaIng/DietPi #1638: NoMachine is currupt

Gokujo, 2018-03-19:
#### Required Information:
- DietPi Version | 6.4?
- SBC Device RPi 3
- Distro | ```Linux DietPi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux```

I tried to install NoMachine, but it failed.
[image]

k-plan, 2018-03-19:
@Gokujo 

> `wget: unable to resolve host address 'dietpi.com'`

Your device has a DNS or network connection problem.

`nomachine_6.0.78_1_armhf.deb` is online and fine:

```
root@Odroid-C2-Stretch:~# wget http://dietpi.com/downloads/binaries/all/nomachine_6.0.78_1_armhf.deb
--2018-03-19 12:36:43--  http://dietpi.com/downloads/binaries/all/nomachine_6.0.78_1_armhf.deb
Resolving dietpi.com (dietpi.com)... 185.101.92.145
Connecting to dietpi.com (dietpi.com)|185.101.92.145|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 39773450 (38M) [application/x-debian-package]
Saving to: ‘nomachine_6.0.78_1_armhf.deb’

nomachine_6.0.78_1_armhf.deb 100%[==============================================>]  37.93M  8.74MB/s    in 4.4s

2018-03-19 12:36:47 (8.68 MB/s) - ‘nomachine_6.0.78_1_armhf.deb’ saved [39773450/39773450]

```

Please check you DNS or internet connection from your device for example by:

[root@Odroid-C2-Stretch:~# ping dietpi.com
PING dietpi.com (185.101.92.145) 56(84) bytes of data.
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=1 ttl=54 time=13.3 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=2 ttl=54 time=13.3 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=3 ttl=54 time=13.4 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=4 ttl=54 time=13.2 ms
64 bytes from 185.101.92.145 (185.101.92.145): icmp_seq=5 ttl=54 time=13.4 ms
^C
--- dietpi.com ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4006ms
rtt min/avg/max/mdev = 13.213/13.357/13.481/0.155 ms
](url)

Gokujo, 2018-03-19:
@Fourdee some mystery... it works now... there are no problems in the line

Fourdee, 2018-03-19:
@Gokujo 

Thanks for the report 👍 

If not DNS, we could also put this down to possible issue with our host provider during install. However, seems fine now and i'll mark this as resolved.

closed 2018-03-19 by Fourdee