MichaIng/DietPi #1640: WIfi Hotspot bug

symbios24, 2018-03-19:
Hi, i used latest DietPi 6.4 and installed the wifi hotspot software
after installation has finished and rebooted i see that the start: isc-dhcp-server is failed to start and the hotspot is not created,also when i go to the network options to change the country code from gb to us it tells me that country code cannot be changed.

symbios24, 2018-03-19:
there was also an error upon isc-dhcp-server installation an exiting code

Fourdee, 2018-03-19:
@symbios24 

Hi,

When creating a ticket, a template will be created that allows you to fill out information that we can use to assist. Without this information in the 1st instance, it will involve a lot of back tracking to find a probable cause.

[image]

Please create a new ticket and fill out the requested information as best as you can. Then we can assist.

closed 2018-03-19 by Fourdee

symbios24, 2018-03-19:
did you try to install hotspot your self to see if it works? neither my card or power supply have problems

symbios24, 2018-03-19:
did you try to install hotspot your self to see if it works? neither my card or power supply have problems installing hotspot on older dietpi version working fine,its v6.4 that has the problem

Fourdee, 2018-03-20:
@symbios24 

> did you try to install hotspot your self to see if it works? neither my card or power supply have problems installing hotspot on older dietpi version working fine,its v6.4 that has the problem

Installed WiFi hotspot on RPi 3. Installs and runs fine.

As explained here (https://github.com/Fourdee/DietPi/issues/1640#issuecomment-374256688) we'll need the required information to assist more efficiently.