MichaIng/DietPi #1643: Disable swap during update to 6.4

thiagobucca, 2018-03-20:
Hi,

I'm trying to update from 6.3 to 6.4 but i'm having some issues because of the size of my SD Card (2GB). I only have 506MB left on my SD Card, and the update process fails during swap partition creation. Is there a way to disable the swap during the update so i can update my build without moving to a bigger SD Card?

Thanks in advance!

#### Required Information:
- DietPi Version | ```6.3```?
- SBC Device: Rpi Zero W
- Power supply used: 5v 2a
- SD card used: 2GB Multilaser
- Distro (EG: Jessie) | ```4.9.80+ #1098```?

bopnrgi, 2018-03-21:
Same problem here, my swap is disabled but update fail and crash the card

Fourdee, 2018-03-21:
Hi,

DietPi requires a 4GB minimum SD card, to ensure a stable experience and sufficient swapfile based on total memory: http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

Previously, it was possible to set swapfile size to 0 in dietpi.txt, however, due to previous issues with swapfile generation on some images, and, issues with insufficient ```/tmp``` tmpfs size, we had to forcefully recreate them for all devices in v6.4 and v6.5 patch, overwriting any previous setting.

Once patched to v6.5, you can manually disable the swapfile. v6.5 should be released next week.

I'll mark this as closed. Please reopen if required.

closed 2018-03-21 by Fourdee

thiagobucca, 2018-03-21:
Hi 

@Fourdee Sorry to reopen this, just one more question:

If i rebuild all my system directly on top of v6.5, i will be able to update to further versions with my 2GB card by disabling swap in dietpi-config?

Thank you