MichaIng/DietPi #1644: Ramdisk image is corrupt or invalid after reboot

enwi, 2018-03-21:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.2
- SBC Device Odroid HC1
- Power supply 5V 5A
- SD card used Londisk extreme 16GB
- Distro jessie

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi?
Not sure

#### Expected behaviour:
Device should start normally after reboot.

#### Actual behaviour:
Device failing to reboot due to "Ramdisk image is corrupt or invalid" after "Verifying Checksum ... Bad Data CRC"

#### Steps to reproduce:
update software using apt-get and reboot.

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
I exported the rootfs to the external harddrive and made some changes in order to boot from said external harddrive.
Output of serial interface:
```
U-Boot 2017.05-15377-gedb23d4 (Aug 24 2017 - 07:09:51 -0300) for ODROID-XU4

CPU:   Exynos5422 @ 800 MHz
Model: Odroid XU4 based on EXYNOS5422
Board: Odroid XU4 based on EXYNOS5422
Type:  xu4
DRAM:  2 GiB
MMC:   EXYNOS DWMMC: 0, EXYNOS DWMMC: 1
MMC Device 0 ( SD ): 14.5 GiB
Card did not respond to voltage select!
mmc_init: -95, time 11
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Press quickly 'Enter' twice to stop autoboot:  0
reading boot.ini
5007 bytes read in 5 ms (977.5 KiB/s)
cfgload: applying boot.ini...
cfgload: setenv initrd_high "0xffffffff"
cfgload: setenv fdt_high "0xffffffff"
cfgload: setenv macaddr "00:1e:06:61:7a:72"
cfgload: setenv bootrootfs "console=tty1 root=UUID=6ffc01d8-06e2-48e6-8b9f-d3da9                                                                            49cb4e5 rootwait rootdelay=10 ro fsck.repair=yes"
cfgload: setenv videoconfig "drm_kms_helper.edid_firmware=edid/1920x1080.bin"
cfgload: setenv vout "hdmi"
cfgload: setenv governor "performance"
cfgload: setenv HPD "true"
cfgload: setenv hdmi_tx_amp_lvl  "31"
cfgload: setenv hdmi_tx_lvl_ch0      "3"
cfgload: setenv hdmi_tx_lvl_ch1      "3"
cfgload: setenv hdmi_tx_lvl_ch2      "3"
cfgload: setenv hdmi_tx_emp_lvl      "6"
cfgload: setenv hdmi_clk_amp_lvl     "31"
cfgload: setenv hdmi_tx_res      "0"
cfgload: setenv hdmi_phy_control "hdmi_tx_amp_lvl=${hdmi_tx_amp_lvl} hdmi_tx_lvl                                                                            _ch0=${hdmi_tx_lvl_ch0} hdmi_tx_lvl_ch1=${hdmi_tx_lvl_ch1} hdmi_tx_lvl_ch2=${hdm                                                                            i_tx_lvl_ch2} hdmi_tx_emp_lvl=${hdmi_tx_emp_lvl} hdmi_clk_amp_lvl=${hdmi_clk_amp                                                                            _lvl} hdmi_tx_res=${hdmi_tx_res} HPD=${HPD} vout=${vout}"
cfgload: fatload mmc 0:1 0x40008000 zImage
reading zImage
3335856 bytes read in 1353 ms (2.4 MiB/s)
cfgload: fatload mmc 0:1 0x42000000 uInitrd
reading uInitrd
4209045 bytes read in 735 ms (5.5 MiB/s)
cfgload: fatload mmc 0:1 0x44000000 exynos5422-odroidxu3.dtb
reading exynos5422-odroidxu3.dtb
52226 bytes read in 9 ms (5.5 MiB/s)
cfgload: fdt addr 0x44000000
cfgload: if test "${cecenable}" = "false"; then fdt rm /cec@101B0000; fi
cfgload: if test "${disable_vu7}" = "false"; then setenv hid_quirks "usbhid.quir                                                                            ks=0x0eef:0x0005:0x0004"; fi
cfgload: setenv bootargs "${bootrootfs} ${videoconfig} smsc95xx.macaddr=${macadd                                                                            r} governor=${governor} ${hdmi_phy_control} ${hid_quirks}"
cfgload: bootz 0x40008000 0x42000000 0x44000000
Kernel image @ 0x40008000 [ 0x000000 - 0x32e6b0 ]
## Loading init Ramdisk from Legacy Image at 42000000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    4208981 Bytes = 4 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... Bad Data CRC
Ramdisk image is corrupt or invalid
mmc block read, dev 0, addr 0x40008000, blk start 2047, blk cnt 16384
Bad Linux ARM zImage magic!
```

Fourdee, 2018-03-21:
@enwi 

Thanks for the report 👍 

I suspect filesystem corruption due to system instability, however, we'll run a local test on latest kernel to verify stable.

enwi, 2018-03-22:
@Fourdee  Which I've already done. Thankfully I had the rootfs on the external drive, so I just needed to reinstall dietpi on the micro sd to get a working init ramdisk and copy my boot.ini from the old installation to the new one. So now everything is up and running again!

Fourdee, 2018-03-22:
@enwi 

Confirmed working HC1 here with latest kernel upgrade. I'll mark this as closed due to potential hardware issue.

closed 2018-03-22 by Fourdee