MichaIng/DietPi #1646: Diet-pi issue with attempted restore from backup

microbiomeparty, 2018-03-22:
#### Required Information:
- DietPi Version | v6.2
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | rsync -alHpvP --delete --log-file=/var/log/dietpi-backup.log /mnt/dietpi-backup/dietpi-backup_userdata/ /mnt/kranghole/dietpi_userdata/
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Backup

#### Expected behaviour:

Was attempting restore from backup with dietpi-backup.

#### Actual behaviour:

hit an error and did not complete.

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
nextcloud_data/appdata_octwabhnjywq/preview/423/
nextcloud_data/appdata_octwabhnjywq/preview/425/
nextcloud_data/appdata_octwabhnjywq/preview/428/
nextcloud_data/appdata_octwabhnjywq/preview/429/
nextcloud_data/appdata_octwabhnjywq/preview/430/
nextcloud_data/appdata_octwabhnjywq/preview/431/
nextcloud_data/appdata_octwabhnjywq/preview/432/
nextcloud_data/appdata_octwabhnjywq/preview/433/
nextcloud_data/appdata_octwabhnjywq/preview/434/
nextcloud_data/appdata_octwabhnjywq/preview/435/
nextcloud_data/appdata_octwabhnjywq/preview/436/
nextcloud_data/appdata_octwabhnjywq/preview/437/
nextcloud_data/appdata_octwabhnjywq/preview/5/
nextcloud_data/appdata_octwabhnjywq/preview/510/
nextcloud_data/appdata_octwabhnjywq/preview/511/
nextcloud_data/appdata_octwabhnjywq/preview/512/
nextcloud_data/appdata_octwabhnjywq/preview/513/
nextcloud_data/appdata_octwabhnjywq/preview/514/
nextcloud_data/appdata_octwabhnjywq/preview/516/
nextcloud_data/appdata_octwabhnjywq/preview/517/
nextcloud_data/appdata_octwabhnjywq/preview/518/
nextcloud_data/appdata_octwabhnjywq/preview/519/
nextcloud_data/appdata_octwabhnjywq/preview/520/
nextcloud_data/appdata_octwabhnjywq/preview/526/
nextcloud_data/appdata_octwabhnjywq/preview/529/
nextcloud_data/appdata_octwabhnjywq/preview/546/
nextcloud_data/appdata_octwabhnjywq/preview/6/
nextcloud_data/appdata_octwabhnjywq/preview/633/
nextcloud_data/appdata_octwabhnjywq/preview/634/
nextcloud_data/appdata_octwabhnjywq/preview/635/
nextcloud_data/appdata_octwabhnjywq/preview/636/
nextcloud_data/appdata_octwabhnjywq/preview/637/
nextcloud_data/appdata_octwabhnjywq/preview/638/
nextcloud_data/appdata_octwabhnjywq/preview/639/
nextcloud_data/appdata_octwabhnjywq/preview/641/
nextcloud_data/appdata_octwabhnjywq/preview/642/
nextcloud_data/appdata_octwabhnjywq/preview/643/
nextcloud_data/appdata_octwabhnjywq/preview/644/
nextcloud_data/appdata_octwabhnjywq/preview/645/
nextcloud_data/appdata_octwabhnjywq/preview/646/
nextcloud_data/appdata_octwabhnjywq/preview/647/
nextcloud_data/appdata_octwabhnjywq/preview/650/
nextcloud_data/appdata_octwabhnjywq/preview/651/
nextcloud_data/appdata_octwabhnjywq/preview/652/
nextcloud_data/appdata_octwabhnjywq/preview/7/
nextcloud_data/files_external/

sent 181,606,905 bytes  received 2,321 bytes  11,006,619.76 bytes/sec
total size is 1,404,307,297  speedup is 7.73
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
```

microbiomeparty, 2018-03-22:
Also, with attempted repeat of dietpi-backup restore, got this error-handler message:
[image]

Fourdee, 2018-03-22:
@microbiomeparty 

Hi,

> rsync error: some files/attrs were not transferred (see previous errors) (code 23) 

I suspect filesystem limitation on ```/mnt/kranghole```, which filesystem type is on this disk?

microbiomeparty, 2018-03-22:
Thanks for the quick reply, it is a 4 terabyte western digital external drive formatted as ext4

Fourdee, 2018-03-22:
@microbiomeparty 

Ok, disable swapfile:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 0 /var/swap
```

And lets find the swapfile it mentions:
```
find / -name .swapfile
```
It should be safe to remove it, if unsure post results here.

Then rerun the restore. Use ```dietpi-config``` > advanced > to setup a new swapfile.

microbiomeparty, 2018-03-23:
thanks so much for the help, after a hard shutdown (which may have cleared the swap file), the restore worked.

Fourdee, 2018-03-23:
@microbiomeparty 

Great to hear 👍 

I'll mark this as closed. Please reopen if required.

closed 2018-03-23 by Fourdee