MichaIng/DietPi #1647: WiringPi not available on BananaPi M3 over prep

Elluminatus, 2018-03-23:
## Creating a bug report/issue:
The Programm wiringPi and other bananaPi spezific programms  are not available
over the software installer when you install dietpi on a banana pi m3 via the prep script.

#### Required Information:
- DietPi Version | 6.4.
- SBC Device (Banana Pi M3)
- Power supply used (5v 2,5A RAVpower)?
- SD card used (transend 8gb)?
- Distro (stretch)

#### Additional Information (if applicable):
- Software title? wiringpi
- Can this issue be replicated on a fresh installation of dietpi: yes

Fourdee, 2018-03-23:
@Elluminatus 

Hi,

#### Currently, WiringPi via ```dietpi-software``` is only supported for the following devices:
- RPi
- Odroids (all)
- BPi Pro

You can try following the install guide here:
http://forum.banana-pi.org/t/how-to-banana-pi-bpi-m3-install-bpi-wiringpi/1056

As WiringPi is a specific compile per device, we simply cant offer support for them all.
As such, i'll mark this as closed. Please reopen if required.

closed 2018-03-23 by Fourdee

Elluminatus, 2018-03-23:
After installing

- apt-get  install make
- apt-get install gcc
- apt-get install build-essentials

it worked for me. Thank you