MichaIng/DietPi #1648: question: move system from as card to emmc by dietpi

Elluminatus, 2018-03-23:
Hi when you own a system like a sbc odroid or a banana pi with a build in emmc it might
be useful to move the whole system from da card to emmc with a dietpi config entry like
mobbing userdata to an other harddisk.

Is this possible?

Greetings Elluminatus

SuBLiNeR, 2018-03-25:
Good question, i tried this already with the "dd" command but it won't boot afterwards, as there must be some different uBoot settings to be done i guess.
Maybe, we'll get an unified solution for this to work. 
@Fourdee : if i setup DietPi from an eMMC loaded system (Jessie), i always get an fail because of missing D-BUS services or similar. Why?

MichaIng, 2018-03-25:
Ref: https://askubuntu.com/questions/3402/how-to-move-boot-and-root-partitions-to-another-drive

Jep, the bootloader needs adjustments after `dd` root + boot partition onto the other drive. The procedure depends on which (kind of) bootloader is used, grub or uboot or RPi or whatever, so there is no unified solution as far as I can see, it depends on the machine/system.

Fourdee, 2018-04-13:
Marking as closed, please reopen if required.

closed 2018-04-13 by Fourdee