MichaIng/DietPi #1651: [REQUEST] Banana Pi M2 Berry - Support / Help with PREP-Script

SuBLiNeR, 2018-03-28:
Hallo together,

yesterday i got a BananaPi M2 Berry gifted and start playing around with it. I found some different based Debian Stretch images, which depend on an old kernel (Guess, yes i know support by BPi is shit), but on mine search i found some images (https://drive.google.com/drive/folders/0B_YnvHgh2rwjR0JsaUltalFXanc) which work quite well.
For myself i thought, i may able to convert this Stretch image directly to DietPi as i've done it with my OrangePi Zero (Kernel 4.14.y). 

But nadda, the PREP-script always fails at the "linux-image-sunx8i" part as it says there is no installation path or similar.

Second try of myself was than to get up and running this SBC with Raspbian Stretch, which even works OOB with no problem. It even runs the PREP-Script, but afterwards isn't able to boot anymore.
As i'm not familar with uBoot-stuff i thought someone here may able to help maybe?

I think there are more ppl like me who are willing to run this SBC with DietPi more than the full bloated images by BPi.

Fourdee, 2018-03-28:
@SuBLiNeR 

Hi,

> But nadda, the PREP-script always fails at the "linux-image-sunx8i" part as it says there is no installation path or similar.

Any chance you could copy/paste the whole terminal output during PREP? With that info, would should be able to debug it.

> but afterwards isn't able to boot anymore

Most likely a kernel package/dep being removed thats required for basic boot. For this, we'll need to see the full list of packages, prior to running PREP:
```
dpkg --get-selections
```

SuBLiNeR, 2018-03-28:
Hold on please, just preparing another image to just let it re-run.

Output fo dpkg --get-selections: https://pastebin.com/zLVfc2YA 

Do you need the output of failed message, too? I used this on a reduced debian image. If i do this on a modified Raspbian installation, the whole PREP runs well, but device won't boot afterwards... just saying.

SuBLiNeR, 2018-04-01:
@Fourdee Do you need anything else for this task being done? I really like to have a DietPi on the BPi M2 Berry than using only a Debian Stretch Lite image....

Fourdee, 2018-04-01:
@SuBLiNeR 

> E: Package 'linux-image-sun8i' has no installation candidate

Think I know what needs to be done, as package isn't available in any of the repos on system, well need to apt-mark the package instead of install it. We'll need to do this in the script.

1 min i'll sent commit and instructions.

Fourdee, 2018-04-01:
@SuBLiNeR 

Done, try:
```
GIT_BRANCH='testing'

GIT_BRANCH=$GIT_BRANCH wget https://raw.githubusercontent.com/Fourdee/DietPi/$GIT_BRANCH/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
sudo GIT_BRANCH=$GIT_BRANCH ./PREP_SYSTEM_FOR_DIETPI.sh
```

SuBLiNeR, 2018-04-01:
@Fourdee Hi there... i hope that's not an April's fool... Gonna try it by tomorrow morning. I'll head back to you if it works! :100:

SuBLiNeR, 2018-04-02:
@Fourdee Hi Fourdee, hell yes, it is now going to install without any error, but the result after 1st reboot is, that i'm falling into a reboot-loop after the first start. The device is booting up and than stops somehow the CPUs to there is no coming even through to boot up normally. I think there is some uBoot-Loader settings in uENV.txt or whatever, which is blocking to boot the system itself.

At this point i'm on my end of knowledge how to get past this. Maybe one of you have some special knowledge or may have an idea on how to get it working at least.

Fourdee, 2018-04-02:
@SuBLiNeR 

> into a reboot-loop after the first start

Gutted.
As I lack the device to try this myself, we'd need to see the serial console output to debug in this case.

SuBLiNeR, 2018-04-02:
hell yes, that's what i'm missing... just have to get one. as of this, i think we do this on hold!

Fourdee, 2018-04-05:
@SuBLiNeR 

I'll mark this as closed + known issue for now. Should I be able to get my hands on the device, or you can run serial debugging, we'll reopen and investigate further.

closed 2018-04-05 by Fourdee