MichaIng/DietPi #1653: RetroPie | Pre-made image

Fourdee, 2018-03-28:
RPi 2/3 (ARMv7).

+ snes9x
+ dosbox
+ Support for dietpi-userdata dir
+ samba?

Fourdee, 2018-03-30:
## Install and configure RetroPie on a DietPi RPi system.

#### Initial setup/configuration
```
#Install Pre-reqs
G_AGI git build-essential

#Install ALSA + Xorg
dietpi-software install 5 6

#Set GPU mem
/DietPi/dietpi/func/dietpi-set_hardware gpumemsplit 192

#Autostart RetroPie on boot
echo 3 > /DietPi/dietpi/.dietpi-autostart_index
```

#### Run RetroPie setup
```
git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
cd RetroPie-Setup
./retropie_setup.sh
```
- Say ```Yes``` to all prompts regarding ```root``` not having access to ```video``` (which is false).
- Select ```Basic Install``` and confirm its installation
- Once completed, exit the RetroPie menu

#### Move RetroPie installation to ```/mnt/dietpi_userdata/RetroPie```
This will allow easy transfer of roms, should Samba/ProFTP server be installed
```
cd "$HOME"
mv RetroPie /mnt/dietpi_userdata/
ln -sf /mnt/dietpi_userdata/RetroPie RetroPie
```

Fourdee, 2018-03-30:
As much as i'd love to have a pre-made DietPi + RetroPie image, it simply adds further maintenance our end, which I lack time for.
As far as I can tell, RetroPie does not support automated installs, if it did, we could of added this back into ```dietpi-software```
As such, i've changed this to a guide.

closed 2018-03-30 by Fourdee

actraiser30, 2019-11-21:
retropie boots way faster in dietpi image,it would be great to add it back to dietpi

actraiser30, 2019-12-03:
after i installed retropie is it safe to remove the installed Build-essential + git ??

MichaIng, 2019-12-03:
@actraiser30 
As long as there is no internal updater that uses Git, yes.

N7-BADA, 2020-08-03:
> ## Install and configure RetroPie on a DietPi RPi system.
> ### Initial setup/configuration
> We recommend doing this on a fresh installation, with no changes to the initial `dietpi-software` screen.
> 
> ```
> #Install ALSA + Xorg + Build-essential + git
> dietpi-software install 5 6 16 17
> 
> #Set GPU mem
> /DietPi/dietpi/func/dietpi-set_hardware gpumemsplit 192
> 
> #Autostart RetroPie on boot
> echo 3 > /DietPi/dietpi/.dietpi-autostart_index
> ```
> 
> ### Run RetroPie setup
> ```
> git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
> cd RetroPie-Setup
> ./retropie_setup.sh
> ```
> 
> * Say `Yes` to all prompts regarding `root` not having access to `video` (which is false).
> * Select `Basic Install` and confirm its installation
> * Once completed, exit the RetroPie menu
> 
> ### Move RetroPie installation to `/mnt/dietpi_userdata/RetroPie`
> This will allow easy transfer of roms, should Samba/ProFTP server be installed, and, also allow for moving of user data with `dietpi-drive_manager` to a USB drive etc.
> 
> ```
> cd "$HOME"
> mv RetroPie /mnt/dietpi_userdata/
> ln -sf /mnt/dietpi_userdata/RetroPie RetroPie
> chown -R dietpi:dietpi /mnt/dietpi_userdata/RetroPie
> ```
> 
> ### Optional: Install Fileserver to transfer roms
> Installing a fileserver will allow you to transfer roms to the device.
> 
> * http://dietpi.com/phpbb/viewtopic.php?f=8&t=15#p19
> 
> ### Completed
> * Simply reboot the system to launch RetroPie

tried following the guide but found out /DietPi/ doesnt exist anymore (at least for me). is it now in /boot/ like the following?

```
#Set GPU mem
/boot/dietpi/func/dietpi-set_hardware gpumemsplit 192

#Autostart RetroPie on boot
echo 3 > /boot/dietpi/.dietpi-autostart_index
```

Joulinar, 2020-08-03:
Yes it's moved to `/boot` recently

N7-BADA, 2020-08-03:
> Yes it's moved to `/boot` recently

thanks, i installed btw and works perfect!

MichaIng, 2020-08-03:
#### Background
`/DietPi` was a tmpfs (RAM disk) where DietPi scripts and configs from `/boot/` where copied to on boot to speed up their parsing and execution. However was kinda fragile as any changes were lost if copying the content back to disk (on shutdown) failed, e.g. on poweroff. And the path was empty if the service that should copy /boot/dietpi content to /DietPi tmpfs failed. And since all changes to those files are intended to be persistent, we switched to native file system caching method to gain the same read/execute performance.

I updated the post above 🙂.

rostomzer, 2020-10-30:
i'm trying this on virtualbox, installed everything from scratch, then installed retropie, but when rebooting, retropie won't startup! and i stay at the console
i tried to instal a full desktop environement and start retropie from start menu and it worked
then retry to set it up to boot first then i get the same error
just to resume, retropie wont start from boot/console. it only works when a desktop environement is active
what should i do ?

MichaIng, 2020-10-30:
Please have a look here: https://github.com/MichaIng/DietPi/issues/3790
```
xinit /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
seems to work from console.

symbios24, 2021-03-11:
The command : echo 3 > /DietPi/dietpi/.dietpi-autostart_index
Does not seem to work anymore in my RaspberryPi and there in not Retropie to select anymore in the dietpi autostart options menu in Gaming/Emulation
Dietpi stays in the console

Joulinar, 2021-03-11:
Hi,

path `/DietPi/dietpi/` did not exist anymore. Content was moved to `/boot/dietpi/` https://github.com/MichaIng/DietPi/issues/1653#issuecomment-667863067

symbios24, 2021-03-11:
I tried in boot also nothing just the same error

MichaIng, 2021-03-11:
We removed the RetroPie autostart option since it did not work anymore, see the above posts and linked issues.

When someone finds time, we can look into it if we can re-add RetroPie as install option, and then tied to it the autostart option.

Please refer to the RetroPie docs about how to boot into it automatically.

tavuntu, 2021-03-12:
Works like charm, thanks!

rostomzer, 2021-03-12:
> Works like charm, thanks!

How?

MichaIng, 2021-03-12:
Answered above last year?
From console outside of a desktop session:
```
xinit /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
From within a desktop session:
```
/opt/retropie/supplementary/emulationstation/emulationstation.sh
```
If you run it as non-root user from console, you might need to add it to the render and/or video group: `usermod -aG video,render $USER`
But that doesn't always work either, so in such case you need to launch it from within a desktop session.

Or are you asking explicitly how to launch it automatically at boot?

rostomzer, 2021-03-12:
ah. yes yes. i thought he found a new way finally its the same thing
outside desktop it works only with root.
other users returns errors even when using `usermod -aG` 
so its working only from desktop

MichaIng, 2021-03-12:
> outside desktop it works only with root.

There is also a way to start an X application via LightDM as non-root user. It technically works like a desktop login, but can be done automatically. This is how I plan to implement a generic autostart option for X applications, as all other ways either have bad security impact or do not work reliably with all GPUs/firmwares. I'll keep you updated once dealing with this topic.

rostomzer, 2021-03-12:
if you mean like this ?
[image]

then i allready tried itm it starts fine but no gamepads (even the keyboard) are detected

MichaIng, 2021-03-13:
Ah yes, LightDM can be configured to automatically login, skipping the mask, but I guess the result would be the same.

tavuntu, 2021-03-18:
> > Works like charm, thanks!
> 
> How?

Sorry, I meant only the the minimum required to run games (retroarch and a few cores), I'm not installing emulation station since I don't need it ( and I don't like it).

lrrosa, 2021-07-04:
Xorg is no longer necessary for Emulationstation and it also needs to run as a regular user, not root.

So, retropie should be downloaded as a regular user, and then "retropie_setup.sh" must be run with "sudo".

I tried to autostart emulationstation, editing /opt/retropie/configs/all/autostart.sh and replacing the first line with this:
su -c "emulationstation" -s /bin/sh dietpi
But when I try to run a game, some phyton scripts complain about being "unable to open /dev/tty" and I get stuck.
Is there any other way to autostart Emulationstation?

If I run it after login, everything works fine.

ravenclaw900, 2021-07-04:
Try running `usermod -aG tty emulationstation`.

lrrosa, 2021-07-05:
> Try running `usermod -aG tty emulationstation`.

The user is "dietpi", but it still didn't work.
What worked for me was to add emulationstation in .profile of user dietpi and autologin.

MichaIng, 2021-07-06:
Hmm strange, if it starts from `.profile` then it should start manually from console as well. Probably a fresh shell was required for the group membership to take effect.

tavuntu, 2021-07-13:
So everything was working properly until 7.2 (I think) now I open retroarch (with a game as parameter) and it just freezes (the whole system, so I have to disconnect the power and start it again). not sure if related to @lrrosa 's issue tho. [This is the script](https://github.com/tavuntu/rucopie/blob/main/root/RucoPie/scripts/run_game.sh) I was using to run the games and it was working ok before, no changes have been made to the repo code.

MichaIng, 2021-07-14:
DietPi v7.3 did not change anything in this regards either, apart of the APT package upgrades, of course.

Any logs or error messages? In case it invokes X, check also the X server logs (from the home of the invoking user).

tavuntu, 2021-07-15:
I'll check the logs and come back here later.

tavuntu, 2021-08-11:
Better late than never, here's the logs for retroarch, not sure if they contain anything useful:
```
[INFO] RetroArch 1.9.7 (Git 8373ac6)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Jul 29 2021
[INFO] Version: 1.9.7
[INFO] Git: 8373ac6
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/opt/retropie/libretrocores/lr-gambatte/gambatte_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/root/.config/retroarch/config/Gambatte/Gambatte.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/root/.config/retroarch/config/Gambatte/USA.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/root/.config/retroarch/config/Gambatte/1942 (USA, Europe).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] Core requested VFS version >= v1, providing v3
[INFO] [Remaps]: Remap directory: "/root/.config/retroarch/config/remaps".
[INFO] [Overrides]: Redirecting save file to "/root/RetroPie/roms/gbc/USA/1942 (USA, Europe).srm".
[INFO] [Overrides]: Redirecting save state to "/root/RetroPie/roms/gbc/USA/1942 (USA, Europe).state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 4.
[INFO] [CONTENT LOAD]: Loading content file: /root/RetroPie/roms/gbc/USA/1942 (USA, Europe).zip#1942 (USA, Europe).gbc
[INFO] Did not find a valid content patch.
[INFO] [CONTENT LOAD]: CRC32: 0x87431672
[INFO] [Environ]: GET_CAN_DUPE: true
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[libretro INFO] Rumble environment supported.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[WARN] [Environ]: SYSTEM DIR is empty, assume CONTENT DIR /root/RetroPie/roms/gbc/USA/1942 (USA, Europe).zip
[INFO] [Environ]: SYSTEM_DIRECTORY: "".
[WARN] [Environ]: SYSTEM DIR is empty, assume CONTENT DIR /root/RetroPie/roms/gbc/USA/1942 (USA, Europe).zip
[INFO] [Environ]: SYSTEM_DIRECTORY: "".
[libretro INFO] [Gambatte]: Got internal game name: 1942.
[INFO] [Environ]: SET_MEMORY_MAPS.
[INFO]    ndx flags  ptr          offset   start    select   disconn  len      addrspace
[INFO]    001 M1A1bc 0x6fb82008 00000000 0000C000 FFFFF000 00000000 00001000 
[INFO]    002 M1A1bc 0x6fb83008 00000000 0000D000 FFFFF000 00000000 00001000 
[INFO]    003 M1A1bc 0x128f5c0 00000000 0000FF80 FFFFFF80 00000000 00000080 
[INFO]    004 M1A1bc 0x6fb7e008 00000000 00008000 FFFFE000 00000000 00002000 
[INFO]    005 M1A1bc 0x128f440 00000000 0000FE00 FFFFFFE0 00000000 000000A0 
[INFO]    006 M1A1bC 0x6fa7e008 00000000 00000000 FFFFC000 00000000 00004000 
[INFO]    007 M1A1bC 0x6fa82008 00000000 00004000 FFFFC000 00000000 00004000 
[INFO]    008 M1A1bc 0x128f440 00000100 0000FF00 FFFFFF80 00000000 00000080 
[INFO]    009 M1A1bc 0x6fb84008 00000000 00010000 FFFFA000 00000000 00006000 
[INFO] [Environ]: SET_SUPPORT_ACHIEVEMENTS: yes.
[INFO] [SRAM]: Skipping SRAM load..
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 32917.50 Hz.
[INFO] [Video]: Video @ 480x432
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: egl_videocore
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shaders]: Configuration file shader set to .
[INFO] [Shaders]: Configuration file shader set to .
[INFO] [Shaders]: Configuration file shader set to .
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[... 19 lines cut ...]
[INFO] [udev]: Added Device mouse#0 X-BOWS NATURE Mouse (/dev/input/event2) .
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 1024 frames
[INFO] ALSA: Buffer size: 2048 frames
[INFO] [Display]: Found display driver: "gl".
[ERROR] [Font]: Failed to create rendering backend: stb-unicode.
[INFO] [Font]: Using font rendering backend: bitmap.
[ERROR] [Font]: Failed to create rendering backend: stb-unicode.
[INFO] [Font]: Using font rendering backend: bitmap.
[ERROR] [Font]: Failed to create rendering backend: stb-unicode.
[INFO] [Font]: Using font rendering backend: bitmap.
[INFO] [Playlist]: Loading history file: [/opt/retropie/configs/all/retroarch/cores/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/opt/retropie/configs/all/retroarch/cores/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/opt/retropie/configs/all/retroarch/cores/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/opt/retropie/configs/all/retroarch/cores/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/opt/retropie/configs/all/retroarch/cores/content_favorites.lpl].

```

Here's a screenshot of how the screen just frozes:
[image]

tavuntu, 2021-08-12:
That is some useful information, I'll try it and come back later, thanks!

tavuntu, 2021-08-13:
not sure if I got this part (non native English speaker):

"The build script seems to do some auto-detection for KMS and DispmanX, so probably it needs to be re-compiled to change the video driver, instead of changing that config file(s) only: https://github.com/RetroPie/RetroPie-Setup/blob/50ed2e7/scriptmodules/system.sh#L317-L346"

I've added the fake KMS to /boot/config.txt and changed the video_driver in just one emulator (to test). No luck so far.

MichaIng, 2021-08-13:
Then I guess you need to recompile RetroPie, e.g. using their setup/build script I linked above. Or where do you get the executables from for RucoPie?

tavuntu, 2021-08-13:
For the setup I do this:
```
/usr/bin/git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
echo 'Entering folder RetroPie-Setup...'
cd RetroPie-Setup
echo 'Starting RetroPie setup...'
./retropie_setup.sh
```

and for emulators I'm downloading the binary versions.

tavuntu, 2021-08-13:
Oh, but you meant changing something within this function maybe? https://github.com/RetroPie/RetroPie-Setup/blob/50ed2e7/scriptmodules/system.sh#L317-L346

MichaIng, 2021-08-13:
Nope, basically before doing these setups steps, on a DietPi system do:
```sh
apt install libraspberrypi-dev # I think is required to build with DispmanX support
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vs4-fkms-v3d
reboot
```
then the RetroPie setup. It should then detect the DRM/KMS driver and available RPi firmware development libraries and set the build flags accordingly to use it.

tavuntu, 2021-08-13:
ooh I see, I missed installing that dev package, I'll try again, thanks again!

MichaIng, 2021-08-13:
For the KMS driver, it check whether `/sys/module/vc4` exists. And for DispmanX, it then checks whether
```sh
ls -A /sys/bus/platform/drivers/vc4_firmware_kms/*.firmwarekms
```
produces an output. If these checks are correct after enabling the fKMS driver as above, could be checked as well. It should also work to set the two environment variables before starting the setup/build:
```sh
export __has_kms=1 __has_dispmanx=1
```

tavuntu, 2021-08-14:
Thanks, I'm currently targeting this for RB Pi 3.

tavuntu, 2021-08-16:
I've been facing a few issues, one of them being the retropie setup not having the binary packages available for the cores... (so my script fails) I'll get back here once I have something.

tavuntu, 2021-08-16:
Actually, I just noticed this popping up when running my script:
[image]

And this may be the cause for the binary packages not being there, I just ran the same RetroPie setup in RB Pi OS and there it seems ok...

MichaIng, 2021-08-17:
Hmm, does this error show up during the setup script execution or afterwards? It is strange since the setup script explicitly checks all possible RPi video drives/features, including the combination of KMS + DispmanX, which is only available with the fake KMS driver you have enabled. So when these checks are done and these flags set, I'd expect it to build a binary that works with those, instead of the legacy non-GL driver. And what they call "experimental" is btw the default on RPi4 with Raspberry Pi OS, so not experimental anymore at all, but well proven. On earlier RPi models indeed the legacy non-GL driver is still default, but e.g. Amiberry recommends and provides builds against the fKMS driver since years, so do we.

Not sure about the issue with the pre-compiled binaries. I though those are compiled as part of the setup script. Else it probably makes sense that legacy pre-compiled binaries show such outdated warnings, respectively on a modern kernel/system no matching binaries are found. So probably the GPU driver checks I mention above are not done, but some flag is required to actually compile binaries freshly? I'll have a look into the script as well, when I find time.

tavuntu, 2021-08-17:
Thanks you. And that warning dialog shows at the beginning btw. I've tried with both kms and fkms just in case, same result for both (I'm using a RB Pi 3).

tavuntu, 2022-02-26:
So just as an update, I'm still stuck in this, getting a few different issues (being probably just one at the end). I'm getting this while starting RetroPie Setup. It checks for dependencies and tries to install them, this is happens when trying to build the `libsdl2-dev (custom)` package, and I'm not sure why it is "custom".

I already mentioned this to the retropie guys (including the binaries not being available/generated for libretro emulators) but they say it's not their issue since DietPi is not supported, and that makes sense. I'm not trying to place blame on anyone, this is surely an interesting issue and I'll keep looking into it at least from time to time, even tho I really lack knowledge on this stuff (low level like drivers, OS libraries, etc.)

[image]

MichaIng, 2022-02-26:
Try to install the Raspberry Pi firmware headers:
```sh
apt install libraspberrypi-dev
```
It seems to depend on DispmanX, which again depends on fKMS, which needs to be actively enabled (as well on RPi OS btw). You can use `dietpi-config` > Display Options > Resolution and pick `vc4-fkms-v3d`, then reboot, or add it manually:
```sh
G_CONFIG_INJECT 'dtoverlay=vc4-f?kms-v3d' 'vc4-fkms-v3d' /boot/config.txt
reboot
```

tavuntu, 2022-02-26:
Forgot to mention, I actually installed `libraspberrypi-dev` and the KMS driver before the Retro Pie setup, however, I didn't install the "fake" one, I did instead:

`/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d`
`reboot`

I was under the impression fKMS was for the pi4 and KMS for the pi3 and older? Maybe I misunderstood. I'm going to try fKMS anyway. I'll let you know.

MichaIng, 2022-02-26:
No, the fake KMS is no hardware KMS but software (fake) KMS, providing however the same interface. Both work on all RPi models but only the fake KMS has support for legacy RPi GPU capabilities/libraries like DispmanX. RetroPie shouldn't use it anymore, IMO, but indeed it seems to provide still better performance, e.g. Amiberry recommends it as well, while one can choose whether to use it or not.

Probably RetroPie has as well build options to use pure DRM/KMS without DispmanX?

tavuntu, 2022-02-26:
That was quick, thanks for clarifying.

>Probably RetroPie has as well build options to use pure DRM/KMS without DispmanX?

Sorry, I'm not sure about that, I'm not even sure if I understand the question.

MichaIng, 2022-02-26:
As said, DispmanX is a legacy RPi-only API, which allows to efficiently paint display output layers over another, obviously efficiently on RPi which it is tailored for. But it is not supported by the general KMS of mainline Linux, hence can only be made available by emulating/faking the KMS API software wise to hook that DispmanX in.

Raspberry Pi OS Bullseye ships with native mainline KMS enabled by default, hence does not support DispmanX without enabling fake KMS manually, and it is likely that this API will be removed completely in a future Raspberry Pi OS (resp. RPi firmware) release, which would break RetroPie completely if there is not way to build it without using DispmanX but only native KMS instead. E.g. Amiberry supports this for a long time, and RetroPie should, if not the case yet, quickly start doing so as well, to not suddenly break for everyone who uses recent RPi OS or recent RPi kernel/firmware packages. So this is something you could ask RetroPie devs about.

tavuntu, 2022-02-26:
Thanks for that info, makes at least a little more sense now.

MichaIng, 2022-02-26:
I'm not sure which distros RetroPie does support, did you use our ARMv6 image or a different one?

Our ARMv6 image matches, kernel, firmware and repo wise 100% the Raspberry Pi OS 32-bit image, hence aside of a different default package composition (the missing libraspberrypi-dev e.g.) it should generally work the very same way.

Our ARMv8 image matches the Raspberry Pi OS 64-bit similarly, hence when RetroPie supports RPi OS 64-bit, it must generally support that DietPi image as well (always keeping possibly missing header/dev packages in mind).

The ARMv7 image is somehow a mixture of both above: It uses the Debian armhf repository (which same as above the official RPi kernel/firmware repo, of course), so there is no equivalent for this provided by RPi Foundation.

Also note that all our images are Bullseye, which, as mentioned, has quite a number of legacy RPi libraries removed already, as RPi OS Bullseye has. So if RetroPie depends on these, it probably supports Buster images only as well, though DispmanX is generally still supported. The SDL2 version they use is very old: 2.0.10 vs 2.0.20, not sure whether it may cause issues as well, the oldest one I tested on RPi Bullseye was 2.0.14 a while ago, which is as well the one shipped by Raspbian and Debian APT repositories.

tavuntu, 2022-02-27:
>I'm not sure which distros RetroPie does support, did you use our ARMv6 image or a different one?

Yes, I used that one. And I'm not 100% sure if it's supported or not but a contribuitor in that other issue told me so.

MichaIng, 2022-02-27:
Found this issue: https://github.com/RetroPie/RetroPie-Setup/issues/3482#issuecomment-1043736381

Looks like you can change the SDL2 branch? On your screen I saw v2.0.10, please try to switch to v2.0.20.

tavuntu, 2022-02-27:
Thanks, I'll try switching the branch and let you know.

tavuntu, 2022-05-20:
Alright, using a newer version of SDL solved the issue I was getting (EGL_DISPMANX_WINDOW_T and stuff). I did change [this line](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/sdl2.sh#L19) to be 2.0.22 since that is the latest (I hope that's ok). The issue about the binaries not being available remaing tho. I mentioned something [in here](https://github.com/RetroPie/RetroPie-Setup/issues/3487).

shibmaster, 2023-07-17:
Hi all,
big ups to the dietpi community!
I am trying to install retropie on a fresh Dietpi install running bookworm on RPi 4B. 
Following the first post, I notice the numbers in the dietpi-software seem to have changed slightly:

`sudo dietpi-software install 5 6 16 17
[sudo] password for pi: 
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing ALSA: Advanced Linux Sound Architecture
[  OK  ] DietPi-Software | Installing X.Org X Server: aka X11 - X Window System implementation
[  OK  ] DietPi-Software | Installing microblog.pub: A self-hosted, single-user, ActivityPub powered microblog.
[  OK  ] DietPi-Software | Installing Git: Clone and manage Git repositories locally
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
`
So after this I installed with `apt install build-essential` and left micronlog.pub installed.
I changed the memory split in dietpi-config to 256MB GPU and rebooted.
I then proceeded to installing retropie from a non-root account:

`git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git`
`cd RetroPie-Setup`
`sudo bash geretropie_setup.sh`

I chose basic install, which encountered some errors: (Excerpt from the log file created by Retropie-setup)
= = = = = = = = = = = = = = = = = = = = =

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libavcodec-dev : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
 libavdevice58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
                 Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)
 libavfilter-dev : Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)
 libavformat58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
                 Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable
E: Unable to correct problems, you have held broken packages.
Could not install package(s): libusb-1.0-0-dev libavcodec-dev libavformat-dev libavdevice-dev.
Update is available - updating ...

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'emulationstation' : EmulationStation - Frontend used by RetroPie for launching emulators
= = = = = = = = = = = = = = = = = = = = =

Did not find needed dependencies: libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev cmake libvlc-dev libvlccore-dev vlc rapidjson-dev omxplayer
. Trying to install them now.
Reading package lists...
Building dependency tree...
Reading state information...
Package omxplayer is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'omxplayer' has no installation candidate
Could not install package(s): libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev cmake libvlc-dev libvlccore-dev vlc rapidjson-dev omxplayer.
Update is available - updating ...

= = = = = = = = = = = = = = = = = = = = =

I tried resolving the dependencies with aptitude but no luck so far.

Am I attempting something impossible at the moment?

Thanks fo hints on how to resolve this.

MichaIng, 2023-07-17:
Indeed the software IDs have changed, notable the build-essential option has been removed, as one can just do `apt install build-essential` as you did, or install the individual required compilers and headers.

Is omxplayer not depending on the legacy graphics stack, and hence incompatible with KMS and obsolete? Also the package has not seen any update since 2019. And yes, it is not even present in the Bullseye suite of the RPi repo: https://archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/Packages

So if RetroPie really still required omxplayer, then it cannot run on any supported Debian or Raspbian version anyway.

Checking its installer scripts: https://github.com/RetroPie/RetroPie-Setup/blob/86f6588/scriptmodules/supplementary/emulationstation.sh#L140-L142
So it attempts to install it only if DispmanX is available, hence either legacy graphics stack enabled or fake KMS. But with real KMS, it won't be installed. So try:
```sh
G_CONFIG_INJECT 'dtoverlay=' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
```
Then repeat the RetroPie installer.

shibmaster, 2023-07-19:
Dear Micha and Joulinar,<
Thank you so much for your lightning fast responses and apologies I could not test the suggestions right away.
I used the following to successfully avoid the omxplayer situation:
`G_CONFIG_INJECT 'dtoverlay=' 'dtoverlay=vc4-kms-v3d' /boot/config.txt`
Thanks for that!

After a reboot I ran basic install again in the retropie_setop.sh
It failed to install a few packages but I was able to resolve most with aptitude, leaving only vlc.

There were a few more "not installable" dependencies to the vlc package, of which one is the libsrt from before.
Joulinar suggested the fix quoted above. Thanks for the tip!

Should I attempt something similar for the other "not installable" dependencies?

 vlc-plugin-base : Depends: libavcodec58 (>= 7:4.2)
                   Depends: libavfilter7 (>= 7:4.0)
                   Depends: libavformat58 (>= 7:4.2) but it is not going to be installed
                   Depends: libdav1d4 (>= 0.1.0) but it is not installable
                   Depends: libflac8 (>= 1.3.0) but it is not installable
                   Depends: libprotobuf-lite23 (>= 3.12.4) but it is not installable
                   Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable
                   Depends: libx264-160 (>= 2:0.160.3011+gitcde9a93) but it is not installable
                   Depends: libx265-192 (>= 3.2) but it is not installable

vlc-plugin-video-output : Depends: libplacebo72 (>= 0.4.0) but it is not installable

As in:
`
cat << '_EOF_' > /etc/apt/preferences.d/dietpi-vlc
Package: libflac* libav* libprotobuf* libsrt* libx26* libplacebo*
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
_EOF_
apt update
apt install vlc
vlc -version
`

Or something similar?
Thanks again for your help :-)

shibmaster, 2023-07-20:
Sort of went another way.
I installed vlc directly with 
`sudo aptitude install vlc`

which completed using the newer dependencies listed on 
[https://packages.debian.org/bookworm/vlc](url)
especially the ones from vlc-plugin-base, which have newer versions.

Then I ran retropie_setup.sh basic install again and it was looking to install
libx11-xcb-dev libavcodec-dev libavformat-dev libavdevice-dev

I aborted the script and used aptitude again for all of these at once, and it managed to find a solution
where only the last one could not be installed.

I tried again the retropie_setup.sh basic install and it ran through with no issues. Did not even try to install anything again, but began compiling immediately.

I rebooted and launched emulationstation with no issues (Sound not working, but that I know from before).
Even ran mupen64plus and it runs Mario Kart smoothly :+1

Cannot thank you guys enough!

Best:-)

[6 more replies]