MichaIng/DietPi #1654: Cron | journalctl shows CRON job gets repeatedly called once every minute.

Fourdee, 2018-03-28:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=2982#p11505

MichaIng, 2018-03-29:
https://github.com/ayufan-rock64/linux-build/blob/master/package/root/etc/cron.d/make_nas_processes_faster

Part of Rock64 image, ugly implementation, spammy like this, not needed for users that don't use these NAS services, should be enough to call the command ONCE after boot, or am I wrong?

@Fourdee 
Do we wan't to remove that file on DietPi-Prep? Or `mv /etc/cron.d/make_nas_processes_faster /etc/cron.d/.make_nas_processes_faster` and activate again on samba/nfs/ftp installation, if this is really the best way to achieve process priorities?

MichaIng, 2018-03-30:
Okay also present on NanoPi Neo, we have to check where this comes from, base image or some APT package update and check if this is really helpful for file severs or not..

Fourdee, 2018-03-30:
@MichaIng 

> Do we wan't to remove that file on DietPi-Prep? Or mv /etc/cron.d/make_nas_processes_faster 

I believe, we should remove the file on all images, let ```dietpi-process_tool``` (and the user) handle it.

> Okay also present on NanoPi Neo

Interesting. cant see it in ```openmediavault_0.2.3_all.deb```, but there are multiple packages i've not checked.

In the preimage used (nanopi-neo_debian-nas-jessie_4.14.0_20171218) OMV is installed, so it must be a OMV addition.

Either way, lets remove the file from all images PREP + patch?

MichaIng, 2018-03-30:
@Fourdee 
> Either way, lets remove the file from all images PREP + patch?

Jep, I think so, PR is up 😃: https://github.com/Fourdee/DietPi/pull/1665

Fourdee, 2018-03-31:
@MichaIng 

Many thanks for the fix. I'll mark this as closed PR should resolve: https://github.com/Fourdee/DietPi/pull/1665

closed 2018-03-31 by Fourdee