MichaIng/DietPi #1655: DietPi 6.5 - apt-get errors... ?

SuBLiNeR, 2018-03-28:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.5
- SBC Device (EG: RPi 3)? RPi3

apt-get update gets you IGN or Error on fetching files from source.

rawwalk13, 2018-03-29:
I got the same problem today on my RPI2 & RPI3.
What i did is just add another working repo to the /apt/sources.list

That solved the problem instantly,  although i dont know if that was the right thing to do .

Fourdee, 2018-03-29:
@SuBLiNeR @rawwalk13 

Thanks for the report 👍 

Confirmed, when upgrading ```raspberrypi-sys-mods``` package which fails to correctly apply the patch (not our bug, but we'll attempt fix).
https://github.com/Fourdee/DietPi/issues/1659

I'll mark this as closed in favor of https://github.com/Fourdee/DietPi/issues/1659

closed 2018-03-29 by Fourdee

Fourdee, 2018-03-29:
@SuBLiNeR @rawwalk13 
Temp fix for now, to get it working until we can fully resolve:
https://github.com/Fourdee/DietPi/issues/1659#issuecomment-377301370