MichaIng/DietPi #1656: DietPi-Update v6.5 | Allo firmware installed on non AlloGUI RPi?

MichaIng, 2018-03-29:
@Fourdee 
https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L529-L538
Shouldn't this be done just on RPi using AlloGUI?

Fourdee, 2018-03-29:
@MichaIng 

> Shouldn't this be done just on RPi's that use AlloGUI?

As this card can be installed via ```dietpi-config```, ideally, we need it preinstalled for all.

Although, we could do a "ondemand net install" of the firmware through ```dietpi-set_hardware```, for those cards, if needed.

MichaIng, 2018-03-29:
@Fourdee 
> Although, we could do a "ondemand net install" of the firmware through dietpi-set_hardware, for those cards, if needed.

Jep, would prefer that way, keeping image/systems as slim as possible. But no need for urgent patch, as the firmware is just ~500k.

MichaIng, 2018-04-19:
PR: https://github.com/Fourdee/DietPi/pull/1718

Fourdee, 2018-04-22:
Good to go 👍 Marking as completed.

closed 2018-04-22 by Fourdee