MichaIng/DietPi #1658: Deluge error

bokiroki, 2018-03-29:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version  -  v6.5  
- SBC Device  -  RPi 3 Model B (armv7l)
- Power supply used  -  official raspberry pi power supply
- SD card used  -  samsung evo+ 32 gb

Dietpi is installed yesterday, with samba, deluge, sonarr, radarr, plex..

i got this failed message on Deluge few times after install... for example i just exited dietpi-drive_manager and got this message..
[image]

Fourdee, 2018-03-29:
Confirmed. Although process appears to be running, most likely a swap in processes during launch.

Fourdee, 2018-03-29:
🈯️ Resolved for v6.6 (and patched) using systemD forking process (which waits for the fork to complete):
```
root@DietPi:~# dietpi-services start
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : deluged
[  OK  ] DietPi-Services | start : deluge-web
[  OK  ] DietPi-Services | start : cron
[ SUB1 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Cron (6386) : Nice      0
[  OK  ] DietPi-Process_tool | Cron (6386) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Cron (6386) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Deluge (Daemon) (6364) : Nice      0
[  OK  ] DietPi-Process_tool | Deluge (Daemon) (6364) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Deluge (Daemon) (6364) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Deluge (Web interface) (6375) : Nice      0
[  OK  ] DietPi-Process_tool | Deluge (Web interface) (6375) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Deluge (Web interface) (6375) : Scheduler SCHED_OTHER 0
```

closed 2018-03-29 by Fourdee