MichaIng/DietPi #1659: Raspbian mirror changed to "raspbian.raspberrypi.org"

MichaIng, 2018-03-29:
I just found the new version of `raspberrypi-sys-mods` trying to auto-adjust the Raspbian mirrors from [archive.raspbian.org](archive.raspberrypi.org) and [mirrordirector.raspbian.org](mirrordirector.raspbian.org) to [raspbian.raspberrypi.org](raspbian.raspberrypi.org) on APT upgrade.

All 3 repos are (still) active providing the same set of packages, so I am not sure if this means that the first two will be shut down by times or something, moving everything into [raspberrypi.org](raspberrypi.org) domain, but this needs to be observed at least I think.

[raspbian.raspberrypi.org](raspbian.raspberrypi.org) is not reachable via https and would be interesting to know if mirror direction is included. I didn't find any new/hint/info about this repo anywhere, not on [raspbian.org](raspbian.org) nor [raspberrypi.org](raspberrypi.org) nor anywhere via search machines.

Fourdee, 2018-03-29:
Notes:
```
root@DietPi:~# cat /etc/apt/sources.list
deb https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch main contrib non-free rpi

###

root@DietPi:~# apt-get dist-upgrade
raspberrypi-sys-mods
Updating apt sources from {mirrordirector,archive}.raspbian.org to raspbian.raspberrypi.org...

###

root@DietPi:~# cat /etc/apt/sources.list
deb https://www.mirrorservice.org/sites/raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
```

Fourdee, 2018-03-29:
#### Temp workaround/fix:
```
cat << _EOF_ > /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
_EOF_
G_AGUP
```

Fourdee, 2018-03-29:
@SuBLiNeR

> why don't they know about this bug at theirs repo?

Feel free to tell them 😃 

Its a terrible patch which does not accommodate any changes to ```/etc/apt/sources.list```. I can only assume it works with Raspbian untouched (untested).

SuBLiNeR, 2018-03-29:
@Fourdee that's why it wasn't working:
it was this in mine: https://www.mirrorservce.org/...... ?
Now as i edited it to only go directly to via http://rasbian.raspberry.org it is working...

Fourdee, 2018-03-30:
Ok:
- Existing images should be fine, we do not run ```G_AGDUG```
- Patch will run ```G_AGDUG``` to force the update, then reset ```/etc/apt/sources.list``` with the new default ```http://raspbian.raspberrypi.org/raspbian```

MichaIng, 2018-03-30:
@Fourdee 
I think there will be no fast fix for this, thus if we want to prevent users from running into this issue, we need to switch the mirror ourself correctly. I hate that there is really no other way and we need to drop mirrordirector for now... Besides, we could also add our default mirrordirector to some sources.list.d/ file instead, leaving sources.list empty, but this is everything else than beautiful as well...

So:
```
if grep -qE '(mirrordirector|archive).raspbian.org' /etc/apt/sources.list; then

    sed -iE "\#(mirrordirector|archive).raspbian.org#c\deb http://raspbian.raspberrypi.org/raspbian $G_DISTRO_NAME main contrib non-free rpi#g" /etc/apt/sources.list

fi
```
And replacing DietPi default mirror and dietpi-config selection menu as well.

MichaIng, 2018-03-30:
@Fourdee 
Jep, I thought wrong, this is the only way, great job 😃!

Fourdee, 2018-03-30:
Ok, great work on this @MichaIng 👍 

Tested RPi with patch, running fine. I'll mark this as completed. We'll need to keep an eye out for when/if ```http://raspbian.raspberrypi.org/raspbian``` goes ```https```, then update accordingly.

closed 2018-03-30 by Fourdee

Fourdee, 2018-03-30:
REF: https://twitter.com/DietPi_/status/979761183359107077