MichaIng/DietPi #1663: Patchfile Update Issue 6.4 to 6.5

buffalo0207, 2018-03-30:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v6.4
- SBC Device     | Asus Tinker Board (armv7l) (index=52)
- Distro         | stretch (index=4)
- Command        | G_AGP: dphys-swapfile
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Patchfile

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
The following package was automatically installed and is no longer required:
  dc
Use 'apt autoremove' to remove it.
The following packages will be REMOVED:
  dphys-swapfile*
0 upgraded, 0 newly installed, 1 to remove and 42 not upgraded.
1 not fully installed or removed.
After this operation, 61.4 kB disk space will be freed.
Setting up util-linux (2.29.2-1+deb9u1) ...
insserv: warning: script 'K16halt' missing LSB tags and overrides
insserv: Script console-setup.sh is broken: missing end of LSB comment.
insserv: exiting now!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package util-linux (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 util-linux
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

yuret, 2018-03-30:
Same problem on RPI3. After that shows on login:
 ───────────────────────────────────────
 DietPi     | 18:54 | Fri 30/03/18
 ───────────────────────────────────────
 v6.5       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 NULL       |
 ───────────────────────────────────────
And external Wi-Fi module stops working:
[   18.670513] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   18.670547] rt2800usb 1-1.4:1.0: Direct firmware load for rt2870.bin failed with error -2
[   18.670554] ieee80211 phy0: rt2x00lib_request_firmware: Error - Failed to request Firmware

Fourdee, 2018-03-30:
@buffalo0207 

Thanks for the report 👍 

> Setting up util-linux (2.29.2-1+deb9u1) ...
insserv: warning: script 'K16halt' missing LSB tags and overrides
insserv: Script console-setup.sh is broken: missing end of LSB comment.
insserv: exiting now!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package util-linux (--configure):

Strange, appears the service script used by ```util-linux``` is broken.

Please paste results of the offending file:
```cat /etc/init.d/console-setup.sh```

Fourdee, 2018-03-30:
@buffalo0207 

> Direct firmware load for rt2870.bin failed with error -2

You'll need to install this package for the firmware (https://packages.debian.org/stretch/firmware-misc-nonfree):
```
G_AGI firmware-misc-nonfree
```

Fourdee, 2018-04-05:
No response, marking as closed. Please reopen if required.

closed 2018-04-05 by Fourdee