MichaIng/DietPi #1667: Updated rpi-update and dietpi-update and this is what happend after

skane165, 2018-03-31:
#### Required Information:
- DietPi Version | v6.5
- SBC Device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | Connection test: https://www.mirrorservice.org/sites/raspbian.raspberrypi.org/raspbian
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-03-31 07:07:04--  https://www.mirrorservice.org/sites/raspbian.raspberrypi.org/raspbian
Resolving www.mirrorservice.org (www.mirrorservice.org)... 212.219.56.184, 2001:630:341:12::184
Connecting to www.mirrorservice.org (www.mirrorservice.org)|212.219.56.184|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

SuBLiNeR, 2018-03-31:
@skane165 Hi there, this problem was already discussed here. See this for reference and you'll have it fixed for now https://github.com/Fourdee/DietPi/issues/1659 #1659

Fourdee, 2018-03-31:
I'll mark this closed, solution via @skane165 post https://github.com/Fourdee/DietPi/issues/1667#issuecomment-377672803

closed 2018-03-31 by Fourdee