MichaIng/DietPi #1668: DietPi issuse

darioristic, 2018-03-31:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-03-31:
@darioristic 

Hi,

Appears to be online here. If the problem still persist, please can you answer the following:
- Is PiHole installed?
- Paste results of ```route```
- DHCP or STATIC IP? Check in ```dietpi-config``` > Networking Options: Adapters

SuBLiNeR, 2018-04-01:
@darioristic @Fourdee Same procedure as everytime - ppl should read your fix upon this problem... it's not the PiHole stuff - it is the problem of the changed repo in the sources.list.

-> nano /etc/apt/sources.list and change the entry to http://raspbian.raspberrypi.org/raspbian #1659

MichaIng, 2018-04-03:
@darioristic 
Not in this case, the mirror URL is still correct, seemed to be a temporary unavailable `www.mirrorservice.org` or indeed some network configuration issue.

But to avoid any feature issue with this, it is no bad idea to anyway adjust `/etc/apt/sources.list` accordingly, if already touching the system 🙂.

SuBLiNeR, 2018-04-04:
@MichaIng 
But this is how i solved this even before #1659 was than marked as being temp solved. Nah, the mirrorservice.org is always a bottleneck since the past.... :1st_place_medal:

Fourdee, 2018-04-07:
No response, marking as closed. Please reopen if required.

closed 2018-04-07 by Fourdee