MichaIng/DietPi #1669: USB 1.0 DAC on Nanopi

paulstephane, 2018-03-31:
## Creating a bug report/issue:

Hello, on the nanopi neo image there seems to be no possibility to activate the USB audio as a USB 1 driver, similar to what can be done on the sparky board (changing uEnv file^
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

SuBLiNeR, 2018-04-01:
@paulstephane I suggest you're using an armbian install as base? - see for reference: http://dietpi.com/phpbb/viewtopic.php?t=1896 - there is a problem with these dev-kernels and i'm not sure if this is the problem at least for your case...

paulstephane, 2018-04-01:
I will look into it, thanks

closed 2018-04-01 by paulstephane