MichaIng/DietPi #1672: Orange pi zero updgrade from Armbian Debian Jessie to DietPi

SmoothSilence, 2018-04-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```DietPi Beta(I think) for "othet" SBC```
- SBC Device (Orange Pi zero H2+ (512))
- Power supply used (5v 2A )
- SD card used (Sandisk ultra 16 Gb class 10 I)?
- Distro (Armbian-Debian-Jessie) 

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

The Orange Pi zero shuld finish the upgrade to the DietPi and reboot. Booting me to the DietPi menu

#### Actual behaviour:

The installation has been terminated. With this message.
[image]

After pressing okay I rebooted the pi, I tried to log into it via SSH but it showed the 'connection timed out' error message.

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

SuBLiNeR, 2018-04-01:
@SmoothSilence Hi there, if you like to go from ARMBIAN to DietPi, you should do following steps:

1) Get actual latest image for your device (OPi Zero) best is Debian (Stretch)
2) after succesful boot into "armbian", do a firmware update via armbian-config
2a) Reboot!!!
3) manually install the above mentioned package - (e.g. apt-get install firmware-realtek)
4) than do the steps with the PREP-script.
5) You should now be on DietPi v6.x after completation of the script and some reboots afterwards.

Please give feedback if this works for you.... as it worked for myself without any problems on this device.

Fourdee, 2018-04-07:
No response, marking as closed. Please reopen if required.

closed 2018-04-07 by Fourdee