MichaIng/DietPi #1674: Dietpi.txt dynamic to static ip

unclehowell, 2018-04-01:
This is a feature request for a solution which will make the dynamic IP static during first run. I was hoping to modify the dietpi.txt to allow the pi to receive the IP address and gateway from the router it is connected to then immediately make it static , but with full autonomy without having to enter the menu etc

Fourdee, 2018-04-01:
@unclehowell 

Interesting idea, thanks for the request 👍 

My only concerns is the router providing the copied IP to another DHCP client, at some point in the future, causing IP conflict.

MichaIng, 2018-04-01:
Jep I actually do the same manually, as my router provides permanent IPs for known devices. By this DHCP server does not need to run etc.

This should only be an optional feature, as suggested via dietpi.txt, in cases user knows that the provided IP will always be reserved for DietPi device.

unclehowell, 2018-04-01:
I think I'll have to use the custom script part of dietpi.txt to get the IP its been assigned and then use it to create a static. But I want to try and keep within the parameters dietpi gives us to avoid deviating from their insisted methods of modifications , for as long as possible.

MichaIng, 2018-10-30:
@Fourdee 
Puhh, wanted to do a start on this, as I like it quite much. No fixed DHCP lease need to be configured or estimated and added to `dietpi.txt`. Theoretically this could be even done on every boot, so wherever you boot your Pi (in what network), it will automatically estimate DHCP, apply as static, thus does not need have DHCP client running in the background.

But this requires to gateway and netmask estimation within boot script and apply to only the correct interface at best. Many of this estimations and settings applying would be doubled with what dietpi-config (and as well preboot) does. I don't like so much doubled code. Would prefer then to estimate more network details within the `obtain_network_details` script, e.g. gateway and netmask as well of both, eth and wifi interfaces, if connected. Then as well add e.g. `net_interface` option to `dietpi-set_hardware` to apply either static or dhcp settings to /etc/network/interfaces.

But I am a bid lost in how to solve this in detail. Usually we use set_hardware/software functions to take 1-2 input arguments apply those and write back to `dietpi.txt`. In this case, it is more complicated:
- Applying static IP settings requires too much values to give those as function arguments (from my point of view?). Those need to be saved in some files to read back by the set_script. `dietpi.txt` would be the obvious solution but then somehow the usual use of set_scripts would be upside down. Read and apply settings from dietpi.txt instead of apply and write back to dietpi.txt. /DietPi/dietpi/.network is no good solution, since it should only contain currently active settings/values and not the target ones.
- Applying current DHCP leased IP/mask/gateway/DNS as static would need to work the other way round: Read values from `/DietPi/dietpi/.network` and write those as static back to `dietpi.txt`. Or leave those as they are, as we currently do not touch dietpi.txt as well when applying static IP via dietpi-config?  Also differentiation between eth and wifi would be good in case of hotspot or other wifi+eth double use...
- Only setting DHCP is really easy, perhaps too simple to create a separate set_hardware function for this.

Maybe I am thinking too complicated currently... Maybe tomorrow I see this a bid more sorted 🤣.

Fourdee, 2018-10-31:
@MichaIng 

Yep, issue is most of the code you need for current IP and valid detection is in ```dietpi-config``` ```Network_GetData()```. Might need to move this outside to ```set_hardware```.

Once the above is obtained, set a static IP. Again, the code for this is in ```dietpi-config```.

Basically a rewrite is needed if we want to do this efficiently, by moving the dietpi-config code to a shared function we can call.

MichaIng, 2018-10-31:
@Fourdee 
> A quick/messy method would be add a input for dietpi-config get_net etc.

Hmm, better for my impression as a quick implementation would be indeed to double all required interface estimation inside /boot first run function, instead of further messing the dietpi-config code-monster :smiley:.

But since this is a non-important request, I am more keen to step-by-step outsource code from dietpi-config to shared scripts/functions. Perhaps we can make a start by doing all interfaces IP+gateway+netmask estimation (if device is connected of course) inside `obtain_network_details`. Then using simple `.network` scrapes within dietpi-config and other network related scripts (e.g. as well dietpi-cloudshell). To not rise script time too much, I would then implement an enhanced interface loop: https://github.com/Fourdee/DietPi/issues/1510#issuecomment-434557203

Jep I think that's it. Step-by-step doing consistent code sorting, then when resulting "API" is as desired, we can add network-related features based on it 😃. I will do a start later.

unclehowell, 2019-01-09:
I want to thank you for your response to this. My project is https://makeitwave.com , just starting out but the basic idea is a UI for dietpi. So the user/ installer can plug in the box, connect to its WiFi, then complete set-up via a webUI

closed 2021-06-26 by MichaIng